Koenkk/zigbee2mqtt #4214: brightness stays 0 even when queried with get or updates via reporting

sjorge, 2020-08-30:
# Bug Report
## What happened

I noticed a lot of bulbs stick with brightness=0, I think this started happening with a change for homeassistant where state=off also publishes brightness=0 (which is not something that might be always wanted, but I am working around this for now in node-red with ignore brightness=0 when state is set to off.

But I was having lots of bulbs stuck on zero so I started to ding into the mqtt messages, asusming maybe some had fail to setup reporting for genLevelCtrl, but I noticed the following:

**get just the brightness for a panel that is turned ON**
```
Zigbee2MQTT:debug 2020-08-30 12:45:00: Received MQTT message on 'zigbee2mqtt/office/lights/panel1/get' with data '{"brightness": ""}'
Zigbee2MQTT:debug 2020-08-30 12:45:00: Publishing get 'get' 'brightness' to 'office/lights/panel1'                                                                                                                                                                                               Zigbee2MQTT:debug 2020-08-30 12:45:00: Received Zigbee message from 'office/lights/panel1', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-30 12:45:01: MQTT publish: topic 'zigbee2mqtt/office/lights', payload '{"brightness":0,"color":{"x":0.459857791746717,"y":0.410598847355503},"color_temp":370,"state":"OFF"}'
Zigbee2MQTT:info  2020-08-30 12:45:01: MQTT publish: topic 'zigbee2mqtt/office/lights/panel1', payload '{"brightness":0,"color_temp":370,"linkquality":60,"state":"OFF","update":{"state":"idle"}}'
```

Looks like it properly reads the currentLevel as 254 (100%)! But it then does not publish this! Brightness stays at 0 for office/lights/panel1! (And for the group it is part of)

**waiting for an attReport**
```
Zigbee2MQTT:debug 2020-08-30 12:45:13: Received Zigbee message from 'office/lights/panel1', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0                                                                                            Zigbee2MQTT:debug 2020-08-30 12:45:13: Received Zigbee message from 'office/lights/panel1', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":370}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-30 12:45:14: MQTT publish: topic 'zigbee2mqtt/office/lights', payload '{"brightness":0,"color":{"x":0.459857791746717,"y":0.410598847355503},"color_temp":370,"state":"ON"}'                                                                                                               Zigbee2MQTT:info  2020-08-30 12:45:14: MQTT publish: topic 'zigbee2mqtt/office/lights/panel1', payload '{"brightness":0,"color_temp":370,"linkquality":60,"state":"ON","update":{"state":"idle"}}'
```

(Looks like filtering out properties doesn't work for groups either, I'll file an issue for that too later.)

Same result, brightness stays at 0?

**Maybe it needs to have both state and brighness freshly received?**

```
Zigbee2MQTT:debug 2020-08-30 12:46:27: Received MQTT message on 'zigbee2mqtt/office/lights/panel1/get' with data '{"brightness": "", "state": ""}'
Zigbee2MQTT:debug 2020-08-30 12:46:27: Publishing get 'get' 'state' to 'office/lights/panel1'
Zigbee2MQTT:debug 2020-08-30 12:46:27: Received Zigbee message from 'office/lights/panel1', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-30 12:46:27: Received Zigbee message from 'office/lights/panel1', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-30 12:46:27: Received Zigbee message from 'livingroom/floor_light/bulb', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":555,"currentX":45774,"currentY":19622}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-30 12:46:28: MQTT publish: topic 'zigbee2mqtt/office/lights', payload '{"brightness":0,"color":{"x":0.459857791746717,"y":0.410598847355503},"color_temp":370,"state":"ON"}'
```

Nope, that doesn't seem to work either! 

## What did you expect to happen
Brightness to not be... well 0 all the time.

## How to reproduce it (minimal and precise)
- turn off and on a trådfri bulb
- wait for an update of brightness
- nothing happens, although looks like the zigbee message do get received correct and have the correct value.

## Debug Info
```
[root@amethyst ~]# mosquitto_sub -t zigbee2mqtt/bridge/info | jq .
{
  "commit": "6452500",
  "coordinator": {
    "meta": {
      "maintrel": 3,
      "majorrel": 2,
      "minorrel": 6,
      "product": 0,
      "revision": 20200728,
      "transportrev": 2
    },
    "type": "zStack12"
  },
  "log_level": "info",
  "network": {
    "channel": 11,
    "extended_pan_id": "0x31d4151f5fefd6c3",
    "pan_id": 6753
  },
  "permit_join": false,
  "version": "1.14.3-dev"
}
```

Koenkk, 2020-08-30:
When the state is OFF, the brightness returned will always be 0 (even if reported otherwise by the device). For Zigbee2MQTT brightness 0 always is OFF, anything else is ON. This is done here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c980b9ccab975892d9b3c0bd03ecdd92e9a8c371/converters/fromZigbee.js#L614

This explains: **get just the brightness for a panel that is turned ON**, the state is OFF here as can be seen in the MQTT message. You wrote that the bulb is ON here, but Zigbee2MQTT doesn't seem to know about that.

The other 2 reports are indeed wrong, as can be seen the state is ON, however still the brightness values are ignored. This is not correct and should be investigated further.

Above this line: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c980b9ccab975892d9b3c0bd03ecdd92e9a8c371/converters/fromZigbee.js#L614 can you add `console.log(meta.state)` and provide the logging of this?

sjorge, 2020-08-30:
I’ll try to get the extra info later this week.

sjorge, 2020-08-31:
> This explains: get just the brightness for a panel that is turned ON, the state is OFF here as can be seen in the MQTT message. You wrote that the bulb is ON here, but Zigbee2MQTT doesn't seem to know about that.

I wonder if this is because reporting sometimes fails (pending testing new fix form other issue) and that i might get updates for genLevelCtrl but not for genOnOff so when the bulb is toggled via a remote, things don't match.

I've added the console.log(), but won't have much time to play today.

sjorge, 2020-08-31:
OK so the above theory is definitely incorrect! I decided to toggle a bulb to be sure and already hit the issue on first go.

```
Zigbee2MQTT:debug 2020-08-31 14:21:11: Received Zigbee message from 'office/desk_light/remote', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 10112
Zigbee2MQTT:debug 2020-08-31 14:21:11: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-31 14:21:11: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
{
  brightness: 0,
  color: { hue: 30, saturation: 78, x: 0.4608, y: 0.3982 },
  color_mode: 1,
  color_temp: 374,
  linkquality: 39,
  state: 'OFF'
}
Zigbee2MQTT:info  2020-08-31 14:21:12: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"action":"on","battery":60,"linkquality":44,"update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-08-31 14:21:12: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":0,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"ON"}'
Zigbee2MQTT:info  2020-08-31 14:21:12: MQTT publish: topic 'zigbee2mqtt/office/desk_light/bulb', payload '{"brightness":0,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_mode":1,"linkquality":39,"state":"ON"}'
Zigbee2MQTT:debug 2020-08-31 14:21:17: Received Zigbee message from 'office/lights/panel3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":370}' from endpoint 1 with groupID 0
```

Koenkk, 2020-09-01:
The debounce might indeed explain it, does the issue disappear when not using debounce?

sjorge, 2020-09-01:
It does indeed seem debounce related
```
Zigbee2MQTT:debug 2020-09-01 14:20:19: Received Zigbee message from 'office/desk_light/remote', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 10112
Zigbee2MQTT:debug 2020-09-01 14:20:19: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-01 14:20:19: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":0,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"ON"}'
Zigbee2MQTT:info  2020-09-01 14:20:19: MQTT publish: topic 'zigbee2mqtt/office/desk_light/bulb', payload '{"brightness":0,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_mode":1,"linkquality":52,"state":"ON"}'
Zigbee2MQTT:debug 2020-09-01 14:20:19: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
{
  brightness: 0,
  color: { hue: 30, saturation: 78, x: 0.4608, y: 0.3982 },
  color_mode: 1,
  color_temp: 374,
  linkquality: 52,
  state: 'ON'
}
Zigbee2MQTT:info  2020-09-01 14:20:19: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"ON"}'
Zigbee2MQTT:info  2020-09-01 14:20:19: MQTT publish: topic 'zigbee2mqtt/office/desk_light/bulb', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_mode":1,"linkquality":52,"state":"ON"}'
Zigbee2MQTT:info  2020-09-01 14:20:20: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"action":"on","battery":60,"linkquality":49,"update":{"state":"idle"}}'
```

Koenkk, 2020-09-01:
Does the following change fix it with debounce enabled: https://github.com/Koenkk/zigbee2mqtt/commit/bdde7fba9d9fcac4683d93a6b5f88ae27ad8caf4 ?

sjorge, 2020-09-01:
```
[root@amethyst /opt/zigbee2mqtt]# curl https://github.com/Koenkk/zigbee2mqtt/commit/bdde7fba9d9fcac4683d93a6b5f88ae27ad8caf4.patch | patch -p1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1433  100  1433    0     0   1933      0 --:--:-- --:--:-- --:--:--  1936
patching file lib/extension/receive.js
[root@amethyst /opt/zigbee2mqtt]# svcadm restart zigbee2mqtt
[root@amethyst /opt/zigbee2mqtt]#
```

Will let you know if I see an improvement with debounce

sjorge, 2020-09-01:
It's doing more weird stuff now

```
Zigbee2MQTT:debug 2020-09-01 19:39:19: Received Zigbee message from 'office/floor_light/remote', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 10111
Zigbee2MQTT:debug 2020-09-01 19:39:19: Received Zigbee message from 'office/floor_light/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-01 19:39:19: MQTT publish: topic 'zigbee2mqtt/office/floor_light', payload '{"brightness":254,"color":{"x":0.4608,"y":0.3982},"color_temp":374,"state":"OFF"}'
Zigbee2MQTT:info  2020-09-01 19:39:20: MQTT publish: topic 'zigbee2mqtt/office/floor_light/remote', payload '{"action":"off","battery":87,"linkquality":84,"update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-09-01 19:39:20: MQTT publish: topic 'zigbee2mqtt/office/floor_light', payload '{"brightness":254,"color":{"x":0.4608,"y":0.3982},"color_temp":374,"state":"OFF"}'
Zigbee2MQTT:info  2020-09-01 19:39:20: MQTT publish: topic 'zigbee2mqtt/office/floor_light/bulb', payload '{"brightness":254,"color":{"x":0.4608,"y":0.3982},"linkquality":68,"state":"OFF"}'
```

So the wanted behavior I believe as when the state is off, we want brightness to be 0 right?
I also still had the console.log() and the meta.state was not logged, but a genLevelCtrl was not reported either.

sjorge, 2020-09-01:
```
Zigbee2MQTT:debug 2020-09-01 19:43:09: Received Zigbee message from 'office/desk_light/remote', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 10112
Zigbee2MQTT:debug 2020-09-01 19:43:09: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-01 19:43:09: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":0,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"ON"}'
Zigbee2MQTT:debug 2020-09-01 19:43:09: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
{
  brightness: 0,
  color: { hue: 30, saturation: 78, x: 0.4608, y: 0.3982 },
  color_mode: 1,
  color_temp: 374,
  linkquality: 89,
  state: 'ON'
}
Zigbee2MQTT:info  2020-09-01 19:43:09: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"ON"}'
Zigbee2MQTT:info  2020-09-01 19:43:10: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"action":"on","battery":87,"linkquality":84,"update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-09-01 19:43:10: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"ON"}'
Zigbee2MQTT:info  2020-09-01 19:43:10: MQTT publish: topic 'zigbee2mqtt/office/desk_light/bulb', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_mode":1,"linkquality":92,"state":"ON"}'
```

The one does seem good, and I am also reminded of the issue where groups don't debounce so the resulting mqtt messages for the group are kind of a mess too, which is not helping I think.

Here is me turning it off again

```
Zigbee2MQTT:debug 2020-09-01 19:45:29: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-01 19:45:29: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"OFF"}'
Zigbee2MQTT:debug 2020-09-01 19:45:29: Received Zigbee message from 'office/desk_light/remote', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 10112
Zigbee2MQTT:info  2020-09-01 19:45:30: MQTT publish: topic 'zigbee2mqtt/office/desk_light', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_temp":374,"state":"OFF"}'
Zigbee2MQTT:info  2020-09-01 19:45:30: MQTT publish: topic 'zigbee2mqtt/office/desk_light/bulb', payload '{"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_mode":1,"linkquality":52,"state":"OFF"}'
Zigbee2MQTT:info  2020-09-01 19:45:30: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"action":"off","battery":87,"linkquality":47,"update":{"state":"idle"}}'
```

Overall it seemed to have made things worse somehow? alteast before it was consistant with brightness being 0 after turning on a bulb until the first attReport that comes in a that was alteast debounce interval after the bulb turned on.

Or never for Hue bulbs due to the lack of reporting on those :(

sjorge, 2020-09-01:
Perhaps the storing of the state should be seperate from the mqtt publishing? where debounce is only applied to the later?

Koenkk, 2020-09-02:
> Perhaps the storing of the state should be seperate from the mqtt publishing? where debounce is only applied to the later?

Yes, that is what the change does.

Regarding the wrong brightness on off, try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/573a3a865260b22042f28b883c5bba9d53a55bcd/converters/fromZigbee.js#L704

```
            if (msg.data.hasOwnProperty('onOff')) {
                const property = postfixWithEndpointName('state', msg, model);
                const payload = {[property]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
                if (msg.data['onOff'] === 0) payload.brightness = 0;
                return payload;
            }
```

NOTE: This is just for testing, as plugs now also will return brightness 0 when toggled manually.

sjorge, 2020-09-02:
Yeah that does seem to work, but as you said it also adds brightness for other things now.

sjorge, 2020-09-02:
I was fiddling it some more (without the 2nd diff applied) and adding logging to receive.js

```
Zigbee2MQTT:debug 2020-09-02 17:50:46: Received Zigbee message from 'office/desk_light/bulb', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
{
  brightness: 254,
  color: { hue: 30, saturation: 78, x: 0.4608, y: 0.3982 },
  color_mode: 1,
  color_temp: 374,
  linkquality: 52,
  state: 'OFF'
}
current: {"brightness":254,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_mode":1,"color_temp":374,"linkquality":52,"state":"OFF"}
new: {"brightness":0,"color":{"hue":30,"saturation":78,"x":0.4608,"y":0.3982},"color_mode":1,"color_temp":374,"linkquality":92,"state":"OFF"}
```

Looks like the meta.state gets logged before we hit the receive.js, is that expected? I think so, but that still leaves the posibility of incorrect data in meta.state right?

Koenkk, 2020-09-04:
There is some discussion ongoing wether brightness 0 should be returned when the bulb is turned off (#2566), let's first see what comes out of that, because reverting this behaviour would also fix this issue.

sjorge, 2020-09-06:
Looks good now! Thanks.

closed 2020-09-06 by sjorge