Koenkk/zigbee2mqtt #4215: get_group_membership fails often

nobodypb, 2020-08-30:
# Bug Report
## What happened
When querying for group membership of multiple devices, most of them fail to answer, while the follwing error is printed:
```
Failed to call 'DeviceGroupMembership' 'onMQTTMessage' (TypeError: Cannot destructure property 'grouplist' of 'response' as it is undefined.
     at DeviceGroupMembership.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/legacy/deviceGroupMembership.js:32:14)
     at async Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:375:21))
```

I only query devices, that have their availability set to 'online' and also tried only querying devices that are connected to mains, with the same result.
It seems like it especially happens if I query all after another fast. If I wait for an state update and then issue the query for only that device, it seems to work.

## What did you expect to happen
Getting an answer with "group_list" set, or at least a log message indicating what went wrong.

## How to reproduce it (minimal and precise)
* Setup multiple devices
* (optional) Add them to groups
* Programmatically query each of them with `zigbee2mqtt/bridge/device/[friendly_name]/get_group_membership`

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: RaspBee I
Adapter firmware version: latest

Koenkk, 2020-08-30:
Looks Deconz specific, @ChrisHae could you look at this?

@nobodypb please already provide herdsman debug logging for easier investigation.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

nobodypb, 2020-09-01:
Yeah, sure.
I've uploaded the log here: https://gist.github.com/nobodypb/d466e58d976aabbd8395f58cf6cf6e61
Please note, that the former error is gone and now replaced by the warning from my commit 8373e49
`Couldn't get group membership of ${entity.device.ieeeAddr}`

ChrisHae, 2020-10-02:
Sry this issue slipped through my attention. I think I will improve the code of the deCONZ adapter to handle fast sending of multiple requestst better.

ChrisHae, 2020-10-09:
Can you retest this issue with the newest zigbee-herdsman commit ? I have slightly raised the time between packets so sending many commands at once could perform a little better. If not then some more complex changes are necessary. Would be nice if you could deliver the log output again.

closed 2020-11-16 by github-actions[bot]