Koenkk/zigbee2mqtt #4216: No state is being sent from OSRAM Smart+ Plug AB32570

smoes, 2020-08-30:
# Bug Report
## What happened

I successfully paired the OSRAM Smart+ Plug AB32570 with Zigbee2MQTT. It is recognized and added as expected. I can even send commands to the plug with Zigbee2MQTT, which are executed properly. However, no state change is being received by Zigbee2MQTT when pressing the hardware button on the plug, to toggle state. It's like there is no message at all. Even in debug mode, nothing shows up. I used 3 different units and all have the same faulty behavior. 

## What did you expect to happen

That toggling the state using the hardware-button results in a state change in Zigbee2MQTT

## How to reproduce it (minimal and precise)

Pair the plug. Press the hardware button on the Plug - no message/state appears in Zigbee2MQTT, not even in debug mode

## Debug Info
Zigbee2MQTT version: Head
Adapter hardware: CC26X2R1
Adapter firmware version:

Koenkk, 2020-08-30:
Can you try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

Please look at the log when doing this to see if it goes OK.

smoes, 2020-08-31:
Hey, bingo! I can see an error in the logs:

```
Aug 31 06:31:49 raspberrypi npm[15642]: Zigbee2MQTT:info  2020-08-31 06:31:49: Configuring '0x7cb03eaa0a0a4d3c'
Aug 31 06:31:59 raspberrypi npm[15642]: Zigbee2MQTT:error 2020-08-31 06:31:59: Failed to configure '0x7cb03eaa0a0a4d3c', attempt 1 (Error: Bind 0x7cb03eaa0a0a4d3c/3 genOnOff from '0x00124b001caa6cb8/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Aug 31 06:31:59 raspberrypi npm[15642]:     at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Aug 31 06:31:59 raspberrypi npm[15642]:     at ontimeout (timers.js:436:11)
Aug 31 06:31:59 raspberrypi npm[15642]:     at tryOnTimeout (timers.js:300:5)
Aug 31 06:31:59 raspberrypi npm[15642]:     at listOnTimeout (timers.js:263:5)
Aug 31 06:31:59 raspberrypi npm[15642]:     at Timer.processTimers (timers.js:223:10))
```

This is the device info zigbee2mqtt carries, btw:

```
{"ieeeAddr":"0x7cb03eaa0a0a4d3c","type":"Router","networkAddress":41672,"model":"AB3257001NJ","vendor":"OSRAM","description":"Smart+ plug","friendly_name":"0x7cb03eaa0a0a4d3c","manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","modelID":"Plug 01","hardwareVersion":1,"softwareBuildID":"V1.05.09","dateCode":"20140331DEOS****","lastSeen":1598189072541}
```

Koenkk, 2020-08-31:
Interesting, does controlling the plug from Zigbee2MQTT work?

smoes, 2020-09-01:
Yes, I just sanity checked. It works, I can switch it on and off using the `set`-topic

Koenkk, 2020-09-01:
Ok, when right after the turn on/off, can you try to configure it again?

smoes, 2020-09-01:
Didn't work - reconfiguring failed with the very same error, when executed right after a state change triggered by Zigbee2MQTT.

Koenkk, 2020-09-02:
Strange! I've ordered this plug recently and should receive it soon, so will check if I get the same issue.

SamNitro, 2020-09-10:
I have the same problem. 
I‘m on 1.14.4-dev.

SamNitro, 2020-09-18:
Is there anything new here?

Koenkk, 2020-09-19:
Still shipping

smoes, 2020-10-14:
Is there a way we can provide you with further details/logs, that could help to solve this issue without you having the device?

Koenkk, 2020-10-14:
Amazon lost my package, so I won't receive it anymore.

I think the only way to further debug this is to sniff the traffic when executing the configure: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

smoes, 2020-10-15:
Oh, sad to hear! 

I have a spare CC2531 lying around. If I find some time, I'll give it a try. Which process do we want to sniff? Physical button push?

Koenkk, 2020-10-15:
Sniff:
- Configure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure
- Physical button push

SamNitro, 2020-10-16:
Sorry for my english, it has been translated.
I tried to switch to the stable version, but I couldn't control anything. Then I restored a Backup and made an update. Now i get the state message from Hardware button.

smoes, 2020-10-20:
> Sorry for my english, it has been translated.
> I tried to switch to the stable version, but I couldn't control anything. Then I restored a Backup and made an update. Now i get the state message from Hardware button.

Which version are you on currently?

SamNitro, 2020-10-20:
I'm on 1.15.0-dev

Commits on Oct 17, 2020
  | info_commit | 9ac660bc |

closed 2020-11-27 by github-actions[bot]