Koenkk/zigbee2mqtt #4217: Best router?

gcormier, 2020-08-30:
I have a CC1352P running, but I have some devices that are just on the edge of signal unfortunately. I'm getting tired of intermittent connections. It mildly improved after moving my WiFi channel but not enough.

What would the best router be to repeat these signals be nowadays? The main concern is door locks (#799) but an Aqara door sensor would also be useful, and I know these don't always seem to work well with the IKEA Tradfi repeaters.

Is CC2531 the best option with the Z-Stack router firmware?

nanobra1n, 2020-08-30:
Look [here](https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623) for some information what routers work with Xiaomi/Aqara devices or are known not to. Personally I use some innr SP224 smart plugs that appear to work. 

I wouldn't expect too much from the CC2531 as a router, but it doesn't hurt to try.

Koenkk, 2020-08-30:
For DiY, https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html is good

xX-Nexus-Xx, 2020-09-02:
I'm using the cheap [IKEAK router](https://www.zigbee2mqtt.io/devices/E1746.html) with great success

gcormier, 2020-09-03:
I tried the tradfri again, it killed my network. For some reason the locks wanted to pair to it perhaps and it had a poor connection to my coordinator (1352P)

I'll try the DiY @Koenkk suggested with a CC2530 PA, external antenna.

gcormier, 2020-09-14:
I ordered one that looked like it had a PA built in but received this

[image]

I don't think this has a power amplifier?

I was expecting to receive this : 

[image]

nanobra1n, 2020-09-15:
Yep, looks like you just got got the CC2530. Complain to the vendor

gcormier, 2020-09-15:
Thx. Sent them a msg. I don't like posting the names or links, don't want to do the SEO for them...

closed 2020-10-23 by github-actions[bot]

gcormier, 2020-11-21:
Follow up, do NOT use

CC2530-UART-Wireless-Core-Development-Board-CC2530F256

From banggood. It does not have the power amplifier as pictured and they won't publish my 1 star review.