Koenkk/zigbee2mqtt #4222: Enabling more debugging info on why interview failed?

osresearch, 2020-08-31:
# Bug Report
## What happened
I'm trying to build a Zigbee device that is interoperable with `zigbee2mqtt` and have not been able to identify why the interview is failing.  For now it is copying the parameters from an existing Ikea lamp, but the interview stops after the first ZCL parameter read. Wireshark shows that both the real Ikea device and the custom firmware report the same :

```
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 4
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 4
    Command: Read Attributes Response (0x01)
    Status Record, String: FLOALT panel WS 30x30
        Attribute: Model Identifier (0x0005)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: FLOALT panel WS 30x30
    Status Record, String: IKEA of Sweden
        Attribute: Manufacturer Name (0x0004)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: IKEA of Sweden
    Status Record
        Attribute: Power Source (0x0007)
        Status: Success (0x00)
        Data Type: 8-Bit Enumeration (0x30)
        Power Source: Mains (single phase) (0x01)
```

On the real firmware the coordinator continues the interview by reading the ZCL version and other parameters, while with the custom firmware the coordinator stops as soon as it receives the attribute response, although it isn't clear why. Even with [`log_level: debug`](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) there is no additional debugging as to what the coordinator is expected:

```
info  2020-08-30T23:23:45: Starting interview of '0x086bd7fffedb6f8c'
info  2020-08-30T23:23:45: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x086bd7fffedb6f8c"}}'
debug 2020-08-30T23:23:47: Received Zigbee message from '0x086bd7fffedb6f8c', type 'readResponse', cluster 'genBasic', data '{"modelId":"FLOALT panel WS 30x30","manufacturerName":"IKEA of Sweden","powerSource":1}' from endpoint 1 with group ID 0
debug 2020-08-30T23:23:47: No converter available for 'L1527' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"FLOALT panel WS 30x30","manufacturerName":"IKEA of Sweden","powerSource":1}'
error 2020-08-30T23:23:57: Failed to interview '0x086bd7fffedb6f8c', device has  not successfully been paired
info  2020-08-30T23:23:57: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x086bd7fffedb6f8c"}}'
```

## What did you expect to happen
Something indicating what feature or parameter is missing that caused the interview to fail.  Is there a way to increase the information displayed when the new device is interviewed?

## How to reproduce it (minimal and precise)
[image]

Flash your Ikea FLOALT lamp with the firmware from [trmm.net/Ikea](https://trmm.net/Ikea) and run `from ZbPy.Device import dev ; dev.join()` at the Micropython prompt. 

## Debug Info
Zigbee2MQTT version: zigbee2mqtt version 1.7.1 (commit #b459c35)
Adapter hardware: CC2531
Adapter firmware version: ???

Koenkk, 2020-08-31:
More info can be found in the herdsman debug logging.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

osresearch, 2020-08-31:
Thanks, that does provide significantly more debugging info.  Still not clear on why it isn't sending the second set of ZCL queries, although it gives me a code path to instrument to see what is happening.

Koenkk, 2020-08-31:
Could you post the logging here?

closed 2020-10-09 by github-actions[bot]