Koenkk/zigbee2mqtt #4223: Osram 4x Switch - No PRESS action from bottom buttons

Jogobo, 2020-08-31:
# Bug Report
## What happened
Added an Osram 4x Switch. Only the top buttons send "PRESS" action. The bottom buttons only send "HOLD" and "RELEASE" actions on long press.

## What did you expect to happen
All 4 Buttons should send "PRESS" action.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2531, CC2530
Adapter firmware version:

Koenkk, 2020-09-01:
Could you share the `data/database.db` entry of thi sdevice?

Jogobo, 2020-09-01:
Sure, here you are...

`{"id":25,"type":"EndDevice","ieeeAddr":"0x000d6f000d5793b6","nwkAddr":64302,"manufId":4364,"manufName":"OSRAM","powerSource":"Battery","modelId":"Switch 4x EU-LIGHTIFY","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":
2064,"inClusterList":[0,1,32,4096,64768],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"Switch 4x EU-LIGHTIFY","manufacturerName":"OSRAM","powerSource":3,"zclVersion":1,"appVersion":1,"stackVersion
":0,"hwVersion":1,"dateCode":"201703130813180","swBuildId":"e.1.0.0M"}}},"binds":"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1}"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1},"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1}]},"2":"profId":260,"epId":2,"devId":2064,"inClusterList":0,4096,64768],"outClusterList":3,4,5,6,8,768,4096],"clusters":"genBasic":"attributes":"stackVersion":0}}},"binds":"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1},"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1}]},"3":"profId":260,"epId":3,"devId":2064,"inClusterList":0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1},"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1}]},"4":{"profId":260,"epId":4,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1}]},"5":{"profId":260,"epId":5,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"genBasic":"attributes":{"stackVersion":0}}},"binds":[]},"6":{"profId":260,"epId":6,"devId":2064,"inClusterList":0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[]}},"appVersion":1,"stackVersion":0,"hwVersion":1,"dateCode":"201703130813180","s
wBuildId":"e.1.0.0M","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1598867264415}`

Jogobo, 2020-09-02:
After changing devices.js I get the following errors in the log:

`Sep  2 11:59:04 jogolinux npm[11079]: ^[[31mZigbee2MQTT:error^[[39m 2020-09-02 1
1:59:04: Failed to configure '0x000d6f000d5793b6', attempt 6 (Error: Bind 0x000d
6f000d5793b6/1 genLevelCtrl from '0x00124b0018e1f479/1' failed (AREQ - ZDO - bin
dRsp after 10000ms)
Sep  2 11:59:04 jogolinux npm[11079]:     at Timeout.waiter.timer.setTimeout [as
 _onTimeout] (/usr/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/wai
tress.js:46:35)
Sep  2 11:59:04 jogolinux npm[11079]:     at ontimeout (timers.js:436:11)
Sep  2 11:59:04 jogolinux npm[11079]:     at tryOnTimeout (timers.js:300:5)
Sep  2 11:59:04 jogolinux npm[11079]:     at listOnTimeout (timers.js:263:5)
Sep  2 11:59:04 jogolinux npm[11079]:     at Timer.processTimers (timers.js:223:
10))`

Those are the actions that I get from the switch so far:

`{"linkquality":105,"battery":100,"voltage":3000,"action":"left_top_click"}
{"linkquality":102,"battery":100,"voltage":3000,"action":"right_top_click"}
{"linkquality":102,"battery":100,"voltage":3000,"action":"left_top_hold"}
{"linkquality":102,"battery":100,"voltage":3000,"action":"left_top_release"}
{"linkquality":102,"battery":100,"voltage":3000,"action":"right_top_hold"}
{"linkquality":105,"battery":100,"voltage":3000,"action":"right_top_hold"}
{"linkquality":105,"battery":100,"voltage":3000,"action":"right_top_release"}
{"linkquality":105,"battery":100,"voltage":3000,"action":"left_bottom_hold"}
{"linkquality":105,"battery":100,"voltage":3000,"action":"left_bottom_release"}
`
"right_top_hold" is being sent twice every time.

The database.db entry looks like this:
`{"id":25,"type":"EndDevice","ieeeAddr":"0x000d6f000d5793b6","nwkAddr":64302,"man
ufId":4364,"manufName":"OSRAM","powerSource":"Battery","modelId":"Switch 4x EU-L
IGHTIFY","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":
2064,"inClusterList":[0,1,32,4096,64768],"outClusterList":[3,4,5,6,8,25,768,4096
],"clusters":{"genBasic":{"attributes":{"modelId":"Switch 4x EU-LIGHTIFY","manuf
acturerName":"OSRAM","powerSource":3,"zclVersion":1,"appVersion":1,"stackVersion
":0,"hwVersion":1,"dateCode":"201703130813180","swBuildId":"e.1.0.0M"}}},"binds"
:[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpo
intID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479
","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00
18e1f479","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"
0x00124b0018e1f479","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":2064,"i
nClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"
genBasic":{"attributes":{"stackVersion":0}}},"binds":[{"cluster":8,"type":"endpo
int","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1},{"cluster":768,"ty
pe":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1}]},"3":{"
profId":260,"epId":3,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList
":[3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}}
,"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479
","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00
18e1f479","endpointID":1}]},"4":{"profId":260,"epId":4,"devId":2064,"inClusterLi
st":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"genBasic":
{"attributes":{"stackVersion":0}}},"binds":[{"cluster":8,"type":"endpoint","deviint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1},{"cluster":768,"ty
pe":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479","endpointID":1}]},"3":{"
profId":260,"epId":3,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList
":[3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}}
,"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f479
","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00
18e1f479","endpointID":1}]},"4":{"profId":260,"epId":4,"devId":2064,"inClusterLi
st":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"genBasic":
{"attributes":{"stackVersion":0}}},"binds":[{"cluster":8,"type":"endpoint","devi
ceIeeeAddress":"0x00124b0018e1f479","endpointID":1}]},"5":{"profId":260,"epId":5
,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,409
6],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[]},"6":{"p
rofId":260,"epId":6,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList"
:[3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},
"binds":[]}},"appVersion":1,"stackVersion":0,"hwVersion":1,"dateCode":"201703130
813180","swBuildId":"e.1.0.0M","zclVersion":1,"interviewCompleted":true,"meta":{
},"lastSeen":1599040494925}
`

Koenkk, 2020-09-02:
I assume you woke up the device by clicking the device right before executing the command?

@hgiesenow  since you added support for it, could you help here? Is there any reason why these were not added?

Jogobo, 2020-10-03:
Any chance that there will be an update?

closed 2020-11-12 by github-actions[bot]