Koenkk/zigbee2mqtt #4224: Support for YSR-MINI-01 with different ID

tbnobody, 2020-08-31:
I have a device which is labled as a YSR-MINI-01 but identifies itself using the id `FB56-ZCW20FB1.2`

I got it to work by adding it to the same entry is the original YSR-MINI-01
```
    // YSRSAI
    {
        zigbeeModel: ['ZB-CL01', 'FB56-ZCW20FB1.2'],
        model: 'YSR-MINI-01',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (RGB+CCT)',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
```

By doing to I am able to set the color, brightness etc. question is, should this id being added to the original entry or is it better to create a new entry? 

If I set the color to e.g. 
```
{
  "state": "ON",
  "color": {
    "r": 0,
    "g": 60,
    "b": 0
  }
}
```
the green channel is 100% on. (--> It seems to ignore the 60 in the g parameter). But if I set the color using:
```
{
  "state": "ON",
  "brightness": 50,
  "color": {
    "r": 0,
    "g": 60,
    "b": 0
  }
}
```
the green color is dimmed correctly. Is this by intention or is it a issue?

Koenkk, 2020-08-31:
Can you set the brightness first and the color after that (as separate messages)? Also try this the other way around.

tbnobody, 2020-08-31:
If I set the color first (e.g. R: 0, G: 0, B: 20)  and then the brightness (e.g. 20) (two messages) it works perfectly. --> Light is dimmed

But if I then set the color to R: 0, G: 0, B: 255 the color stays dimmed. I have to change the brightness value back to 255 to get the full brightness

Koenkk, 2020-08-31:
that's expected, rgb won't influence the brightness.

tbnobody, 2020-08-31:
ah ok... that explains a lot effects I see here. regarding the additional zigbee id, should I just create a pull request or can you change it directly?

Koenkk, 2020-09-01:
please make a pr, assuming this can be closed now.

closed 2020-09-01 by Koenkk