Koenkk/zigbee2mqtt #4225: How to avoid Zigbee2MQTT to send some cached endpoint states when rebooting?

sebi5361, 2020-08-31:
I am using Zigbee2MQTT together with HomeAssistant.
I have a Xiaomi contact sensor that triggers a notification every time `"contact":false`.
My issue is that if I restart Zigbee2MQTT `"contact":false` is resent to the broker triggering a false notification.
The solution I have found not to trigger a false notification is to set Zigbee2MQTT setting `cache_state: false`.
However I get this Zigbee2MQTT warning:
_In order for HomeAssistant integration to work properly set `cache_state: true_

Thus two questions:
- how to avoid Zigbee2MQTT to send some cached endpoint states when rebooting?
- why is it important to have `cache_state: true` in order for HomeAssistant integration to work properly?

Koenkk, 2020-09-04:
> how to avoid Z2M to send some cached endpoint states when rebooting?
Use [cache_state_send_on_startup](https://www.zigbee2mqtt.io/information/configuration.html), but this won't solve your problem as when e.g. a battery report is send `contact: false` will be send again (if `cache_state: true`).

> why is it important to have cache_state: true in order for HA integration to work properly?

Because Home Assistant requires all values in the messages, you will get warnings/error in Home Assistant if you set this `false`.

What I actually advise to do is to fix your automations. You shouldn't execute them when `contact: false` is send but when the state changes from `contact: true` -> `contact: false`.

sebi5361, 2020-09-04:
> What I actually advise to do is to fix your automations. You shouldn't execute them when `contact: false` is send but when the state changes from `contact: true` -> `contact: false`.

Indeed this is what I have done...
However as `contact: false` when rebooting Z2M and as the last state is `contact:true`, a state change occurs still...
- Why the last known state isn't the cached state?
- Is there a way to force the cached state to `contact:true`?

Koenkk, 2020-09-05:
When contact is true, and you stop Zigbee2MQTT, what do you see in `data/state.json` (this is where the state is persisted), this is the state that will be resend on startup.

sebi5361, 2020-09-05:
I see `..."0x00158d000544b19e": {"battery": 100,"contact": true,"linkquality": 0,"voltage": 3065},...` in `data/state.json`
...and indeed it persists and is resent on startup!

The mistake came from me ...sorry for this. Thanks for your expalanations and your excellent work!!

closed 2020-09-05 by sebi5361

tomasbedrich, 2023-06-18:
Hello, what is the guidance to resolve the same issue when HA integration is required?