Koenkk/zigbee2mqtt #4228: Aqara roller_shutter always reports running: "true"

rengaboy, 2020-08-31:
# Bug Report
## What happened
I have some Aqara curtains (ZNCLDJ11LM) and roller shutters (ZNGZDJ11LM). Both curtains and shutters are reported by zigbee2mqtt as "Aqara curtain motor (ZNCLDJ11LM)", both are working, but shutter reports "running" attribute as always true, even when they are idle.
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2538
Adapter firmware version: zStack30x

Koenkk, 2020-09-01:
Could you post the debug logging when you open the curtain, it stopped and close it again?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

rengaboy, 2020-09-01:
With pleasure.
closing:
`debug 2020-09-01 19:47:57: Received MQTT message on 'homeassistant/shutter_hall/set' with data 'CLOSE'
debug 2020-09-01 19:47:57: Publishing 'set' 'state' to 'shutter_hall'
debug 2020-09-01 19:47:57: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440516,"presentValue":100}' from endpoint 1 with groupID 0
info  2020-09-01 19:47:57: MQTT publish: topic 'homeassistant/shutter_hall', payload '{"options":{"reverse_direction":true,"hand_open":false},"linkquality":78,"position":100,"running":true}'
debug 2020-09-01 19:47:58: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":2}' from endpoint 1 with groupID 0`
closed:
`debug 2020-09-01 19:48:36: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":0}' from endpoint 1 with groupID 0
debug 2020-09-01 19:48:36: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":63140608,"presentValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 19:48:36: MQTT publish: topic 'homeassistant/shutter_hall', payload '{"options":{"reverse_direction":true,"hand_open":false},"linkquality":84,"position":0,"running":true}'`
opening:
`debug 2020-09-01 19:48:36: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":0}' from endpoint 1 with groupID 0
debug 2020-09-01 19:48:36: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":63140608,"presentValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 19:48:36: MQTT publish: topic 'homeassistant/shutter_hall', payload '{"options":{"reverse_direction":true,"hand_open":false},"linkquality":84,"position":0,"running":true}'`
opened:
`debug 2020-09-01 19:50:33: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":0}' from endpoint 1 with groupID 0
debug 2020-09-01 19:50:33: Received Zigbee message from 'shutter_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":63140608,"presentValue":100}' from endpoint 1 with groupID 0
info  2020-09-01 19:50:33: MQTT publish: topic 'homeassistant/shutter_hall', payload '{"options":{"reverse_direction":true,"hand_open":false},"linkquality":84,"position":100,"running":true}'`

Koenkk, 2020-09-01:
@yozik04  @Evildime is the running attribute working for you?

rengaboy, 2020-09-01:
maybe this could help?
this is log from curtains, that reports status well.
closing
`debug 2020-09-01 23:56:10: Received MQTT message on 'homeassistant/curtain_hall/set' with data 'CLOSE'
debug 2020-09-01 23:56:10: Publishing 'set' 'state' to 'curtain_hall'
debug 2020-09-01 23:56:10: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440516,"presentValue":100}' from endpoint 1 with groupID 0
info  2020-09-01 23:56:10: MQTT publish: topic 'homeassistant/curtain_hall', payload '{"position":100,"running":true,"linkquality":42}'
debug 2020-09-01 23:56:10: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":2}' from endpoint 1 with groupID 0`
closed
`debug 2020-09-01 23:56:17: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":0}' from endpoint 1 with groupID 0
debug 2020-09-01 23:56:17: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 23:56:17: MQTT publish: topic 'homeassistant/curtain_hall', payload '{"position":0,"running":false,"linkquality":47}'`
opening
`debug 2020-09-01 23:55:31: Received MQTT message on 'homeassistant/curtain_hall/set' with data 'OPEN'
debug 2020-09-01 23:55:31: Publishing 'set' 'state' to 'curtain_hall'
debug 2020-09-01 23:55:31: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440515,"presentValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 23:55:31: MQTT publish: topic 'homeassistant/curtain_hall', payload '{"position":0,"running":true,"linkquality":44}'
debug 2020-09-01 23:55:31: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":1}' from endpoint 1 with groupID 0`
opened
`debug 2020-09-01 23:55:39: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genBasic', data '{"1028":0}' from endpoint 1 with groupID 0
debug 2020-09-01 23:55:39: Received Zigbee message from 'curtain_hall', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":100}' from endpoint 1 with groupID 0
info  2020-09-01 23:55:39: MQTT publish: topic 'homeassistant/curtain_hall', payload '{"position":100,"running":false,"linkquality":42}'`

rengaboy, 2020-09-01:
One more strange thing - ZNCLDJ11LM (curtains) have "hand_open" option, but dont report it, and ZNGZDJ11LM (shutter), have no "hand_open" option, because they are roller shutters, but they do report that option.

yozik04, 2020-09-02:
@Koenkk @rengaboy 
Running attribute works fine for me:
[image]
Screenshot from MQTT Explorer

rengaboy, 2020-09-02:
@yozik04 
Do you have roller shutters or curtains?

yozik04, 2020-09-02:
Curtains. Aqara curtain motors (ZNCLDJ11LM) by Xiaomi

rengaboy, 2020-09-02:
@yozik04 Curtains works fine for me too. Shutters don’t.

Koenkk, 2020-09-02:
I think the roller shutter doesn't support `running`. The running is true when `61440` == `0`, however the roller shutter never reports 0 and when closing/closed the value stays `63140608`.

rengaboy, 2020-09-02:
I’ll try to add shutters to Xiaomi hub tomorrow and sniff traffic. Maybe it’s some kind of old firmware issue.

rengaboy, 2020-09-03:
Xiaomi MiHome app reports running action fine, so i see in the app, that curtain is closing or opening. Also i found one more problem with shutter status updates, when i fire toggle action to shutter, it starts moving, then i fire action one more time, shutter updates attributes in topics "current_position" and "position" one time to 100>98, when it was opened or 0>2, when it was closed, but when i fire action third, fourth, etc. times, topics stays on previous states.

Koenkk, 2020-09-04:
@rengaboy I currently don't understand how the Xiaomi gateway determines that, as you can see in https://github.com/Koenkk/zigbee2mqtt/issues/4228#issuecomment-684996858 the received messages for open/opening are identical expect the position. Maybe there is some hidden traffic which we currently don't parse. Are you able to sniff Zigbee traffic from the Xiaomi gateway using: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

rengaboy, 2020-09-23:
@Koenkk I made sniffing and got some traffic, how can i share saved packets with you?

Koenkk, 2020-09-25:
On telegram (@koenkk)

closed 2020-11-03 by github-actions[bot]