Koenkk/zigbee2mqtt #4230: Danalock state not updating automatically

Rurouni, 2020-08-31:
Hi, 
I have Danalock V3 integrated into HA via MQTT discovery.
Everything works but the state update is not happening automatically, I have to manually publish empty payload to `zigbee2mqtt/<friendlyLockName>/get` and then it immediately updates in HA to correct state. 
I don't know what side is responsible for publishing update into `state_topic` is it zigbee2mqtt side or HA should poll... reading docs on "MQTT lock" in HA it feels like they just expect the message to be retained in state topic and updated on every change...

I've tried adding :
```
    retain: true
    retrieve_state: true
```
to device in configuration.yaml and also tried adding:
```
advanced:
  report: true
```
but neither had any effect and I am now stuck and need pointers for where to start looking next

### Debug Info
Zigbee2Mqtt: 1.14.0
Adapter hardware: CC2531
Adapter firmware version: zStack12 (20190608)

Thank you!

Koenkk, 2020-09-01:
Can you try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , while doing so look at the log if it went OK.

Rurouni, 2020-09-02:
Yes, that did the trick, thank you!

closed 2020-09-02 by Rurouni

VincentSC, 2020-11-03:
This worked for me too. My guess is the that this was caused by the lock not being calibrated yet when I first connected. When I calibrated, I had to switch the rotation-direction for locking the door.

Hooshee, 2021-01-03:
> 
> 
> Yes, that did the trick, thank you!

Coulc you please tell me what exactly you have done/post your relevant current devices topic content? I have the same issue, but as a relatively newby to the z2m, I'm a bit stuck... thank you :)

Koenkk, 2021-03-09:
It's https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure now

philibertc, 2021-03-10:
Hello @Koenkk,
Thank you for your support. It worked for an outlet with an integrated button on it that didn't report its status.
But for a tradfri light when I click on "Reconfigure" the lqi of the lamp doesn't change and nothing happens.

frogale, 2021-03-27:
@Koenkk I have this problem wit non-reporting lock state very often (every 1-2 weeks). Every time I remove my Danalock and pair it again, it is ok for some time. But this reconfigure feature also works and it is much faster for me. Is it a problem with Z2M or Danalock itself? I have the latest firmware and the latest version of Z2M. Maybe I could perform preventive reconfigures every few hours or so, but that is not the solution.

Koenkk, 2021-03-27:
@frogale it's a guess, but I think the problem is at the Danalock side, the device shouldn't lose it configuration.

VincentSC, 2021-03-27:
Also think it's the Danalock, based on their bug discussions and my own experience. Their code seems to be a mess and not thought through. For instance battery levels are incorrect and their app has more spelling-errors than the website of AliExpress.

It's a shame, as I was doubting between Nuki and Danalock and Z2M's support was the reason I chose for the latter.

arcanatigris, 2021-06-10:
It seems this occurs when the route changes, like through a repeater or the like.

InToSSH, 2021-08-01:
> It seems this occurs when the route changes, like through a repeater or the like.

Yes, I had an IKEA repeater and when the lock switched to it, it stopped responding. But even after disconnecting the repeater and lock sitting idle for a while, it still lock/unlocks but doesn't report it's state.