Koenkk/zigbee2mqtt #4231: zigbee2mqtt fails to start when network_key is set to GENERATE (edge case)

elfhack, 2020-08-31:
I'll forgo the usual bug template as this is a more complicated situation.

On my home server i now have 2 zigbee adapters, a generic cc2531 (with udev rules renaming it to /dev/zigbee) and a new and shiny zzh (under /dev/zzh). I'm now trying to run 2 instances of zigbee2mqtt, with the zzh instance being completely separate from the 2531 one. I run both of them via kubernetes, as statefullsets (so no user errors with sharing data have a chance to occur). They both also have their own topics on mqtt, so no chance of collision here.

The 2531 now has around 40 devices under itself and is running smoothly.
The zzh has nothing, and I'm not trying to migrate any data from the 2531 instance. Completely fresh configs, different pan_id et al.

Now, if i run zigbee2mqtt for zzh, while the zigbee2mqtt for 2531 instance is running, with network_key set to GENERATE
```
advanced:
  network_key: GENERATE
```

it will crash like so

```
> zigbee2mqtt@1.14.3 start /app
> node index.js

Zigbee2MQTT:info  2020-08-31 20:37:59: Logging to console only'
Zigbee2MQTT:debug 2020-08-31 20:37:59: Can't load state from file /data/state.json (doesn't exist)
Zigbee2MQTT:info  2020-08-31 20:37:59: Starting Zigbee2MQTT version 1.14.3 (commit #f8066e8)
Zigbee2MQTT:info  2020-08-31 20:37:59: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-08-31 20:37:59: Using zigbee-herdsman with settings: '{"network":{"panID":8149,"extendedPanID":[253,173,170,153,17,35,68,245],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/data/database.db","databaseBackupPath":"/data/database.db.backup","backupPath":"/data/coordinator_backup.json","serialPort":{"path":"/dev/zzh"},"adapter":{"concurrent":null}}'
Zigbee2MQTT:error 2020-08-31 20:39:09: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-31 20:39:09: Failed to start zigbee
Zigbee2MQTT:error 2020-08-31 20:39:09: Exiting...
Zigbee2MQTT:error 2020-08-31 20:39:09: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:171:23
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-31T18_39_09_710Z-debug.log
```

However, if I either stop the 2531 instance or configure the network key statically like so (the key here doesn't matter, any will do)

```
advanced:
  network_key: [141, 242, 220, 79, 156, 22, 117, 55, 201, 141, 159, 24, 130, 173, 250, 27]
```

Then there's no problem at all. 

I'm guessing you have some sort of scan there that detects a second zigbee network on the same channel and this makes it fail? Not sure about it, since even if i set the channel to let's say 15 (the 2531 network is on channel 11) it would fail as well.

And to reiterate : both instances have no contact with each other - separate containers, separate config files, separate devices.

Koenkk, 2020-09-01:
Can you provide the herdsman debug logging of this issue?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

elfhack, 2020-09-11:
Sorry for the lag
```
Using '/data' as data directory
Creating configuration file...

> zigbee2mqtt@1.14.4 start /app
> node index.js

Zigbee2MQTT:info  2020-09-11 14:42:03: Logging to console only'
Zigbee2MQTT:debug 2020-09-11 14:42:04: Can't load state from file /data/state.json (doesn't exist)
Zigbee2MQTT:info  2020-09-11 14:42:04: Starting Zigbee2MQTT version 1.14.4 (commit #654817a)
Zigbee2MQTT:info  2020-09-11 14:42:04: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-09-11 14:42:04: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/data/coordinator_backup.json","databaseBackupPath":"/data/database.db.backup","databasePath":"/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":8149},"serialPort":{"path":"/dev/zzh"}}'
2020-09-11T12:42:04.222Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[73,220,141,253,23,7,1,132,14,188,159,168,123,155,223,154],"panID":8149,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/zzh"},"databasePath":"/data/database.db","databaseBackupPath":"/data/database.db.backup","backupPath":"/data/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-09-11T12:42:04.223Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/zzh and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-09-11T12:42:04.225Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-09-11T12:42:04.678Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-09-11T12:42:04.679Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-09-11T12:42:05.683Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-09-11T12:42:05.685Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-09-11T12:42:05.691Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2020-09-11T12:42:05.692Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2020-09-11T12:42:05.692Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2020-09-11T12:42:05.693Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2020-09-11T12:42:05.695Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-11T12:42:05.695Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2020-09-11T12:42:05.696Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2020-09-11T12:42:05.702Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,1,2,7,1,126,61,52,1,0,255,255,255,255,28]
2020-09-11T12:42:05.702Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7,1,126,61,52,1,0,255,255,255,255,28]
2020-09-11T12:42:05.702Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,1,2,7,1,126,61,52,1,0,255,255,255,255] - 28
2020-09-11T12:42:05.703Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200830}
2020-09-11T12:42:05.703Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-11T12:42:05.703Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16
2020-09-11T12:42:05.703Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200830})
2020-09-11T12:42:05.705Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-09-11T12:42:05.706Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74]
2020-09-11T12:42:05.720Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62]
2020-09-11T12:42:05.720Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62]
2020-09-11T12:42:05.720Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62
2020-09-11T12:42:05.720Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-09-11T12:42:05.721Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-11T12:42:05.721Z zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid
2020-09-11T12:42:05.721Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-09-11T12:42:05.721Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74]
2020-09-11T12:42:05.731Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62]
2020-09-11T12:42:05.731Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62]
2020-09-11T12:42:05.731Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62
2020-09-11T12:42:05.731Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-09-11T12:42:05.731Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-09-11T12:42:05.731Z zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid
2020-09-11T12:42:05.732Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}}
[... cut ...]
homeassistant: true
        permit_join: false
        mqtt:
          base_topic: zzh
          server: 'mqtt://mosquitto.homeassistant'
        serial:
          port: /dev/zzh
          disable_led: false
        advanced:
          # network_key: [141, 242, 220, 79, 156, 22, 117, 55, 201, 141, 159, 24, 130, 173, 250, 27]
          network_key: GENERATE
          pan_id: 0x1fd5
          log_level: debug
          log_output:
            - console
          availability_timeout: 300
          report: false
```

Koenkk, 2020-09-11:
Are you able to sniff the traffic on channel 11 during this startup? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

elfhack, 2020-09-11:
> Are you able to sniff the traffic on channel 11 during this startup? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Unfortunately no. That would require a 3rd zigbee adapter, and I just don't have that many :(

closed 2020-10-19 by github-actions[bot]