Koenkk/zigbee2mqtt #4234: Windows build fails because of unix-dgram

blackfyre, 2020-09-01:
# Bug Report
## What happened
Fresh checkout from the repo on windows, per documentation. After `npm ci` the build fails because of the `unix-dgram` dependency is looking for a file that's not present on windows.

## What did you expect to happen
Start the Zigbee2MQTT app

## How to reproduce it (minimal and precise)

## Debug Info
```bash
galiczm@INV18587 MINGW64 /c/projects/zigbee2mqtt (master)
$ npm ci
npm WARN prepare removing existing node_modules/ before installation

> @serialport/bindings@9.0.1 install C:\projects\zigbee2mqtt\node_modules\zigbee-herdsman\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> serialport@9.0.1 postinstall C:\projects\zigbee2mqtt\node_modules\zigbee-herdsman\node_modules\serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

> unix-dgram@2.0.3 install C:\projects\zigbee2mqtt\node_modules\unix-dgram
> node-gyp rebuild

C:\projects\zigbee2mqtt\node_modules\unix-dgram>if not defined npm_config_node_gyp (node "C:\Users\galiczm\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-g
yp\bin\node-gyp.js" rebuild )  else (node "C:\Users\galiczm\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  unix_dgram.cc
  win_delay_load_hook.cc
c:\projects\zigbee2mqtt\node_modules\unix-dgram\src\unix_dgram.cc(9): fatal error C1083: Cannot open include file: 'unistd.h': No such file or directory [C:\projects\zigbee2mqtt\node_modules\unix-dgram\build\u
nix_dgram.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\galiczm\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Windows_NT 10.0.17763
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\galiczm\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebu
ild"
gyp ERR! cwd C:\projects\zigbee2mqtt\node_modules\unix-dgram
gyp ERR! node -v v12.13.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
added 770 packages in 61.965s

```

Koenkk, 2020-09-01:
I see you are running it via MINGW64, can you try running `npm install` in the Zigbee2MQTT directory with the default command prompt?

blackfyre, 2020-09-01:
The problem is with both with Git Bash & Powershell. Below is the essentially same output from powershell on another system.

The problem is caused by the `unix-dgram` dependency.

```
PS D:\projects\zigbee2mqtt> npm install

> unix-dgram@2.0.3 install D:\projects\zigbee2mqtt\node_modules\unix-dgram
> node-gyp rebuild

D:\projects\zigbee2mqtt\node_modules\unix-dgram>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  unix_dgram.cc
  win_delay_load_hook.cc
d:\projects\zigbee2mqtt\node_modules\unix-dgram\src\unix_dgram.cc(9): fatal error C1083: Cannot open include file: 'uni
std.h': No such file or directory [D:\projects\zigbee2mqtt\node_modules\unix-dgram\build\unix_dgram.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd D:\projects\zigbee2mqtt\node_modules\unix-dgram
gyp ERR! node -v v12.18.2
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN tsutils@3.17.1 requires a peer of typescript@>=2.8.0 || >= 3.2.0-dev || >= 3.3.0-dev || >= 3.4.0-dev || >= 3.5.0-dev || >= 3.6.0-dev || >= 3.6.0-beta || >= 3.7.0-dev || >= 3.7.0-beta but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules\zigbee-herdsman\node_modules\jest-haste-map\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules\zigbee-herdsman\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules\zigbee-herdsman-converters\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: unix-dgram@2.0.3 (node_modules\unix-dgram):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: unix-dgram@2.0.3 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

added 1583 packages from 713 contributors and audited 775 packages in 40.156s

38 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```

Koenkk, 2020-09-02:
I've just check on a windows system and I get the same error. However it seems that `npm start` starts Zigbee2MQTT as expected, so not sure if this is really a problem.

yurycooliq, 2020-09-14:
`npm install --global windows-build-tools`

Or, if you are using Yarn:

`yarn global add windows-build-tools`

closed 2020-10-22 by github-actions[bot]

afrostewie, 2020-10-28:
I stumbled on this, and have the same problem. blackfyre were you able to solve it?

blackfyre, 2020-10-29:
@afrostewie It's a dependency issue... it should go away once the dependency is updated.
And yes, it doesn't imply that everything is in working order on windows...

KiLLeRRaT, 2020-11-16:
I'm stuck with this too.

blackfyre, 2020-11-19:
> I'm stuck with this too.

don't worry, It'll work just fine

tjarb, 2021-08-25:
Hi, well I'm also stuck on windows.
I extracted the ZIGBEE2MQTT folder under c:\program files\zigbee2mqtt-master
For the Conbee2 stick I've changed the device port to \\.\COM3 in the data\configuration.yaml file

I've got running node-js 14 (had node-js10 but gave same results)

Added the windows build tools etc

the npm ci generates errors on the 
`
> c:\Program Files\zigbee2mqtt-master>npm ci
>
> @serialport/bindings@9.2.0 install c:\Program Files\zigbee2mqtt-master\node_modules\zigbee-herdsman-converters\node_modules\zigbee-herdsman\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
> @serialport/bindings@9.2.0 install c:\Program Files\zigbee2mqtt-master\node_modules\zigbee-herdsman\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
> unix-dgram@2.0.3 install c:\Program Files\zigbee2mqtt-master\node_modules\unix-dgram
> node-gyp rebuild
> c:\Program Files\zigbee2mqtt-master\node_modules\unix-dgram>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
>  unix_dgram.cc
> win_delay_load_hook.cc
> c:\program files\zigbee2mqtt-master\node_modules\unix-dgram\src\unix_dgram.cc(9): fatal error C1083: Cannot open includ
e file: "unistd.h": No such file or directory [c:\Program Files\zigbee2mqtt-master\node_modules\unix-dgram\build\unix_d
gram.vcxproj]
> gyp ERR! build error
> gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
> gyp ERR! stack     at ChildProcess.emit (events.js:375:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Windows_NT 10.0.17763
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd c:\Program Files\zigbee2mqtt-master\node_modules\unix-dgram
gyp ERR! node -v v14.17.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
added 1124 packages in 20.963s
`

npm start turns in an error and exit of the service.
`
> c:\Program Files\zigbee2mqtt-master>npm start
> zigbee2mqtt@1.21.0 start c:\Program Files\zigbee2mqtt-master
> node index.js
>
Zigbee2MQTT:info  2021-08-25 16:22:23: Logging to console and directory: "c:\Program Files\zigbee2mqtt-master\data\log\2021-08-25.16-22-20" filename: log.txt
Zigbee2MQTT:info  2021-08-25 16:22:23: Starting Zigbee2MQTT version 1.21.0 (commit #unknown)
Zigbee2MQTT:info  2021-08-25 16:22:23: Starting zigbee-herdsman (0.13.124)
Zigbee2MQTT:error 2021-08-25 16:22:43: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-08-25 16:22:43: Failed to start zigbee
Zigbee2MQTT:error 2021-08-25 16:22:43: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-08-25 16:22:43: Exiting...
Zigbee2MQTT:error 2021-08-25 16:22:43: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (c:\Program Files\zigbee2mqtt-master\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:102:27)
    at Controller.start (c:\Program Files\zigbee2mqtt-master\node_modules\zigbee-herdsman\src\controller\controller.ts:123:29)
    at Zigbee.start (c:\Program Files\zigbee2mqtt-master\lib\zigbee.js:67:27)
    at Controller.start (c:\Program Files\zigbee2mqtt-master\lib\controller.js:106:27)
    at start (c:\Program Files\zigbee2mqtt-master\index.js:94:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.21.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.21.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Administrator.HYPERV01.003\AppData\Roaming\npm-cache\_logs\2021-08-25T14_22_43_704Z-debug.log

c:\Program Files\zigbee2mqtt-master>`

I see a lot of issues corresponding this error. Is there a known solution?