Koenkk/zigbee2mqtt #4235: Switching from LAUNCHXL-CC1352P-2 to ZZA (CC2652 Stick)

Diddlik, 2020-09-01:
I want to switch from CC1352P-2 to ZZA zig-a-zig-ah! CC2652, is it enough to change only the panID or do I have to repair all devices?

Zigbee2mqqt: 0.14.3

Koenkk, 2020-09-01:
I've never tested it, but repairing shouldn't be necessary (just plug, change port and start)

copystring, 2020-09-01:
I have done this with the LAUNCHXL-CC26X2R1. No problem.

Diddlik, 2020-09-02:
I tried it. Port changed to 
serial:
  port: /dev/ttyUSB0 

But now I get this error:
**Zigbee2MQTT:error 2020-09-02 10:11:07: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-09-02 10:11:07: Failed to start zigbee
Zigbee2MQTT:error 2020-09-02 10:11:07: Exiting...
Zigbee2MQTT:error 2020-09-02 10:11:07: Error: Error while opening serialport 'Er                                   ror: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapt                                   er/z-stack/znp/znp.js:136:32)**

Koenkk, 2020-09-02:
@Diddlik can you provide the herdsman debug logging of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

closed 2020-10-11 by github-actions[bot]