Koenkk/zigbee2mqtt #4237: Brightness of dimmer set to 0% when switching to OFF with Node Red or when device configured in HomeBridge

georgeavd, 2020-09-01:
# Bug Report
Thus issue is related to #3782.

## What happened
After the upgrade of Zigbee2MQTT to version 1.14.3 when switching the dimmer to OFF the brightness is also set to 0%.
When switching to ON the dimmer is set to ON but with a brightness level of 0%.

In the previous firmware 1.13.4 (when I'm not mistaken) the brightness level was not touched and the dimmer (ROBB_200-004-0) was switched ON to the last brightness level. I performed the upgrade to 1.14.3 to be able to detect ROBB_200-007-0 (8 button wall switch)

Home Bridge is also configured. When I remove the switch from the Home Bridge configuration and and only use the 8-button wall switch the previous brightness level is kept.
When the dimmer is switch OFF via NodeRed also the brightness is set to 0%.
Also after adding the switch to the Home Bridge configuration the brightness level is set to 0% when switching OFF with the 8-button wall switch.

## What did you expect to happen
I expect the brightness level to be left untouched and not set to 0% when switching the device to OFF.

## How to reproduce it (minimal and precise)
The brightness is set to 0% when the dimmer:
- is switched OFF via Node Red or Home Bridge;
- is included in a Home Bridge configuration but only the 8-button wall switch is used to control the device.

When the device is not included in a Home Bridge configuration and only controlled via the 8-button wall switch the brightness level is maintained and after switching the dimmer to ON restored to the last level.
But as soon as the buttons in Node Red are used the brightness is set to 0% when the device is set to OFF.

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC2531, CC2530
Adapter firmware version: 20190608

Koenkk, 2020-09-02:
> I expect the brightness level to be left untouched and not set to 0% when switching the device to OFF.

This is expected behaviour. For Zigbee2MQTT brightness 0 = OFF, anything else is ON.

However when turning it on again the brightness should be restored. First update to zigbee2mqtt 1.14.4 and try to reproduce the bug. If you can provide the debug logging here.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

georgeavd, 2020-09-03:
Indeed brightness 0 = OFF, but OFF ≠ brightness 0. But that's not whats happening now.

Zigbee2MQTT is not the right place for changing the brightness parameter when switching OFF a lightbulb. It should act like a wall switch/dimmer when switching ON and OFF via Node Red, HomeAssistant or Homekit, but it's not.

When switching ON a dimmer it's expected that it returns to the last used brightness setting, but that information is now lost because of the brightness parameter being forced/changed to 0 bij the zigbee2MQTT interface. That's not correct.

Zigbee2MQTT should not change parameters (based on MQTT status messages) this leads to unexpected behaviour for all other users. If someone wants a slider in the GUI being also set to 0 when a dimmer is switched to OFF is something that should be solved in somewhere else (E.g. HomeAssistant or Node Red) but not in the zigbee2MQTT-interface.
Zigbee2MQTT is not the right place for this. zigbee2MQTT should be transparent.

As requested I have tested this on 1.14.4 as well. The brightness level is till forced to 0 when switching OFF.

Koenkk, 2020-09-04:
Let's continue in #2566

closed 2020-09-04 by Koenkk