Koenkk/zigbee2mqtt #4238: Aquara motion sensor strange problem

Maurotb, 2020-09-01:
# Bug Report
## What happened

My motion sensor (4 sensors) works correctly for 1 year, now
every day i need to repair.
After repair a sensor, for 15/20 minuts it trigger continnuosly, after this time period, work for some hours and disconnect again.
When sensor is disconnected, in log i see nothing for this motion sensors.
Other door sensors and switch works correctly

This is log after repair one sensor (during 15 29 minutes up/down) 
info  2020-09-01 21:58:02: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:02: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:02: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":true,"battery":100,"voltage":3005}'
info  2020-09-01 21:58:07: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:08: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:08: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:09: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:09: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:09: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:09: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":true,"battery":100,"voltage":3005}'
info  2020-09-01 21:58:14: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:15: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:15: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:15: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:15: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":true,"battery":100,"voltage":3005}'
info  2020-09-01 21:58:20: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:21: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:21: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:21: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:21: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:21: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:21: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":true,"battery":100,"voltage":3005}'
info  2020-09-01 21:58:26: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:28: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:28: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":false,"battery":100,"voltage":3005}'
debug 2020-09-01 21:58:28: Received Zigbee message from 'Sensore-Movimento-Corridoio-2', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-09-01 21:58:28: MQTT publish: topic 'zigbee2mqtt/Sensore-Movimento-Corridoio-2', payload '{"illuminance":0,"illuminance_lux":0,"linkquality":42,"occupancy":true,"battery":100,"voltage":3005}'

## Version
Docker 1.14.3

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2

CC2531

Adapter firmware version:
CC2531_DEFAULT_20190608

Maurotb, 2020-09-04:
I have try to change my cc2531 and downgrade zigbee2mqtt to 1.13.0 ,same result , after rejoin,
i have up/down for some minuts, after work for some ours and disconnect....

Koenkk, 2020-09-05:
Make sure to check https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html#troubleshooting-device-stops-sending-messagesdisconnects-from-network

Maurotb, 2020-09-14:
now work... i think is interference...

closed 2020-09-14 by Maurotb