Koenkk/zigbee2mqtt #4239: Honeywell smoke don't work fully behind router?

andriej, 2020-09-01:
# Bug Report
## What happened
I've wanted to automate selftest of Honeywell Xiaomi smoke sensors, which are working well with Z2M on Conbee stick. Although all of them are paired, I couldn't manage to set sensitivity and perform self-test.

Setting sensitivity on them or self-test always failed:
```
igbee2mqtt/bridge/log {"type":"zigbee_publish_error","message":"Publish 'set' 'selftest' to 'Honeywell Smoke z' failed: 'TypeError: Write 0x00158d0002f37e32/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Cannot read property 'frame' of null)'","meta":{"friendly_name":"Honeywell Smoke Garaż"}}
zigbee2mqtt/CC2591 Extender Piętro {"led":true,"linkquality":39}
zigbee2mqtt/bridge/log {"type":"zigbee_publish_error","message":"Publish 'set' 'selftest' to 'Honeywell Smoke y' failed: 'TypeError: Write 0x00158d0002f375c9/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Cannot read property 'frame' of null)'","meta":{"friendly_name":"Honeywell Smoke Korytarz"}}
zigbee2mqtt/bridge/log {"type":"zigbee_publish_error","message":"Publish 'set' 'selftest' to 'Honeywell Smoke x' failed: 'TypeError: Write 0x00158d0002f376c8/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Cannot read property 'frame' of null)'","meta":{"friendly_name":"Honeywell Smoke Przedpokój"}}

But!

I've managed to get them working by getting them out from the roof and coming close to coordinator.
They instantly react for self-test or sensitivity change - as long as I don't place them back on their ceiling spot.

In between the only routers I have are cc2530+cc2591 (firmware from z2m website) and Ikea tradfri extenders.
I guess it's something with IAS zones or... idk.
```
## What did you expect to happen
Hear self-test or be able to change sensitivity
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: conbee2  latest firmware
Adapter firmware version:

Koenkk, 2020-09-02:
This might be caused by the router that the smoke sensor is connected to. The smoke sensor only checks for the request every 30 seconds, some routers drop the messages earlier so the smoke sensor will never get the request.

To investigate further a sniff is required, but I have doubts this can be solved from the Zigbee2MQTT side. I assume that you keep the smoke sensor at the same location after it's paired (because switching parents could also cause this).

andriej, 2020-09-02:
I have tried two approaches (I have 3 story house).
I have tried to pair smoke sensors from their final location, so 2 floors away from conbee2 coordinator - of course they did pair through router(s), although no self-test or possibility to change sensitivity.

So I took those sensors who are far from coordinator next to it - I've paired them (without removing, they were reporting no problem all the time) next to coordinator - set sensitivity worked and self-test too.

So I moved them to final location and clicked 'self-test' on them once. No further self-test or sensitivity-set works.
Right now the only routers I have are Ikea Extenders (Tradfri ones) and CC2530+CC2591 which are the beasts in terms of range provided.

McGiverGim, 2020-09-03:
You are sure that id does not work at all? There are reports that say that it can last until 5 minutes to execute the self-test: https://github.com/Koenkk/zigbee2mqtt/issues/3818

In my case it takes some time too, when under a router. Using an Aqara relay as router.

andriej, 2020-09-03:
Is it possible that Aqara does relay specific messages better than i.e. CC2530+CC2591 or Ikea Extender?
I'm sure because having 4 smoke sensors on different floors I can hear if they do or do not beep. And they don't when they are away from coordinator (conbee2) at the moment.

andriej, 2020-09-03:
Also - the error message is different. In linked Issue there is clear 'timeout' while on mine it's totally different I guess.

Koenkk, 2020-09-04:
> Also - the error message is different. In linked Issue there is clear 'timeout' while on mine it's totally different I guess.

I missed this (thought it was a timeout). `Cannot read property 'frame' of null` seems to be an issue for the conbee implementation, please provide the herdsman debug logging so @ChrisHae can take a look at this.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Mouky66, 2020-09-07:
I have the same error with a Osram light bulb:
Conbee I as Coordinator and Osram Smart+ plug as Router

info  2020-09-07 09:23:44: MQTT publish: topic 'zigbee2mqtt/0xf0d1b800001054de', payload '{"brightness":0,"linkquality":0,"state":"OFF","update_available":true}'
error 2020-09-07 09:24:44: Publish 'set' 'state' to '0xf0d1b800001054de' failed: 'TypeError: Read 0xf0d1b800001054de/1 genLevelCtrl(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)'
info  2020-09-07 09:24:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xf0d1b800001054de' failed: 'TypeError: Read 0xf0d1b800001054de/1 genLevelCtrl([\"currentLevel\"], {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Cannot read property 'frame' of null)'","meta":{"friendly_name":"0xf0d1b800001054de"},"type":"zigbee_publish_error"}'

ChrisHae, 2020-09-08:
@andriej
> So I moved them to final location and clicked 'self-test' on them once. No further self-test or sensitivity-set works.
> Right now the only routers I have are Ikea Extenders (Tradfri ones) and CC2530+CC2591 which are the beasts in terms of range provided.
Are you sure that the Honeywell sensors choose new parents when the old parent is lost? Because I know that some Xiamoi devices (or perhaps all) don't choose new parents. That means if you pair it new the conbee2 stick and then move it out of range it wont choose a new parent and so stop working correctly. (So that is what Koenkk said already) A zigbee sniffing log would really help in this case.

But also a zigbee-heards log would also help a lot with deconz adapter log output enabled. I could not reproduce those "Type Error" messages for myself yet.

ChrisHae, 2020-09-08:
@Mouky66 could you please also enable deconz adapter debug output:
add -e DEBUG=zigbee-herdsman:* to your docker run command or add this to your docker-compose file:

    environment:
      - DEBUG=zigbee-herdsman:*

or start npm with: DEBUG=zigbee-herdsman:* npm start

andriej, 2020-09-08:
> Are you sure that the Honeywell sensors choose new parents when the old parent is lost? Because I know that some Xiamoi devices (or perhaps all) don't choose new parents. That means if you pair it [...]

If they wouldn't, I probably wouldn't see messages from those sensors, right?
I was sniffing MQTT topic and seeing those devices at that moment AFAIR
 
> But also a zigbee-heards log would also help a lot with deconz adapter log output enabled. I could not reproduce those "Type Error" messages for myself yet.

Will try to provide them (from add-on in HA) when I'll be able to

andriej, 2020-09-09:
One more field update (since it's too late to try things out with smoke sensors as they make noise):
DJT11LM Vibration sensor from Xiaomi, I set it's sensitivity next to coordinator (no other option) to high and moved to final place. It's working there, reports sent etc. Altthough I wanted to change it to medium.
According to docs, I'm pushing this payload to MQTT:

topic: zigbee2mqtt/Xiaomi Vibration Garaż/set
payload: '{"sensitivity": "medium"}'
qos: 2
retain: false

and press button every second until it's picked up by sensors.
It never will, as I've been trying for 10 minutes, constantly...

Full error log: https://pastebin.com/wx7z3c80
The error itself shows up pretty quick after the MQTT message:
```
zigbee2mqtt/Xiaomi Vibration Garaż/set {"sensitivity": "medium"}
zigbee2mqtt/Xiaomi Vibration Garaż {"angle":59,"angle_x":5,"angle_x_absolute":85,"angle_y":4,"angle_y_absolute":86,"angle_z":-84,"battery":100,"linkquality":62,"sensitivity":"high","strength":3, "voltage":3085}
zigbee2mqtt/bridge/log {"message":"Publish 'set' 'sensitivity' to 'Xiaomi Vibration Garaż' failed: 'Error: Write 0x00158d000483bb1a/1 genBasic({\"65293\":{\"value\":11,\"type\":32}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Status 'undefined')'","meta":{"friendly_name":"Xiaomi Vibration Garaż"},"type":"zigbee_publish_error"}
```
and previously I've also catched:
```
zigbee2mqtt/bridge/log {"message":"Publish 'set' 'sensitivity' to 'Xiaomi Vibration Garaż' failed: 'TypeError: Write 0x00158d000483bb1a/1 genBasic({\"65293\":{\"value\":11,\"type\":32}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Cannot read property 'frame' of null)'","meta":{"friendly_name":"Xiaomi Vibration Garaż"},"type":"zigbee_publish_error"}
```

all that on fully working mesh (with no LQI issues as 2xcc2591 extenders present) with conbee2 stick.

andriej, 2020-09-09:
Log from the Vibration-sensor issue with zigbee-herdsman debug turned on: https://pastebin.com/bMAribH2 (full log),
part of it:
```
Zigbee2MQTT:debug 2020-09-10 00:29:24: Received MQTT message on 'zigbee2mqtt/Xiaomi Vibration Garaż/set' with data '{"sensitivity": "medium"}'
Zigbee2MQTT:debug 2020-09-10 00:29:24: Publishing 'set' 'sensitivity' to 'Xiaomi Vibration Garaż'
2020-09-09T22:29:24.275Z zigbee-herdsman:controller:endpoint Write 0x00158d000483bb1a/1 genBasic({"65293":{"value":11,"type":32}}, {"timeout":35000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null})
2020-09-09T22:29:24.278Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2020-09-09T22:29:24.279Z zigbee-herdsman:deconz:adapter [ { attrId: 65293, attrData: 11, dataType: 32 } ]
2020-09-09T22:29:24.283Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xa6d5 EP:1 SeqNr. 127 request id: 29
2020-09-09T22:29:24.286Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 29 status: 0
2020-09-09T22:29:24.304Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-09-09T22:29:24.334Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 128
2020-09-09T22:29:24.337Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xa6d5 request id: 29 confirm status: 0
2020-09-09T22:29:24.338Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send
2020-09-09T22:29:26.231Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-09T22:29:26.250Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 129
2020-09-09T22:29:26.254Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 129 srcAddr: 0xa6d5 destAddr: 0x0 profile id: 0x104 cluster id: 0x0
2020-09-09T22:29:26.254Z zigbee-herdsman:deconz:frameParser response payload: 28,95,17,14,10,5,0,66,18,108,117,109,105,46,118,105,98,114,97,116,105,111,110,46,97,113,49,1,255,66,46,1,33,13,12,3,40,23,4,33,168,51,5,33,150,0,6,36,0,0,0,0,0,8,33,8,3,10,33,0,229,152,33,40,0,153,33,0,0,154,37,210,255,236,3,207,0
2020-09-09T22:29:26.260Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":14,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":5,"dataType":66,"attrData":"lumi.vibration.aq1"},{"attrId":65281,"dataType":66,"attrData":{"1":3085,"3":23,"4":13224,"5":150,"6":[0,0],"8":776,"10":58624,"152":40,"153":0,"154":[207,65863634]}}]},"address":42709,"endpoint":1,"linkquality":62,"groupID":null}'
Zigbee2MQTT:debug 2020-09-10 00:29:26: Received Zigbee message from 'Xiaomi Vibration Garaż', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3085,"10":58624,"152":40,"153":0,"154":[207,65863634],"3":23,"4":13224,"5":150,"6":[0,0],"8":776},"modelId":"lumi.vibration.aq1"}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2020-09-10 00:29:26: MQTT publish: topic 'zigbee2mqtt/Xiaomi Vibration Garaż', payload '{"angle":59,"angle_x":5,"angle_x_absolute":85,"angle_y":4,"angle_y_absolute":86,"angle_z":-84,"battery":100,"linkquality":62,"sensitivity":"high","strength":3,"voltage":3085}'
2020-09-09T22:29:26.266Z zigbee-herdsman:deconz:adapter response received
2020-09-09T22:29:26.267Z zigbee-herdsman:controller:endpoint Write 0x00158d000483bb1a/1 genBasic({"65293":{"value":11,"type":32}}, {"timeout":35000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Status 'undefined')
Zigbee2MQTT:error 2020-09-10 00:29:26: Publish 'set' 'sensitivity' to 'Xiaomi Vibration Garaż' failed: 'Error: Write 0x00158d000483bb1a/1 genBasic({"65293":{"value":11,"type":32}}, {"timeout":35000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Status 'undefined')'
Zigbee2MQTT:debug 2020-09-10 00:29:26: Error: Write 0x00158d000483bb1a/1 genBasic({"65293":{"value":11,"type":32}}, {"timeout":35000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Status 'undefined')
    at Endpoint.checkStatus (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:166:23)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:195:26)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
Zigbee2MQTT:info  2020-09-10 00:29:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'sensitivity' to 'Xiaomi Vibration Garaż' failed: 'Error: Write 0x00158d000483bb1a/1 genBasic({\"65293\":{\"value\":11,\"type\":32}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Status 'undefined')'","meta":{"friendly_name":"Xiaomi Vibration Garaż"},"type":"zigbee_publish_error"}'
[... cut ...] 4],"3":22,"4":13224,"5":150,"6":[0,0],"8":776},"modelId":"lumi.vibration.aq1"}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2020-09-10 00:29:27: MQTT publish: topic 'zigbee2mqtt/Xiaomi Vibration Garaż', payload '{"angle":59,"angle_x":5,"angle_x_absolute":85,"angle_y":4,"angle_y_absolute":86,"angle_z":-84,"battery":100,"linkquality":62,"sensitivity":"high","strength":3,"voltage":3085}'
2020-09-09T22:29:27.835Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 131
```

ChrisHae, 2020-09-15:
I've improved reading responses in the latest commit. Because it seems that the deconz adapter took the wrong response from the sensor and thus throw an error: status undefined. I also fixed the cryptic error messages: Cannot read property 'frame' of null. I think this error was mainly because of a timeout waiting for a response.
Perhaps this will not cure your initial problem.
But there is also a new FW available for ConBee2: http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26650700.bin.GCF and RaspBee2
http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26650700.bin.GCF
which improved routing. Perhaps you try it out.

andriej, 2020-09-15:
I did upgrade of the firmware as linked [here](https://github.com/Koenkk/zigbee-herdsman/issues/72#issuecomment-692509807)
[image]

I have issues installing zigbee2mqtt-edge, but will try out today later on.

ChrisHae, 2020-09-15:
Sry, reading the coordinator version is bugged with the ne FW. I'm already on it. All other functions should work.

andriej, 2020-09-15:
Ok, I've switched to -edge and will try to see if there are any changes.
Everything else is working like you say.

ChrisHae, 2020-09-16:
Uploaded a fix for the wrong coordinator version. With that reading the FW version works with the new and the old FW.

ChrisHae, 2020-09-24:
@andriej do you have some new experiences with the new firmware? Please note there was an update in the z2mqtt dev branch yesterday. There last commited code produced some errors while pairing new devices.

So I tested for myself with the aqara vibration sensor. And it joined my network without problems and I could configure it. After a while I could not configure it anymore. No differences if the sensor was near the ConBee2 or far away.

So I spoke with @manup who has more experience with these sensors and it seems the vibration sensor is only configurable a short. time after it joined the network. And then it wakes up once an hour and is only configurable for a short time. 

I have sniffed this with wireshark and I can see "indirect transaction expiry" when the sensor sleeps. That means the parent can't forward data requests to it. It wakes up from time to time and sends "mac data requests" so this would be the time where it can be configured.

So the application should store the configure command and send it when the sensor awakes and reports to the coordinator. Perhaps it is similar with the honeywell sensors. @koenkk is something like this implemented in z2mqtt?

ChrisHae, 2020-09-24:
[image]

andriej, 2020-09-24:
@ChrisHae I didn't try latest dev yet, but my mesh became stable enough. Sensors are present and working, self test without change

andriej, 2020-10-25:
Will test this week again

closed 2020-12-03 by github-actions[bot]