Koenkk/zigbee2mqtt #4240: Tuya (TS0505A) RGBW+CCT LED not recognized

MSchutt, 2020-09-01:
I have bought: https://www.alibaba.com/product-detail/Factory-direct-sales-Smart-Zigbee-GU10_62501812912.html and unfortunatley they are not supported. They seem to be Tuya devices (atleast that's my guess). I have tried to control them via generic (fromZigbee, toZigbee) but I only seem to be able to control on/off but not brightness or color. 
Log from zigbee2mqtt
 `{"type":"devices","message":{"ieeeAddr":"0x14b457fffe1a7996","type":"Router","networkAddress":16252,"model":"TS0505A","vendor":"-","description":"-","friendly_name":"0x14b457fffe1a7996","manufacturerID":4098,"modelID":"TS0505A","hardwareVersion":1,"lastSeen":1599003895225}`

What would the steps be to get dimming etc. working? Do I need to buy a Tuya bridge and then sniff the ZigBee packets? Is there an easier approach? Any help would be greatly appreciated!

MSchutt, 2020-09-02:
I forgot: entry from `database.db` for the device

`{"id":8,"type":"Router","ieeeAddr":"0x086bd7fffefb66bf","nwkAddr":42085,"manufId":4098,"manufName":"_TZ3000_sosdczdl","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,10,4,5,6,8,768],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":1,"offWaitTime":0}},"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0013","65506":21,"modelId":"TS0505A","manufacturerName":"_TZ3000_sosdczdl","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"61440":0,"colorTemperature":255}},"genLevelCtrl":{"attributes":{"currentLevel":255}}},"binds":[]}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1599004841967}`

Koenkk, 2020-09-02:
For sure TuYa! (because of the TS prefix in the modelID).

What errors do you see in the zigbee2mqtt debug log when setting the brightness?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

MSchutt, 2020-09-02:
Hi! Thanks for your quick reply.
The weird thing is that Z2M thinks that all works as intended since there are no errors.. it recognizes the on/off state correctly and thinks that color_temp & brightness are correct (?) but in fact they aren't!
These are the logs from Z2M when I change the brightness to 112:

> zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-02 22:36:07: Received MQTT message on 'zigbee2mqtt/0x086bd7fffefb6695/set' with data '{"state": "ON", "brightness": 112}'
> zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-02 22:36:07: Publishing 'set' 'brightness' to '0x086bd7fffefb6695'
> zigbee2mqtt      | Zigbee2MQTT:info  2020-09-02 22:36:07: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffefb6695', payload '{"state":"ON","linkquality":63,"brightness":112,"color_temp":255}'

I set the brightness to 112.. no errors, but the lights are still at 100% (same issue for color temperature and color)

Koenkk, 2020-09-02:
Since it's TuYa it could be that a custom protocol is used to set brightness :( 

@kirovilya do know more about this?

MSchutt, 2020-09-02:
Damn that's too bad :(

Quick update: I tested color temperature again and it worked with values from 0-255 (warm-white to cold-white) - home assistant sent too high numbers so it failed..
So state & color_temperature works with generic (toZigbee & fromZigbee) but color and brightness still does not!

MSchutt, 2020-09-02:
Just to be 100% sure: is this the correct payload I am sending via MQTT?

`{ "state": "on", "brightness": 1, "color_temp": 1, "color":{"x":0.38402,"y":0.378951509567474} }`

This turns the light on & sets its color_temp to 1 (brightness and color are not recognized) - am I sending the wrong payload?

Koenkk, 2020-09-03:
yes, thats correct, you can also try only setting brightness.

MSchutt, 2020-09-05:
Thank you. Unfortunately it does not work.
I will receive a Tuya Gateway next week. I will try to sniff the packages and try to implement it. Do you have some detailed resources on how to do that? (best practices, common pitfalls etc.)

Koenkk, 2020-09-05:
https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-1-b20227251d46 and https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d should help.

Also a general guide can be found here: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

zhouF96, 2020-09-18:
Hi, How is every thing going on? 2 days ago I have received a spot light shipped with TS0505A module, and tested with 
extend: generic.light_onoff_brightness_colortemp_colorxy
every thing seemed to worked well. @MSchutt 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1558

MSchutt, 2020-09-18:
Hi. I tried it out again and it just does not work. I have recieved my gateway yesterday and it works with TuYa Gateway.. It seems like the lamps also do not respond with any color at all (so the response payload never include the current color while it does with color_temp and state). 👎 Maybe I recieved different LEDs than you guys. Don't know what's wrong here unfortunately.

MSchutt, 2020-09-18:
@zhouF96 How does you log like when you pair the device?

> zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:08: Received Zigbee message from '0x086bd7fffefb6695', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:08: Received Zigbee message from '0x086bd7fffefb6695', type 'readResponse', cluster 'genBasic', data '{"appVersion":65}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:08: Received Zigbee message from '0x086bd7fffefb6695', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:08: Received Zigbee message from '0x086bd7fffefb6695', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:08: Received Zigbee message from '0x086bd7fffefb6695', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:08: Received Zigbee message from '0x086bd7fffefb6695', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:08: Successfully interviewed '0x086bd7fffefb6695', device has successfully been paired
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:08: Device '0x086bd7fffefb6695' is supported, identified as: TuYa RGB+CCT light controller (TS0505A)
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"RGB+CCT light controller","friendly_name":"0x086bd7fffefb6695","model":"TS0505A","supported":true,"vendor":"TuYa"},"type":"pairing"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:08: Received MQTT message on 'zigbee2mqtt/bridge/config/device_options' with data '{"friendly_name":"0x086bd7fffefb6695","options":{"retain":true}}'
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:08: Changed device specific options of '0x086bd7fffefb6695' ({"retain":true})
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:09: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:09: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffefb6695', payload '{"linkquality":44,"state":"ON"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:09: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'lightingColorCtrl', data '{"61440":0}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:09: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":0}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:09: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":255}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:09: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffefb6695', payload '{"brightness":255,"linkquality":44,"state":"ON"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:09: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":255}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:09: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffefb6695', payload '{"brightness":255,"color_temp":255,"linkquality":44,"state":"ON"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:13: Received Zigbee message from '0x086bd7fffefb6695', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:19: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":0}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:19: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2020-09-18 17:58:19: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffefb6695', payload '{"brightness":255,"color_temp":255,"linkquality":47,"state":"ON"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:20: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":255}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:20: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":65}' from endpoint 1 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 17:58:20: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genBasic', data '{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0013"}' from endpoint 1 with groupID 0
[... cut ...]
Maybe there is a difference? I have noticed that the appVersion is quite low for my LEDs (I have seen other devices at appVersion 80 or something... Also if I leave the device without sending anything the LED keeps sending:

> zigbee2mqtt      | Zigbee2MQTT:debug 2020-09-18 18:17:36: Received Zigbee message from '0x086bd7fffefb6695', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":65}' from endpoint 1 with groupID 0

I am still beginner level with Z2M so I am not sure if that's normal behaviour. Thank you!

MSchutt, 2020-09-18:
And for completness-sake the entry from `database.db`

> {"id":2,"type":"Router","ieeeAddr":"0x086bd7fffefb6695","nwkAddr":68,"manufId":4098,"manufName":"_TZ3000_sosdczdl","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,10,4,5,6,8,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0013","65506":21,"modelId":"TS0505A","manufacturerName":"_TZ3000_sosdczdl","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"offWaitTime":0}},"lightingColorCtrl":{"attributes":{"61440":0,"colorTemperature":1}},"genLevelCtrl":{"attributes":{"currentLevel":255}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1600445856765}

zhouF96, 2020-09-18:
> @zhouF96 How does you log like when you pair the device?
>
>
> Maybe there is a difference? I have noticed that the appVersion is quite low for my LEDs (I have seen other devices at appVersion 80 or [...]

sorry, currently I don't have it in my office , and I'will receive a new one next Monday, Currently I can only provide the info  in my **database.db**

it looks like this:

`{"id":2,"type":"Router","ieeeAddr":"0x842e14fffe9a5572","nwkAddr":8133,"manufId":4098,"manufName":"_TZ3000_k5t1kmxl","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_k5t1kmxl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1600148218504}`

MSchutt, 2020-10-09:
Quick update here: I have managed to implement the protocol for Tuya.. it's quite different - what do I do now? It's the same modelID, but I don't want to overwrite the other's generic implementation. @Koenkk

Koenkk, 2020-10-09:
@MSchutt please submit a pr using a `fingerprint` instead of `zigbeeModel` (e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/f2eda27772801c2bb670ad1297409a9f8ea651d4/devices.js#L1276). Use: `fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_sosdczdl'}],`. Zigbee2MQTT prefers fingerprint identifciation of zigbeeModel identification. As for the others (I expect) the manufacturer won't match, it will fallback to the zigbeeModel definition of TS0505A.

closed 2020-10-16 by Koenkk

VincentSC, 2020-12-15:
FYI: Livarno Lux LED strip (Lidl) has the same name and is recognized as such. https://www.lidl-shop.nl/LIVARNO-LUX-RGB-LED-verlichtingsband-Zigbee-Smart-Home/p100306624

MSchutt, 2020-12-15:
Thank you! Does it work for you? After more detailed testing I found some minor bugs in my implementation which I am currently working on!

VincentSC, 2020-12-16:
Yes, it just works without problems