Koenkk/zigbee2mqtt #4241: Not able to add new device after the very first boot up

saury, 2020-09-02:
# Bug Report
## What happened

It seems my coordinator is not able to add new device by triggering the permit join event after the very first boot up. The only way to add new devices is to change the `channel` in z2m configuration and restart which would reset all the device, then add all devices include the new ones by the first time the clean service been booted.  

I'm not sure if it's an software or hardware issue.

Here's some logs:

```
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-09-02T10:12:33: PM2 log: Launching in no daemon mode
2020-09-02T10:12:33: PM2 log: App [npm:0] starting in -fork mode-
2020-09-02T10:12:33: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.3 start /zigbee2mqtt-1.14.3
> node index.js
Zigbee2MQTT:info  2020-09-02 10:12:41: Logging to console and directory: '/share/zigbee2mqtt/log/2020-09-02.10-12-41' filename: log.txt
Zigbee2MQTT:info  2020-09-02 10:12:46: Starting Zigbee2MQTT version 1.14.3 (commit #unknown)
Zigbee2MQTT:info  2020-09-02 10:12:46: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-09-02 10:12:48: zigbee-herdsman started
Zigbee2MQTT:info  2020-09-02 10:12:48: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
Zigbee2MQTT:info  2020-09-02 10:12:50: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.3","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"network":{"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11},"log_level":"info","permit_join":true}'
```

## What did you expect to happen

Can add new device by triggering the permit join event.

## How to reproduce it (minimal and precise)

1. start z2m service
2. wait for several minutes later
3. unable to add new devices anymore

## Debug Info
Zigbee2MQTT version: latest
Adapter hardware: CC2530
Adapter firmware version: 20190619

Koenkk, 2020-09-02:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all ?

saury, 2020-09-03:
> Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all ?

Tried all of them except reflashing the coordinator. I bought a 2nd hand coordinator and has no flash tool.

closed 2020-10-11 by github-actions[bot]