Koenkk/zigbee2mqtt #4242: Error every several days

zhapwey, 2020-09-02:
# Bug Report
## What happened

The installation starts and works correctly, it is working for several days well, and after 2 or 5 days mqtt fails, it stays stopped without sending or receiving anything, and if I stop the zigbee2mqtt service and try to raise it again, it gives me an error in the adapter usb.
Sep 02 09:17:35 raspberrypi npm[675]: (node:823) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - version after 6000ms
Sep 02 09:17:35 raspberrypi npm[675]:     at Timeout._onTimeout (/opt/zigbee/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Sep 02 09:17:35 raspberrypi npm[675]:     at listOnTimeout (internal/timers.js:549:17)
Sep 02 09:17:35 raspberrypi npm[675]:     at processTimers (internal/timers.js:492:7)
Sep 02 09:17:35 raspberrypi npm[675]: (node:823) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an asyn
Sep 02 09:17:35 raspberrypi npm[675]: (node:823) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not

I have to restart the device and everything works correctly again.

and the general status of z2m remains online.

## What did you expect to happen

That will not be blocked

## How to reproduce it (minimal and precise)

It's very random
## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC26X2R1

Koenkk, 2020-09-02:
- How many devices do you have on your network?
- What CC26X2R1 firmware are you running?
- Please provide the herdsman debug logging of the moment before the crash

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

zhapwey, 2020-09-08:
Hello:
can you help me modify the z2m service because i am unable to make it work for the service to invoke the program in debug mode. If I launch it by hand it works correctly but I want to launch it as a service to save the entire log.

Koenkk, 2020-09-09:
```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment="DEBUG=zigbee-herdsman*"
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=file:/var/log/zigbee2mqtt.log
StandardError=file:/var/log/zigbee2mqtt.log
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

Log can be found in `/var/log/zigbee2mqtt.log`

zhapwey, 2020-09-09:
Thank you very much I did not give with it, I have separated error and output in two different files.
Very thankful, It works great

zhapwey, 2020-09-17:
hello @Koenkk 
I've been analyzing the behavior of the network these days.
What happens to me is that when I restart the raspberry everything works correctly and everything connects, after a few days the Availability pings begin to fail. After this, the devices are disconnected from the network and everything stops communicating.

If I restart the raspberry that has z2m it reconnects everything correctly.

I have two networks, one inside the house that works correctly and I mount it to try to reach the gate and I have two cc2530 + cc2592, one at home and the other next to the gate with a relay + pressure sensor.
In case it gives you clues that it may be happening.

Best Regards,

Koenkk, 2020-09-17:
There are some known issues in 1.14.4 regarding availability and stability decrease when running for several days.

This should have been fixed in the latest dev branch: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

closed 2020-10-25 by github-actions[bot]