Koenkk/zigbee2mqtt #4245: Zemismart Relay is not update relay status using conbee 2 as coordinator

alexandrepossebom, 2020-09-02:
# Bug Report
## What happened

I am using conbee 2 as coordinator and everything is working well.

I added LXN59-2S7LX1.0 device and it's working but the physical switch don't update the MQTT, but if I send command via MQTT it's turns on and off the relay as expected.

I tried with my old cc2531 coordinator to add this device and everything works.

## What did you expect to happen

Physical switch update the MQTT.

## How to reproduce it (minimal and precise)

Just add this device using conbee 2 as coordinator.

## Debug Info
Zigbee2MQTT version:1.14.4
Adapter hardware: conbee 2
Adapter firmware version:

Koenkk, 2020-09-04:
Can you try changing the entry in `devices.js` to

```js
    {
        zigbeeModel: ['LXN59-2S7LX1.0'],
        model: 'LXN59-2S7LX1.0',
        vendor: 'Nue / 3A',
        description: 'Smart light relay - 2 gang',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        meta: {multiEndpoint: true},
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZW-EU-02'},
        ],
        endpoint: (device) => {
            return {'left': 1, 'right': 2};
        },
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1 , coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint1 );
            const endpoint2 = device.getEndpoint(2);
            await bind(endpoint2 , coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint2 );
        },
    },
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to edit `devices.js`.

alexandrepossebom, 2020-09-04:
> Can you try changing the entry in `devices.js` to
>
> ```js
>     {
>         zigbeeModel: ['LXN59-2S7LX1.0'],
>         model: 'LXN59-2S7LX1.0',
>         vendor: 'Nue / 3A',
>         description: 'Smart light [...]

Thank you @Koenkk it's almost works.

Now I receive update in MQTT but its create new 'state' field and doesn't matter if I change physical switch 'left' or 'right' it's update new state field named 'state' and not 'state_left' or 'state_right'.

For example: 
Before use physical switch:
```
{"linkquality":8,"state_left":"ON","state_right":"ON"}
``` 
after:
```
{"linkquality":8,"state":"OFF","state_left":"ON","state_right":"ON"}
```

Koenkk, 2020-09-04:
Should work with 

```js
    {
        zigbeeModel: ['LXN59-2S7LX1.0'],
        model: 'LXN59-2S7LX1.0',
        vendor: 'Nue / 3A',
        description: 'Smart light relay - 2 gang',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        meta: {multiEndpoint: true},
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZW-EU-02'},
        ],
        endpoint: (device) => {
            return {'left': 1, 'right': 2};
        },
        meta: {configureKey: 2, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1 , coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint1 );
            const endpoint2 = device.getEndpoint(2);
            await bind(endpoint2 , coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint2 );
        },
    },
```

alexandrepossebom, 2020-09-04:
@Koenkk now it's works!

Thank you a lot!

If you want I can open a PR with your fix and linked with this issue.

Koenkk, 2020-09-04:
Yes please do 👍

alexandrepossebom, 2020-09-04:
I tested this fix with cc2531 too and it's worked too.

Koenkk, 2020-09-04:
Thanks, will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-04 by Koenkk