Koenkk/zigbee2mqtt #4248: [Feature request] Intergrating with E-Byte E800-DTU

whc2001, 2020-09-02:
I just bought an E800-DTU (Z2530-485-27) from E-Byte for 160 CNY (~25 USD). As what the model number implies, it is based on the E18 ZigBee module (also from E-Byte), which has an CC2530, communicate via RS485, and has a maximum of 500mW (27dBm) of transmit power. Seems like someone in #2633 and #4128 is also interested in it. After got it and fiddled around, I found that it has a proprietary firmware that is designed to automatically communicate with E-Byte's modules, has proprietary software protocol (confirmed by PCAP, which from the E18-2G4U04B I bought together, that is a clone of the CC2531 USB dongle with sniffer firmware, which worked great), and won't work for our home automation devices. After teardown, I found the E18-2G4Z27SI module, the PA (metal box under the E18 module daughter board) and the RS485 daughter board:
[image]

I immediately noticed the pin header row on the main daughter board. After poking with multimeter and referring to their E18 module's pin number:
[image]
[image]

 I am able to figure out the pinout of the debugging port (it has SPI, UART and CC-Debugger at the same time, but seems like flashing only needs the CC port):
[image]

So I tried the [Tutorial for CC2530 module](https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc2530-optionally-with-a-cc2591-or-cc2592-rf-frontend) since they are both based on CC2530 and serial interface, I don't have CC debugger so I used the [flash-cc2531 with RPi](https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html#with-raspberry-pi-3min) method, which I also dumped the proprietary firmware from E-byte (change extension to HEX after downloading):
[EByte_E800_Stock.hex.txt](https://github.com/Koenkk/zigbee2mqtt/files/5164630/EByte_E800_Stock.hex.txt)

The flashing process went smoothly, but after flashing, I am not able to make it work. The serial interface at pin 10 and 11 always has TX=LOW and RX=HIGH, and sending anything from host over serial port will get a 0x00 back. The RS485 module is based on MAX485 or some clone, and has the manual half-duplex switch pin to control the data flow direction, but I am not able to trace where it goes, and it's always HIGH (enables module->host) even when I am sending data over from host. I am pretty much stuck here.

whc2001, 2020-09-04:
Tried to flash the stock firmware back, and it still does not come alive. I may have to buy another module and a CC debugger.

whc2001, 2020-09-04:
I ripped the module apart, and I am confused about the design: The module pin 12 goes to the base of a transistor, and the collector of that transistor goes to last third of the pin header on the left, so when pin 12 is LOW (according to stock firmware, normally running), pin 5 of the daughter board is HIGH (assume it has pull-up somewhere). Then that pin goes two ways: one is the cathode of the RUN (actually RUN ERR) LED, when it's running the LED should not light up, and this is correct. But it also goes to the 13 of a 74HC14 on the back of the board, which is a schmit trigger with inverter. and the pin 12 of that 74HC14 (corresponding to pin 13 which is input, 12 is output) directly goes to the MAX485 duplex pin. When the pin 5 of the daughter board is HIGH, the pin 12 of the 74HC14 should be LOW, and the MAX485 is in receiving state (host -> module). But I am not sure if the module wants to send something to the host, how will it control the pin 12? I destroyed the E18 module for now so I am not able to try it out, but from my memory the RUN led only light up when it's not ready, and won't light up when the module is transmitting data to the host.

whc2001, 2020-09-04:
Cracked open the metal cover of the E18 module, there are two chips inside: one is the expected CC2530, another is a mysterious QFN16 chip that has silkscreen printing as:
```
SKY
321
7501C
```
I assume it's an PA chip (the metal can under the daughter board is actually a DC-DC module), but I am not able to find any information in any way about this chip.

whc2001, 2020-09-04:
Tried interfacing with the E18 module directly, flashed stock and N-Stack firmware, no activity on serial port at all. Seems like I completely killed it, and now it's going to the dustbin. Although the price and specification looks tempting, I wouldn't recommend purchasing this device unless you are very experienced with CC25XX chipset. Better just go with one of the devices that this project already supported officially.

bojanpotocnik, 2020-10-16:
Have anyone tried using ethernet version? http://www.ebyte.com/en/product-class.aspx?cid=4&pid=101 It looks like it currently uses their cloud.

bojanpotocnik, 2020-10-16:
EBYTE just responded to my question
> Hello, what is the difference between E180-DTU (ZG120-ETH) and E800-DTU (Z2530-ETH-27)? Which Zigbee chip is inside E180-DTU? And which ethernet chip? It is possible to support one of these adapters in zigbee2mqtt?

with

> E180 is newer model than E800
> E180 is Zigbee 3.0，E800 is old version of Zigbee
> E180 is EFR32 chip, E800 is CC2530 chip.
> And MQTT is said is from OneNet?
> E180 supports MQTT protocol

but looking at https://github.com/Koenkk/zigbee-herdsman/issues/168 I see that EFR32 is not yet supported? I think I will order E180 and research their MQTT support.

whc2001, 2020-10-16:
> Have anyone tried using ethernet version? http://www.ebyte.com/en/product-class.aspx?cid=4&pid=101 It looks like it currently uses their cloud.

I think the -ETH version is just -DTU with a MCU that interfaces ethernet port to UART. The firmware inside wireless module should be the same.

whc2001, 2020-10-16:
> EBYTE just responded to my question
> 
> > Hello, what is the difference between E180-DTU (ZG120-ETH) and E800-DTU (Z2530-ETH-27)? Which Zigbee chip is inside E180-DTU? And which ethernet chip? It is possible to support one of these adapters in zigbee2mqtt?
> 
> with
> 
> > E180 is newer model than E800
> > E180 is Zigbee 3.0，E800 is old version of Zigbee
> > E180 is EFR32 chip, E800 is CC2530 chip.
> > And MQTT is said is from OneNet?
> > E180 supports MQTT protocol
> 
> but looking at [Koenkk/zigbee-herdsman#168](https://github.com/Koenkk/zigbee-herdsman/issues/168) I see that EFR32 is not yet supported? I think I will order E180 and research their MQTT support.

I have contacted the customer service of EBYTE (which apparently has a strict management code, I just wrote an honest review that mentions it cannot be used with other devices except their own ZigBee module, not negative review or something, which after he politely requested me to "try to remove" the review (which is not possible from both buyer and seller) or he will face a penalty), and confirmed that this module has its own firmware inside, can be only used with EBYTE's ZigBee module, and they won't disclose hardware design into. The E180 is pretty much the same, which means it's not possible to use them to interface to smart home appliances, until someone reverse engineered the module hardware and transplant the ZStack firmware for it.

whc2001, 2020-10-17:
I just bought a CC Debugger clone and another E18-2G4Z27SI module. Hope this time I can at lease dump the original firmware out successfully. Also seems like this chip CAN be used for user applications. Not sure why the ZStack firmware for CC2530 not working.

Datasheet (Chinese): https://atta.szlcsc.com/upload/public/pdf/source/20190812/C411295_2C4124A5C79537A05B9FBDABDA8E299E.pdf

whc2001, 2020-10-20:
Bad news: the stock firmware is protected

[image]

whc2001, 2020-10-20:
I erased and flashed the CC2530 firmware to the module, and nothing on UART0 of that module (P1.4/P1.5), which is the E800 used for RS485. But when I connected the UART to another set of pin for UART0 (P0.2/P0.3), I got something on reset: 

```
FE 06 41 80 01 02 00 02 06 03 C3 
```

Seems like we are progressing here.

whc2001, 2020-10-20:
I connected the module to a fresh instance of ZigBee2MQTT, and packet sniffed the serial port, seems like a LOT of communication is going on!

```
root@raspberrypi:/home/pi/interceptty# interceptty -s 'ispeed 115200 ospeed 115200' -l /dev/ttyUSB0 /dev/ttyVCP0 | interceptty-nicedump
<  ef fe 00 21 01 20             | ▒▒[NUL]![SOH]
>  fe 02 61 01 79 01 1a          | ▒[STX]a[SOH]y[SOH][SUB]
<  fe 00 21 02 23                | ▒[NUL]![STX]#
>  fe 0e 61 02 02 00 02 06 03 90 | ▒[SO]a[STX][STX][NUL][STX][ACK][ETX]▒
>  15 34 01 02 00 00 00 00 da    | [NAK]4[SOH][STX][NUL][NUL][NUL][NUL]▒
<  fe 03 21 08 00 0f 00 25       | ▒[ETX]![BS][NUL][SI][NUL]%
>  fe 03 61 08 00 01 55 3e       | ▒[ETX]a[BS][NUL][SOH]U>
<  fe 03 21 08 00 0f 00 25       | ▒[ETX]![BS][NUL][SI][NUL]%
>  fe 03 61 08 00 01 55 3e       | ▒[ETX]a[BS][NUL][SOH]U>
<  fe 03 21 08 84 00 00 ae       | ▒[ETX]![BS]▒[NUL][NUL]▒
>  fe 06 61 08 00 04 00 08 00 00 | ▒[ACK]a[BS][NUL][EOT][NUL][BS][NUL][NUL]
>  63                            | c
<  fe 03 21 08 63 00 00 49       | ▒[ETX]![BS]c[NUL][NUL]I
>  fe 03 61 08 00 01 00 6b       | ▒[ETX]a[BS][NUL][SOH][NUL]k
<  fe 01 26 04 62 41             | ▒[SOH]&[EOT]bA
>  fe 13 66 04 00 62 10 6e 30 3c | ▒[DC3]f[EOT][NUL]b[DLE]n0<
>  a4 15 62 69 e8 48 9d a4 13 06 | ▒[NAK]bi▒H▒▒[DC3][ACK]
>  c9 c9 33 64                   | ▒▒3d
<  fe 03 21 08 83 00 00 a9       | ▒[ETX]![BS]▒[NUL][NUL]▒
>  fe 04 61 08 00 02 62 1a 17    | ▒[EOT]a[BS][NUL][STX]b[SUB][ETB]
<  fe 03 21 08 2d 00 00 07       | ▒[ETX]![BS]-[NUL][NUL][BEL]
>  fe 0a                         | ▒[LF]
>  61 08 00 08 dd dd dd dd dd dd | a[BS][NUL][BS]▒▒▒▒▒▒
>  dd dd 6b                      | ▒▒k
<  fe 00 27 00 27                | ▒[NUL]'[NUL]'
>  fe 0e 67 00 00 85 50 4f 1f 00 | ▒[SO]g[NUL][NUL]▒PO[US][NUL]
>  4b 12 00 00 00 07 09 00 bb    | K[DC2][NUL][NUL][NUL][BEL][TAB][NUL]▒
<  fe 04 25 05 00 00 00 00 24    | ▒[EOT]%[ENQ][NUL][NUL][NUL][NUL]$
>  fe 01 65 05 00 61 fe 13 45 85 | ▒[SOH]e[ENQ][NUL]a▒[DC3]E▒
>  00 00 00 00 00 0d f2 2f 0d 0c | [NUL][NUL][NUL][NUL][NUL][CR]▒/[CR][FF]
>  6e 0b 08 06 05 04 03 02 01 68 | n[VT][BS][ACK][ENQ][EOT][ETX][STX][SOH]h
<  fe 03 25 4a f2 84 0b 11       | ▒[ETX]%J▒[VT][DC1]
>  fe 13 65 4a 00 84 0b 00 00 00 | ▒[DC3]eJ[NUL]▒[VT][NUL][NUL][NUL]
>  00 00 00 00 00 00 00 00 00 00 | [NUL][NUL][NUL][NUL][NUL][NUL][NUL][NUL][NUL][NUL]
>  00 00 00 b3                   | [NUL][NUL][NUL]▒
<  fe 04 25 05 00 00 00 00 24    | ▒[EOT]%[ENQ][NUL][NUL][NUL][NUL]$
>  fe 01 65 05 00 61 fe 13 45 85 | ▒[SOH]e[ENQ][NUL]a▒[DC3]E▒
>  00 00 00 00 00 0d f2 2f 0d 0c | [NUL][NUL][NUL][NUL][NUL][CR]▒/[CR][FF]
>  6e 0b 08 06 05 04 03 02 01 68 | n[VT][BS][ACK][ENQ][EOT][ETX][STX][SOH]h
<  fe 00 27 00 27                | ▒[NUL]'[NUL]'
>  fe 0e 67 00 00 85 50 4f 1f 00 | ▒[SO]g[NUL][NUL]▒PO[US][NUL]
>  4b 12 00 00 00 07 09 00 bb    | K[DC2][NUL][NUL][NUL][BEL][TAB][NUL]▒
<  fe 05 25 04 00 00 00 00 f2 d6 | ▒[ENQ]%[EOT][NUL][NUL][NUL][NUL]▒▒
>  fe 01 65 04 00 60 fe 0e 45 84 | ▒[SOH]e[EOT][NUL]`▒[SO]E▒
>  00 00 00 00 00 08 f2 e0 a1 05 | [NUL][NUL][NUL][NUL][NUL][BS]▒▒[ENQ]
>  00 00 00 00 71                | [NUL][NUL][NUL][NUL]q
<  fe 05 25 04 00 00 00 00 2f 0b | ▒[ENQ]%[EOT][NUL][NUL][NUL][NUL]/[VT]
>  fe 01 65 04 00 60 fe 0e 45 84 | ▒[SOH]e[EOT][NUL]`▒[SO]E▒
>  00 00 00 00 00 08 2f 04 01 05 | [NUL][NUL][NUL][NUL][NUL][BS]/[EOT][SOH][ENQ]
>  00 00 00 00 e8                | [NUL][NUL][NUL][NUL]▒
<  fe 05 25 04 00 00 00 00 0d 29 | ▒[ENQ]%[EOT][NUL][NUL][NUL][NUL][CR])
>  fe 01 65 04 00 60 fe 10 45 84 | ▒[SOH]e[EOT][NUL]`▒[DLE]E▒
>  00 00 00 00 00 0a             | [NUL][NUL][NUL][NUL][NUL][LF]
>  0d 04 01 05 00 00 01 19 00 00 | [CR][EOT][SOH][ENQ][NUL][NUL][SOH][EM][NUL][NUL]
>  ce                            | ▒
<  fe 05 25 04 00 00 00 00 0c 28 | ▒[ENQ]%[EOT][NUL][NUL][NUL][NUL][FF](
>  fe 01 65 04 00 60 fe 0e 45 84 | ▒[SOH]e[EOT][NUL]`▒[SO]E▒
>  00 00 00 00 00 08 0c 5e c0 05 | [NUL][NUL][NUL][NUL][NUL][BS][FF]^▒[ENQ]
>  00 00 00 00 50                | [NUL][NUL][NUL][NUL]P
<  fe 05 25 04 00 00 00 00 6e 4a | ▒[ENQ]%[EOT][NUL][NUL][NUL][NUL]nJ
>  fe 01 65 04 00 60 fe 0e 45 84 | ▒[SOH]e[EOT][NUL]`▒[SO]E▒
>  00 00 00 00 00 08 6e 04 01 05 | [NUL][NUL][NUL][NUL][NUL][BS]n[EOT][SOH][ENQ]
>  00 00 00 00 a9                | [NUL][NUL][NUL][NUL]▒
<  fe 05 25 04 00 00 00 00 0b 2f | ▒[ENQ]%[EOT][NUL][NUL][NUL][NUL][VT]/
>  fe 01 65 04 00 60 fe 12 45 84 | ▒[SOH]e[EOT][NUL]`▒[DC2]E▒
>  00 00 00 00 00 0c 0b 04 01 00 | [NUL][NUL][NUL][NUL][NUL][FF][VT][EOT][SOH][NUL]
[... 50 lines cut ...]
>  fe 01 65 36 00 52 fe 01 45 cb | ▒[SOH]e6[NUL]R▒[SOH]E▒
>  00 8f fe 03 45 b6 00 00 00 f0 | [NUL]▒▒[ETX]E▒[NUL][NUL][NUL]▒
```

And seems like ZigBee2MQTT recognized the device:

```
Zigbee2MQTT:info  2020-10-20 16:14:48: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-20.16-14-48' filename: log.txt
Zigbee2MQTT:info  2020-10-20 16:14:48: Starting Zigbee2MQTT version 1.14.4 (commit #654817a)
Zigbee2MQTT:info  2020-10-20 16:14:48: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-10-20 16:14:51: zigbee-herdsman started
Zigbee2MQTT:info  2020-10-20 16:14:51: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2020-10-20 16:14:51: Currently 0 devices are joined:
Zigbee2MQTT:warn  2020-10-20 16:14:51: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-10-20 16:14:51: Allowing new devices to join.
[... cut ...]
Zigbee2MQTT:info  2020-10-20 16:14:51: Connecting to MQTT server at mqtt://localhost
^CZigbee2MQTT:error 2020-10-20 16:14:53: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 16:14:53: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2020-10-20 16:14:53: Disconnecting from MQTT server
Zigbee2MQTT:info  2020-10-20 16:14:53: zigbee-herdsman stopped
```

So, the first challenge of supporting this device, is that the E800 uses another optional pin set of UART0 (P1.4/P1.5) rather than the default pins (P0.2/P0.3).

whc2001, 2020-10-20:
I think there are still two challenges: first is that the RS485 half duplex will require a direction control pin to control the data direction, which I have not analyzed which pin is doing that. As I am using standalone module for testing we can ignore this for now. 

Another is that the datasheet states that P1.0 and P1.1 controls the internal PA. When you receive you put P1.0 (LNA_EN) on HIGH, and when you transmit you put P1.1 (PA_EN) on HIGH. When doing nothing you put both pins on LOW.

However I just tried a MiJia pushbutton, and seems like it works. I don't have any ZigBee devices that allows me to try coordinator to device communication for now, but seems like device to coordinator communication is fine. So I am not sure the PA_EN and LNA_EN really need to be set to work, or it does not matter.

```
Zigbee2MQTT:info  2020-10-20 17:01:25: Device '0x00158d00026952e8' joined
Zigbee2MQTT:info  2020-10-20 17:01:25: Starting interview of '0x00158d00026952e8'
Zigbee2MQTT:error 2020-10-20 17:01:25: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:25: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":{"friendly_name":"0x00158d00026952e8"},"type":"device_connected"}
Zigbee2MQTT:error 2020-10-20 17:01:25: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:25: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"interview_started","meta":{"friendly_name":"0x00158d00026952e8"},"type":"pairing"}
Zigbee2MQTT:error 2020-10-20 17:01:26: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:26: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"announce","meta":{"friendly_name":"0x00158d00026952e8"},"type":"device_announced"}
Zigbee2MQTT:error 2020-10-20 17:01:26: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:26: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"battery":100,"linkquality":110,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:32: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:32: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"action":"single","battery":100,"linkquality":92,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:32: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:32: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"battery":100,"click":"single","linkquality":92,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:35: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:37: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:37: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"action":"double","battery":100,"linkquality":107,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:37: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:37: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"battery":100,"click":"double","linkquality":107,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:41: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:41: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"action":"hold","battery":100,"linkquality":107,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:41: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:41: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"battery":100,"click":"long","linkquality":107,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:42: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:42: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"action":"release","battery":100,"duration":816,"linkquality":105,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:42: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:42: Cannot send message: topic: 'zigbee2mqtt/0x00158d00026952e8', payload: '{"battery":100,"click":"long_release","duration":815,"linkquality":105,"voltage":3052}
Zigbee2MQTT:error 2020-10-20 17:01:45: Not connected to MQTT server!
Zigbee2MQTT:info  2020-10-20 17:01:48: Successfully interviewed '0x00158d00026952e8', device has successfully been paired
Zigbee2MQTT:info  2020-10-20 17:01:48: Device '0x00158d00026952e8' is supported, identified as: Xiaomi MiJia wireless switch (WXKG01LM)
Zigbee2MQTT:error 2020-10-20 17:01:48: Not connected to MQTT server!
Zigbee2MQTT:error 2020-10-20 17:01:48: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"interview_successful","meta":{"description":"MiJia wireless switch","friendly_name":"0x00158d00026952e8","model":"WXKG01LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}
```

bojanpotocnik, 2020-10-21:
Thank you for your insights. If I understand correctly, you have RS-485 version. Last week I ordered E180-DTU (ZG120-ETH), eager to see which interface is used there - hoping for UART + some useful SoC for ethernet. I will probably receive it in the middle of November.

I also ordered [Eport Pro-EP20](https://www.aliexpress.com/item/32934139895.html) which could be, in combination with [CC2530+CC2591](https://www.aliexpress.com/item/4000118023903.html) module, good alternative to achieve E180-DTU (ZG120-ETH) functionality by using more open SDK.

whc2001, 2020-10-21:
> Thank you for your insights. If I understand correctly, you have RS-485 version. Last week I ordered E180-DTU (ZG120-ETH), eager to see which interface is used there - hoping for UART using original set of pins + some useful SoC for ethernet. I will probably receive it in the middle of November.
> 
> I also ordered [Eport Pro-EP20](https://www.aliexpress.com/item/32934139895.html) which could be, in combination with [CC2530+CC2591](https://www.aliexpress.com/item/4000118023903.html) module, good alternative to achieve E180-DTU (ZG120-ETH) functionality by using more open SDK.

Yep the one I got is the E800-485-27 which uses RS485 (although I already destroyed it). It's a pity that E180 uses the the EFR32 chipset that ZStack firmware does not currently support. As what I can see, their DTU products are really meant to use with their own modules since they will have the same firmware. If you want to use it with other brand products then you will have to reflash it with something else.

whc2001, 2020-10-21:
I traced and double checked the wires regarding to RS485 duplex switching, and it really goes like this:

[image]

The direction of MAX485 transciver is solely controled by the RUN_LED pin of the module (negated by NPN transistor, and negated again by 74HC14D Schmitt inverter). However I don't even think the RUN_LED should light up during device to host transmission. This pin should only be toggled when the software encounter an error for the stock firmware. Anyone have any idea about that?

whc2001, 2020-10-21:
Just found that the `E18-2G4Z27SI_UserManual-H` datasheet mentioned how to modify source code to control PA, and ZStack provided compilation instruction. Maybe I'll try to edit and compile my own ZStack firmware.

whc2001, 2020-10-21:
## Enable PA

### HAL\Target\Config\hal_board_cfg.h

```
#define xHAL_PA_LNA
```
change to
```
#define HAL_PA_LNA
```

### MAC\Low Level\System\mac_radio_defs.c -> macRadioTurnOnPower()

Code between comments "...during sleep." and "For any RX...", only keep
```
       /* P1_1 -> PAEN */
      RFC_OBS_CTRL0 = RFC_OBS_CTRL_PA_PD_INV;
      OBSSEL1       = OBSSEL_OBS_CTRL0;
      
      /* P1_0 -> EN (LNA control) */
      RFC_OBS_CTRL1 = RFC_OBS_CTRL_LNAMIX_PD_INV;
      OBSSEL0       = OBSSEL_OBS_CTRL1;
```

### MAC\High Level\mac_pib.c

Field before "/* phyTransmitPower */", change to
```
#if defined (HAL_PA_LNA)
  27,                                          /* phyTransmitPower */
#else
  0,
#endif
```

## Switch UART0 from P0.2/P0.3 to P1.4/P1.5

### HAL\Target\CC2530ZNP\Drivers\hal_uart.c -> _hal_uart_dma.c -> HalUARTInitDMA()
Comment all but the line setting bit 0 of PERCFG to 1 in the block below:
```
#if (HAL_UART_DMA == 1)
  PERCFG &= ~HAL_UART_PERCFG_BIT;
#else
  PERCFG |= HAL_UART_PERCFG_BIT;    // <--Only keep this line!!!
#endif
```

whc2001, 2020-10-21:
Tested the 2G4Z27SI along with the CC2531 USB stick with MiJia pushbutton, it appears that the 2G4Z27SI does not have a significant improvement regarding to signal reception distance (even a little worse in some cases). I wonder if it's because the LNA is not working correctly.

whc2001, 2020-11-23:
Contacted EByte again and they still does not want to disclose any design detail about E800. Gonna buy another one and do some testing with stock firmware. Now the only thing that's uncertain is about the relation of RS485 direction and RUN LED.

whc2001, 2020-11-25:
The new unit arrived, and sure enough, the RUN LED blinks when the unit is sending data to the host. Just updated the ZStack code modify tutorial for RS485 direction switch. Gonna rip the new E800 apart and flash the firmware later.

whc2001, 2020-11-29:
Measured RUN_LED and TX pin on original firmware. Not sure why the logic seems to be inverted. One useful information is that seems like the RUN_LED pin toggles ~5.5ms before and after transmitting. Modifying the ZStack code corresponding to that.

[image]

whc2001, 2020-11-30:
Note for myself: 7P header from daughter board to main board:
```
MODULE_BAUD_RST_BTN P1.7
/MODULE_NWK_LED P1.2（H=In Network L=No Network）
MODULE_RX P1.4
MODULE_TX P1.5
/MODULE_RUN_LED P1.3（H=Normal L=Error / 485 Transmitting）
VCC
GND
```

whc2001, 2020-12-01:
\ Seems like it worked!!! /

```
Zigbee2MQTT:info  2020-12-01 14:31:01: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-12-01.14-31-01' filename: log.txt
Zigbee2MQTT:info  2020-12-01 14:31:02: Starting Zigbee2MQTT version 1.14.4 (commit #654817a)
Zigbee2MQTT:info  2020-12-01 14:31:02: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-12-01 14:31:06: zigbee-herdsman started
Zigbee2MQTT:info  2020-12-01 14:31:06: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2020-12-01 14:31:06: Currently 1 devices are joined:
Zigbee2MQTT:info  2020-12-01 14:31:06: 0x00158d00026952e8 (0x00158d00026952e8): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
Zigbee2MQTT:info  2020-12-01 14:31:06: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2020-12-01 14:31:06: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:info  2020-12-01 14:31:06: Connected to MQTT server
Zigbee2MQTT:info  2020-12-01 14:31:06: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2020-12-01 14:31:06: MQTT publish: topic 'zigbee2mqtt/0x00158d00026952e8', payload '{"battery":100,"linkquality":106,"voltage":3022}'
Zigbee2MQTT:info  2020-12-01 14:31:06: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"654817a","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.14.4"}'
Zigbee2MQTT:info  2020-12-01 14:31:10: MQTT publish: topic 'zigbee2mqtt/0x00158d00026952e8', payload '{"action":"single","battery":100,"linkquality":135,"voltage":3022}'
Zigbee2MQTT:info  2020-12-01 14:31:10: MQTT publish: topic 'zigbee2mqtt/0x00158d00026952e8', payload '{"battery":100,"click":"single","linkquality":135,"voltage":3022}'
Zigbee2MQTT:info  2020-12-01 14:31:12: MQTT publish: topic 'zigbee2mqtt/0x00158d00026952e8', payload '{"action":"double","battery":100,"linkquality":130,"voltage":3022}'
Zigbee2MQTT:info  2020-12-01 14:31:12: MQTT publish: topic 'zigbee2mqtt/0x00158d00026952e8', payload '{"battery":100,"click":"double","linkquality":130,"voltage":3022}'
Zigbee2MQTT:info  2020-12-01 14:31:16: MQTT publish: topic 'zigbee2mqtt/0x00158d00026952e8', payload '{"action":"triple","battery":100,"linkquality":135,"voltage":3022}'
Zigbee2MQTT:info  2020-12-01 14:31:16: MQTT publish: topic 'zigbee2mqtt/0x00158d00026952e8', payload '{"battery":100,"click":"triple","linkquality":135,"voltage":3022}'
```

Some problems:
    1. LNK_LED does not work (for stock firmware it dims when joined into a network, not sure how to implement)
    2. UART_BAUD_RESET button does not work and pushing it would cause unknown effect (do ZStack and zigbee2mqtt really use button for anything?)
    3. Seems like if the E800 is accidentally power cycled while zigbee2mqtt is running, it will not connect back to devices and all operation of devices have no effect until you restart zigbee2mqtt (is it designed to be that, or are there some problem with the firmware?)

I am aware that this issue is kept open for way too long, so I'm gonna close it for now, and later refine the writeup of firmware modify steps a bit and put it along the compiled firmware into a new repository. Also I think tests are still needed, since I only have the E800 and a MiJia pushbutton. Anyone have this please try it if you'd like, and look for any problems. Thanks everyone for the supporting process.

closed 2020-12-01 by whc2001

whc2001, 2021-02-05:
Anyone willing to test it here is the compiled HEX file: https://github.com/whc2001/zigbee2mqtt_E800