Koenkk/zigbee2mqtt #4251: Light entity missing in HASS, other entities exist

syphernl, 2020-09-03:
# Bug Report
## What happened
I have a bunch of Tradfri lights in my Z2M network. One of those lights suddenly disappeared from HomeAssistant (not unavailable, just completely gone). The `linkquality` and `update_available` sensors are there. It still shows up on the Zigbee Map. 

I have deleted it from the network and paired it again, however the light still doesn't show up. 

During re-pairing it seems to have send the correct config details for the bulb for auto-discovery in HASS:
```json
info  2020-09-03 09:11:28: MQTT publish: topic 'homeassistant/light/0x14b457fffe77f4b8/light/config', payload '{"availability_topic":"zigbee2mqtt/lamp_overloop/availability","brightness":true,"brightness_scale":25
4,"color_temp":true,"command_topic":"zigbee2mqtt/lamp_overloop/set","device":{"identifiers":["zigbee2mqtt_0x14b457fffe77f4b8"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (LED1732G11)","name":"lamp_overloop","sw_version":"Zigbee2MQTT 1.14.4"},"json_attributes_topic":"zigbee2mqtt/lamp_overloop","name":"lamp_overloop_light","schema":"json","state_topic":"zigbee2mqtt/lamp_overloop","unique_id":"0x14b457fffe77f4b8_light_zigbee2mqtt"}'
```
If I send `off` to `zigbee2mqtt/lamp_overloop/set` the light turns off properly.
There are no errors/warnings in the logs of HA.

## What did you expect to happen
The light to show up in HomeAssistant

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: 20190619

syphernl, 2020-09-03:
I fixed it by removing any references to `0x14b457fffe77f4b8` in files within the `.storage` folder of HASS. After this the light was properly discovered again. ISTM this might be a HASS bug, although very hard / impossible to reproduce at this time.

closed 2020-09-03 by syphernl

rodsan0, 2021-07-06:
Hi!

I replicated this by renaming the `friendly_name` of my Tradfri lights on z2m's config.

G5K-org, 2024-05-16:
I could also replicate it by renaming one of my AwoX Bulbs. I fixed it by removing `core.entity_registry` in 
`~/.homeassistant/.storage`

Stephan1908, 2024-11-12:
Hello, I have the same problem,
- do I have to delete the whole file (core.entity_registry)?
- do I have to restart HA after deleting it?

g-regor, 2024-11-24:
> Hello, I have the same problem,
> 
> * do I have to delete the whole file (core.entity_registry)?
> * do I have to restart HA after deleting it?

Maybe you are using too old version of Home Assistant?

I am running Home Assistant 2023.7.3 and since zigbee2mqtt 1.33.0, 2023.8.0 is required (and 2024.4. since 1.37.0).

I was somehow running fine with 1.40.0, until MQTT server was reset. Probably because of the still retained messages.  
I had to go back to 1.32.2 to make it work.