Koenkk/zigbee2mqtt #4253: Phantom clicks and events from Aqara switches

SeTeM, 2020-09-03:
## What happened
Switch: Aqara QBKG11LM (lumi.ctrl_ln1.aq1) and Aqara QBKG12LM (lumi.ctrl_ln2.aq1).
zigbee2mqtt config & HA mqtt config: https://gist.github.com/SeTeM/47eddf5d4a636b88d36d1429974ec585

I have a lot of Aqara switches and all of them send clicks and actions every 5 minutes, but actually no one clicks on them.

HA clicks|mqtt clics|HA actions|mqtt actions
---|----|----|----
[image]|[image]|[image]|[image]

## What did you expect to happen
Not to spam with phantom events.
Maybe this behavior caused by any z2m configuration option, but I didn't find anything related to that in the docs.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2538
Coordinator version: 20200211

Koenkk, 2020-09-04:
Can you catch a debug log of the moment this happens?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

SeTeM, 2020-09-18:
@Koenkk, sorry for the delay, I was on vacation.

Here is a log with all rows related to one of my switches started from the restart of zigbee2mqtt:
https://gist.github.com/SeTeM/47eddf5d4a636b88d36d1429974ec585#file-switch-office_ceiling_light_debug-log

The switch is off. Messages at 2020-09-18 14:37:37 and 2020-09-18 14:42:37 look the same for me:
```
debug 2020-09-18 14:37:37: Received Zigbee message from 'switch.office_ceiling_light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0

debug 2020-09-18 14:42:37: Received Zigbee message from 'switch.office_ceiling_light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
```

Koenkk, 2020-09-18:
This is strange! `Received Zigbee message from` means that the device actually send a Zigbee message saying it was clicked. Would you be able to sniff the traffic of this? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2020-10-26 by github-actions[bot]

Skyman64, 2022-12-01:
One thing that I have noticed with these mini switches is that they can report information aside from a person doing a click as an action like the temperature (yes, they have temp sensors). So if you are acting on just an “action” with no detail, you may get activity triggered by a temp change or other sensor actionand as far as most people are concerned, this looks like a phantom click. If you instead trigger specifically on a single click, a double click, or a hold+release as details for the action, then the apparent phantoms should disappear.