Koenkk/zigbee2mqtt #4254: Adding Immax 4in1 Sensor Z3.0

Nithador, 2020-09-03:
Hi guys,
I tried to add support for the Immax motion 4in1 sensor  it should support occupancy, temperature, relative humidity and lux, but I am unable to make it work.
I followed the guide for adding devices and added following in the devices.js:

    {
        zigbeeModel: ['4in1-Sensor-ZB3.0'],
        model: '07047L',
        vendor: 'Immax',
        description: 'Intelligent motion sensor',
        supports: 'occupancy, temperature, illuminance, humidity',
        fromZigbee: [
	    fz.ias_occupancy_alarm_1_with_timeout,
            fz.battery, fz.temperature, fz.illuminance, fz.humidity, fz.ignore_iaszone_report,
        ],
        toZigbee: [],
	meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
	    const binds = ['msTemperatureMeasurement', 'msRelativeHumidity','msIlluminanceMeasurement'];
            await bind(endpoint, coordinatorEndpoint, binds);
            await configureReporting.temperature(endpoint);
            await configureReporting.humidity(endpoint); 
            await configureReporting.illuminance(endpoint); 
        },
    },	

and when I wave in front of the sensor I get this in the logs: https://pastebin.com/Pqfw6YCP

When I listen to zigbee2mqtt/Sensor 1 it gives

Message 10 received on zigbee2mqtt/Sensor 1 at 2:12 PM:

{
    "battery_low": false,
    "humidity": 60.82,
    "linkquality": 89,
    "occupancy": true,
    "tamper": false,
    "temperature": 24.16
}

So I believe that it can handle everything except lux, however the device and entities does not appear in hass.io.
When I tried publishing a packet zigbee2mqtt/Sensor 1/get with payload {"temperature": ""} I got No converter available for 'temperature' () in the logs and it is the same for occupancy and humidity.

In the devices.js I tried different settings and basically I got a feeling it behaves the same when it was configured like this (removing all the converters besides fz.ias_occupancy_alarm_1_with_timeout, even fz.ias_occupancy_alarm_1 works, but I think the device has a timeout):

    {
        zigbeeModel: ['4in1-Sensor-ZB3.0'],
        model: '07047L',
        vendor: 'Immax',
        description: 'Intelligent motion sensor',
        supports: 'occupancy, temperature, illuminance, humidity',
        fromZigbee: [
	    fz.ias_occupancy_alarm_1_with_timeout
        ],
        toZigbee: [],
    },
	
Do You know how to proceed from here?

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: Z-Stack_Home_1.2

Nithador, 2020-09-03:
Also a link to the devices. Not much info there though.
https://www.immaxneo.cz/en/products/security/immax-neo-smart-sensor-4v1/

Koenkk, 2020-09-04:
- Can you share the `data/database.db`?
- What did you add to `homeassistant.js`?

Nithador, 2020-09-04:
In `database.db` I found these records regarding the coordinator and the sensor (or should I post the whole thing?):
https://pastebin.com/jKK9MvT1

In `homeassistant.js` I had
'07047L': [cfg.binary_sensor_occupancy, cfg.sensor_temperature, cfg.sensor_humidity, cfg.sensor_illuminance, cfg.binary_sensor_baterry_low],

Just found out `homeassistant.js`, it was rewritten by the update to the version 1.14.4.

Koenkk, 2020-09-04:
- Don't understand why the illuminance isn't working, according to db it should work with this config. Could it be that you made changes to the configure() method without bumping the `configureKey`? After every change to `configure()` you have to bump the `configureKey`, so `configureKey: 2`, in this case. You can also force it through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (look at the log if everything went OK, you may need to wakeup the device right before sending it).
- Don't worry about the HA discovery for now, once we add it as supported device this will be taken care of.

Nithador, 2020-09-04:
I bumped up `configureKey` to 2, ran configure() for the sensor and now it returns illuminance. Thank You.
https://pastebin.com/qEKXM9d4

Now I guess only things missing are pushing the changes, right?

Nithador, 2020-09-05:
By pushing I meant, should I create pull request on the `devices.js` in [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters)? And another pull request for the `homeassistant.js` in this repo?

Koenkk, 2020-09-05:
Yes exactly, could you do that?

Nithador, 2020-09-05:
I did that. Just not sure if I've done it right for the zigbee2mqtt repo, since I made a pull request to master instead of dev branch.

Koenkk, 2020-09-05:
Merged the prs, thanks! Assuming this can be closed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-05 by Koenkk