Koenkk/zigbee2mqtt #4255: Failed to execute LQI for <Gledopto devices>

MaZZly, 2020-09-03:
# Bug Report
## What happened

Every 3h in the zigbee2mqqt logs there is the following lines logged:
```
2020-08-31.20-00-39/log.txt:error 2020-09-03 15:00:19: Failed to execute LQI for 'ledstrip.room1'
2020-08-31.20-00-39/log.txt:error 2020-09-03 15:00:19: Failed to execute LQI for 'ledstrip.room2'
2020-08-31.20-00-39/log.txt:error 2020-09-03 15:00:19: Failed to execute LQI for 'ledstrip.projector'
2020-08-31.20-00-39/log.txt:error 2020-09-03 15:00:19: Failed to execute LQI for 'light.corridoor3'
2020-08-31.20-00-39/log.txt:error 2020-09-03 15:00:19: Failed to execute LQI for 'light.corridoor1'
2020-08-31.20-00-39/log.txt:error 2020-09-03 15:00:19: Failed to execute LQI for 'light.corridoor2'
```
No other devices in the network (~80 devices total) log anything like that.

I realized that all of the above devices are "Gledopto"-brand devices:
- cooridoor lights are https://www.zigbee2mqtt.io/devices/GL-D-005Z.html
- projector ledstrip is https://www.zigbee2mqtt.io/devices/GL-C-008-2ID.html
- 2 room ledstrips are https://www.zigbee2mqtt.io/devices/GL-MC-001.html

## What did you expect to happen
Not to see this in the logs :+1: 

## How to reproduce it (minimal and precise)

Having a Gledopto device in the network I'm guessing.

## Debug Info
Zigbee2MQTT version: edge
Adapter hardware: CC1352R1
Adapter firmware version: Z-Stack_3.x.0 from 20200417

nanobra1n, 2020-09-03:
I see this as well for my Geldopto devices, though for some others as well (not as often). This is usually preceded by
`Starting network scan (includeRoutes 'false')` which I presume is initiated by Zigbee2mqttAssistant.

MaZZly, 2020-09-05:
Hmm you might be onto something here as I also have Zigbee2mqttAssistant installed.

I tried clicking the "Launch new network scan" and after a little while, I saw the same messages come up in the log with the current time.

Don't the Gledopto devices support network scans? if so, those devices should maybe be configured so that the network scan ignores them?

lpuzzello, 2020-09-08:
I'm facing the same issue with 4 Gledopto bulbs (GL-B-008Z) and one Led strip Controller (GL-C-008) . I saw others threads ( not related to z2m) and even on Gledopto reddit, that this might be a hardware limitation of these products that are not expected to work as a router. Which is a shame.

closed 2020-10-17 by github-actions[bot]

kevincaradant, 2020-11-06:
Hi guys, 

I had a lot of GLEDOPTO products but I'm wondering why ZigBee2MQTT doesn't have the LQI signal for these products ? 
I got this issue since the begin 1.12.0 but I decided to create an issue only today (missing time before)

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-06 22:33:39: MQTT publish: topic 'zigbee2mqtt/gledopto_1_couloir', payload '{"brightness":111,"color":{"x":0.313059507478705,"y":0.323172994053855},"color_temp":153,"state":"ON"}'
```
At the starting of Zigbee2MQTT, I got `ERROR: Failed to execute LQI for...` for each GLEDOPTO devices.
It's not random, in my case I had always this answer.

Under zigbee2mqttAssistant I have this attribute with a defined value, so how is it possible to have this value under Zigbee2MQTTAssistant but not under Zigbee2MQTT ? Under Home Assistant I didn't have any value as Zigbee2MQTT doesn't have it.

Is it the same issue as you ?

Thank you in advance

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC1352P-2 and CC2531
Adapter firmware version: 20201026

robertcsakany, 2020-11-23:
I have the very same issue, but it is on Xiaomi bulbs too. Is there any posibility that all device template containing info about LQI scan?

jmgutierrezm, 2021-03-01:
I have the same issue with the Gledopto GL-B-008Z bulbs

mihing, 2021-09-18:
Same issue with the Gledopto GL-C-009 led dimmer.

Koenkk, 2021-09-18:
Most gledopto devices don't support the LQI command, so not something that can be fixed from z2m.