Koenkk/zigbee2mqtt #4256: Failed to interview '0x00124b000923fac0', device has not successfully been paired

Tobias007, 2020-09-03:
Hi, I tried to pair the following Lonsonho device:

https://www.aliexpress.com/item/4001160044091.html?spm=a2g0s.9042311.0.0.27424c4dILnY0W

But I always get 'device has not successfully been paired'...

here the log:

https://hastebin.com/poruhoqaqo.sql

Thanks a lot in advance!

Tobias

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-09-04:
Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails

Tobias007, 2020-09-05:
Hi Koen,

yep, I checked those points already (CC2531 is connected via extension cable), tried several times. Batteries changed (actually it were brand new ones). I get always this:

Zigbee2MQTT:error 2020-09-05 12:20:42: Failed to interview '0x00124b000923fac0', device has not successfully been paired
Zigbee2MQTT:info  2020-09-05 12:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:info  2020-09-05 12:20:48: Device '0x00124b000923fac0' joined
Zigbee2MQTT:info  2020-09-05 12:20:48: Starting interview of '0x00124b000923fac0'
Zigbee2MQTT:info  2020-09-05 12:20:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b000923fac0"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-09-05 12:20:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:info  2020-09-05 12:20:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b000923fac0"},"type":"device_announced"}'
Zigbee2MQTT:error 2020-09-05 12:21:10: Failed to interview '0x00124b000923fac0', device has not successfully been paired
Zigbee2MQTT:info  2020-09-05 12:21:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:warn  2020-09-05 12:21:17: Received message from unsupported device with Zigbee model 'undefined'
Zigbee2MQTT:warn  2020-09-05 12:21:17: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

cheers, Tobias

Koenkk, 2020-09-05:
It could be that the device falls asleep during, pairing, can you try to keep it awake by pressing a button on it (every second) until pairing is completed?

Tobias007, 2020-09-06:
Tried it. But the pairing seems still to fail. What I have observed:  

The device (a PIR motion sensor) seems to somehow have "checked in" to the network
a) After a failed pairing - if I press the button for 10sec - I blinks three times - then the log states: Device '0x00124b000923fac0' left the network
b) When I press the button again 10sec: It blinks 3 times and then it blinks 3 times fast - then the pairing process seems to start: However after 20sec - the pairing failed message appears again. 
c) again: after 10sec pressing button - "it has left the network" ... so it seems it was somehow "half" paired?

Zigbee2MQTT:warn  2020-09-06 08:15:20: Device '0x00124b000923fac0' left the network
Zigbee2MQTT:info  2020-09-06 08:15:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00124b000923fac0"},"type":"device_removed"}'
Zigbee2MQTT:info  2020-09-06 08:15:43: Device '0x00124b000923fac0' joined
Zigbee2MQTT:info  2020-09-06 08:15:43: Starting interview of '0x00124b000923fac0'
Zigbee2MQTT:info  2020-09-06 08:15:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b000923fac0"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-09-06 08:15:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:info  2020-09-06 08:15:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b000923fac0"},"type":"device_announced"}'
Zigbee2MQTT:warn  2020-09-06 08:15:51: Device '0x00124b000923fac0' left the network
Zigbee2MQTT:info  2020-09-06 08:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00124b000923fac0"},"type":"device_removed"}'
Zigbee2MQTT:error 2020-09-06 08:16:05: Failed to interview '0x00124b000923fac0', device has not successfully been paired
Zigbee2MQTT:info  2020-09-06 08:16:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:info  2020-09-06 08:16:33: Device '0x00124b000923fac0' joined
Zigbee2MQTT:info  2020-09-06 08:16:33: Starting interview of '0x00124b000923fac0'
Zigbee2MQTT:info  2020-09-06 08:16:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b000923fac0"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-09-06 08:16:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:info  2020-09-06 08:16:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b000923fac0"},"type":"device_announced"}'
Zigbee2MQTT:warn  2020-09-06 08:16:39: Device '0x00124b000923fac0' left the network
Zigbee2MQTT:info  2020-09-06 08:16:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00124b000923fac0"},"type":"device_removed"}'
Zigbee2MQTT:info  2020-09-06 08:16:52: Device '0x00124b000923fac0' joined
Zigbee2MQTT:info  2020-09-06 08:16:52: Starting interview of '0x00124b000923fac0'
Zigbee2MQTT:info  2020-09-06 08:16:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b000923fac0"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-09-06 08:16:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:info  2020-09-06 08:16:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b000923fac0"},"type":"device_announced"}'
Zigbee2MQTT:error 2020-09-06 08:16:55: Failed to interview '0x00124b000923fac0', device has not successfully been paired
Zigbee2MQTT:info  2020-09-06 08:16:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:error 2020-09-06 08:17:14: Failed to interview '0x00124b000923fac0', device has not successfully been paired
Zigbee2MQTT:info  2020-09-06 08:17:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b000923fac0"},"type":"pairing"}'
Zigbee2MQTT:warn  2020-09-06 08:17:16: Received message from unsupported device with Zigbee model 'undefined'
Zigbee2MQTT:warn  2020-09-06 08:17:16: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-09-06 08:17:22: Received message from unsupported device with Zigbee model 'undefined'
Zig

I tried also combinations to press button in 1sec cycles again and again after pairing is initiated ... but I doesn't seem to make any difference ..

Any idea how to proceed?

Koenkk, 2020-09-06:
Could you create a sniff of the pairing? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Tobias007, 2020-09-13:
Hi, finally i got the sniffing thing working :). It's really a good documentation but I struggled more on the Ubuntu side to get it running in a VBox and to get the USB CC2531 working in it. 

It seems during pairing process the device is visible as 0x3db4.

However I couldn't figure anything out of the log ... really cryptic :o)

[Motion-Log.zip](https://github.com/Koenkk/zigbee2mqtt/files/5213634/Motion-Log.zip)

Tobias007, 2020-09-13:
Another observation: Even if the pairing seems to fail - the device is sending once motion is detected a data package to the CC2531. I have sniffed it and it seems to be some info available about the sensors state (however zigbee2mqtt  says: Zigbee2MQTT:warn  2020-09-13 13:32:51: Received message from unsupported device with Zigbee model 'undefined')

Attached the file with the data packages: 

[motion_detected.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5213649/motion_detected.pcapng.zip)

Koenkk, 2020-09-13:
@Tobias007 please send your network key to me on telegram (@koenkk) Otherwise I cannot decrypt the sniff.

Koenkk, 2020-09-14:
@Tobias007 I've checked your sniff and something strange is going on. Can you provide me the herdsman debug logging of the pairing?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Tobias007, 2020-09-15:
@Koenkk Here you go:

[herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/5221827/herdsman.log)

Koenkk, 2020-09-15:
The problem is that the motion sensor stops doing "Data request" to the coordinator very shortly after joining. These are required to send queries to the motion sensor. Because of this interview fails. 

I'm not sure if we can get this device to be supported, sniffing traffic when pairing it with the original gateway may reveal more info on how we can do this.

Koenkk, 2020-09-18:
@Tobias007 no problem, thats always a safer choice! :) I assume this can be closed now.

closed 2020-09-18 by Koenkk