Koenkk/zigbee2mqtt #4268: HEIMAN HS2SK Smart metering plug Interview failed because can not get node descriptor

Stormweaker, 2020-09-04:
# Bug Report
## What happened

Adding [HEIMAN HS2SK](https://www.zigbee2mqtt.io/devices/HS2SK.html) devices but interview fails.

## What did you expect to happen

Device to connect without problems.

## How to reproduce it (minimal and precise)

Long press power button of the device until blinking to start pairing.
Wait.

## Debug Info
Zigbee2MQTT version: 1.14.4 (commit #654817a)
Adapter hardware: CC2531
Adapter firmware version: 3.2.6
Log with Zigbee-herdsman debug logging : https://pastebin.com/SQBfz268
A bit of info for the log : I have 5 Aqara sensors working. I try to pair two different plugs in this log: the first plug is part of the "joined devices" but "not recognised", the second plug is not in the joined devices.

## FAQ advice on interview fail
> Try pairing the device closer to the coordinator.

Coordinator is ~3m away from the first device, ~1 m away from the second device

> There can be too much interference, try connecting the coordinator USB through an USB extension cable. This problem occurs a lot when used in combination with a Raspberry Pi 4.

Using a 1 m cable.

> If it’s a battery powered device, try replacing the battery.

Not battery powered.

> Try repairing the device again for 2 or 3 times.

Tried a few times more.

Koenkk, 2020-09-04:
> Adapter firmware version: 3.2.6

This is not a firmware version, it should be one of https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin , which one are you running?

Stormweaker, 2020-09-04:
Sorry, it's CC2531_DEFAULT_20190608.

Koenkk, 2020-09-05:
@Stormweaker I'm not sure if it will fix it, but can you try to reflash your stick? (wont' require repairing of other devices). Curious if this fixes the problem.

Stormweaker, 2020-09-05:
Well that may have fixed the issue.
Before seeing your message I paired an IKEA light bulb (with no issues).
Then I reflashed the stick with CC2531_DEFAULT_20190608.
After a restart the IKEA bulb would not respond, 'No network route' (205) error.
I reflashed the stick with CC2531_SOURCE_ROUTING_20190619.zip.
Now the IKEA bulb is responsive and I managed to pair the two meter plugs. 

So I don't know if just reflashing would have fixed the issue but with the source_routing firmware it works great.

I'm going to pair my other devices, wait a few days to see if I have any similar issue and close this one if I don't.

Thanks a lot.

Stormweaker, 2020-09-15:
No issue to report on that, I close the issue.

closed 2020-09-15 by Stormweaker