Koenkk/zigbee2mqtt #4271: Massive pairing problems

guardiande, 2020-09-05:
# Bug Report
## What happened

### Problem case 1

I needed to re-pair some Xiaomi devices since I removed the router that they we're attached to and Xiaomi devices do not seem to be able to reassociate themselves. Doh.

I was not able to re-pair the devices and was not able to pair any other device. I tried everything to make it work:

* Reset the CC2531.
* Restart the Raspberry.
* Plug the CC2531 into different USB ports.
* Apply several extension cables.
* Starting with a fresh configuration.

In the end this is the procedure that worked:

* Stop Z2M.
* Save the current config.
* Change the network key.
* Start Z2M (all devices gone obviously).
* Stop Z2M.
* Restore config.
* Start Z2M (no devices found).
* Stop Z2M.
* Restore config.
* Start Z2M (old devices available).

After that pairing works most of the time.

For the other cases it does not work for I applied the following procedure:

* Setup an extra Raspberry Pi with its own CC2531.
* Copy over the installation from the primary device.
* Stop the primary device.
* Clear the configuration on the secondary.
* Start Z2M on secondary (empty database).
* Pair the device.
* Stop secondary.
* Copy the newly created entry from database.db to the primary database.
* Start the primary.

Device should work afterwards.

**Question 1**: Any idea why this needs to be done this way?

### Problem case 2

Devices seem to drop off the network.

I paired an OSRAM smart plug+ the way described above with a secondary installation. I moved it over to the primary setup and it seemed to work for a while (a few hours). Then I noticed it did not send alive messages for a long time (several hours).

I re-paired the device and had to resort to the secondary device method again. In the process of that I also paired another OSRAM smart plug+. Both devices paired successfully and worked on the secondary system.

After I moved the database entries over to the primary device only one of the devices worked.

**Question 2**: Is there anything generally wrong with the secondary device procedure? Is there any reason that should not work?

**Bonus questions**:

* How can I find out the parent for a battery powered device (battery powered devices to seem to answer to requests for creating the network map)?
* How can I find out whether a CC2531 coordinator does not have room for further paired devices? Is it just that the pairing does not start or fails?

## What did you expect to happen

Devices to be successfully paired.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: 20190619

guardiande, 2020-09-05:
**Update for second problem case**

Now the second OSRAM plug works as expected. I seemed to have taken some time for Zigbee to settle.

Koenkk, 2020-09-05:
> Problem case 1

This is indeed a known issue, most of the times it helps to just reflash the CC2531 (should provide the same effect as what you did now). Some improvements have been done in #3177, you can find an updated firmware in that thread.

> How can I find out the parent for a battery powered device (battery powered devices to seem to answer to requests for creating the network map)?

End devices don't answer network map requests, their parent (Router) does. You should be able to see this in the network map, but results may vary for Xiaomi devices as some routers will kick them out of their child table because they sleep for too long. (This does not mean they are disconnected from the network!)

> How can I find out whether a CC2531 coordinator does not have room for further paired devices? Is it just that the pairing does not start or fails?

You can see al direct connections in the network map, however note that other routers should still be able to join other devices into the network.

guardiande, 2020-09-05:
> You can see al direct connections in the network map, however note that other routers should still be able to join other devices into the network.

That's interesting. All my directly to the coordinator paired Xiaomi devices show as disconnected. Is this expected?

[image]

Koenkk, 2020-09-05:
Unfortunately yes, below also goes for coordinator.

> You should be able to see this in the network map, but results may vary for Xiaomi devices as some routers will kick them out of their child table because they sleep for too long. (This does not mean they are disconnected from the network!)

closed 2020-10-14 by github-actions[bot]