Koenkk/zigbee2mqtt #4272: OTA update's log issue.

GuGu927, 2020-09-05:
# Bug Report
## What happened
1. when "Update available" log happen
zigbee2mqtt/bridge/log {"message":"Update available for 'dimmer_livingroom'","meta":{"device":"dimmer_livingroom","status":"available"},"type":"ota_update"}

2. when do ota_update
zigbee2mqtt/bridge/log {"meta":{"device":"dimmer_livingroom","status":"update_in_progress"},"msg":"Updating 'dimmer_livingroom' to latest firmware","type":"ota_update"}

## What did you expect to happen
1 payload with "message" and
2 payload with "msg"
They are mismatch.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version:1.14.4
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-09-05:
Issues in the legacy api will not be fixed anymore, please use the new api (https://github.com/Koenkk/zigbee2mqtt/issues/3281)

closed 2020-09-05 by Koenkk

GuGu927, 2020-09-06:
@Koenkk Thx for your support.
There's a problem using new api with HA docker version.(https://github.com/danielwelch/hassio-zigbee2mqtt)
When it be possible?

Koenkk, 2020-09-06:
@GuGu927 please open an issue here to support that option: https://github.com/danielwelch/hassio-zigbee2mqtt/issues