Koenkk/zigbee2mqtt #4273: E1744 too much delay with mqtt updates

ale1800, 2020-09-05:
Hi everyone. I have an ikea E1744 remote correctly recognized from zigbee2mqtt. The single and double press are almost ok (about 1,5 sec from press to mqtt update). The problem is with right and left rotation. There is a lot of delay. If I start rotating and then stop (during rotation nothing happens), after 5 or 6 seconds zigbee2mqtt receives mqtt updates about rotation all at once so is impossible to understand something. I tried re-pairing 2 or 3 times and also changed the firmware of cc2531 but no luck. Zigbee2mqtt is on his latest version. How can I reduce latency to have "live" rotation mqtt updates ?

Thank you all!

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: CC2531_SOURCE_ROUTING_20190619 and CC2531_DEFAULT_20190608

Koenkk, 2020-09-05:
Did you disable the legacy intergration? (https://www.zigbee2mqtt.io/devices/E1744.html#legacy-integration)

ale1800, 2020-09-05:
> Did you disable the legacy intergration? (https://www.zigbee2mqtt.io/devices/E1744.html#legacy-integration)

Thanks for answering. Yes i disabled Legacy but nothing changed

Koenkk, 2020-09-05:
The debug logging should show you when the remote send zigbee commands (you will see `Received zigbee message...`). Do you also see the delay there?

ale1800, 2020-09-05:
Yes. I Just tried and there's the delay

ale1800, 2020-09-05:
If you want i can upload a zip file with a video of the problem. maybe it's easier

Koenkk, 2020-09-05:
Yes, video would help (showing both the E1744 where actions are executed on and the z2m logging).

Please also attach the logging as a txt.

ale1800, 2020-09-06:
[video.zip](https://github.com/Koenkk/zigbee2mqtt/files/5178932/video.zip)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5178928/log.txt)
As you can see from the txt and the video , updates of rotation are showed all at once at the same time. After the single press update I immediately  rotate but updates are showed after 7/8 seconds

Koenkk, 2020-09-07:
There is indeed a huge delay, looks that the delay comes from the network.

If you could sniff the traffic (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) you should be able to find out where the delay comes from (e.g. remote itself or any routers).

ale1800, 2020-09-07:
I have only one cc2531 stick. Maybe I might need another one.  However i have 4 zigbee devices, only this one with rotation gives me problem. Maybe is a defective remote

Koenkk, 2020-09-07:
@ale1800 I've noticed that you are holding the device in air, have you tried mounting it? I believe the device will not send any updates until it stopped moving.

ale1800, 2020-09-07:
Just tried but no differences

closed 2020-10-15 by github-actions[bot]