Koenkk/zigbee2mqtt #4282: Conbee2 stopped working. Silent: nothing obvious in logs

VincentSC, 2020-09-06:
# Bug Report
## What happened

Context. After lots of troubles (Murphy moved in and I kept getting obstructions like broken cables and other simply stupid stuff - out of scope, but demotivated me to work on smarthome for a few months), I replaced the (probably broken) CC2531 USB stick with the Conbee II I initially bought for smarthome. It worked for a few hours, where I could re-pair all the devices and get the old config running as it used to (node-red). I had started it (npm start) manually, and left ssh open. A few hours of not using it, all zigbee communication seemed to just have stopped working. I was not sure it was a broken ssh pipeline that froze z2m and caused the trouble, so eventually I restarted the device and started it cleanly (just like I did when it still worked), including unplugging the adapter to make sure power was really off. But still the same. No response. 

When I press a zigbee-button there was no response; really nothing shows up in the log. When I pressed a nodered-button mosquitto registers it but z2m did not always receive it (and the few it did, it took well over a minute). The mqtt client showed it immediately (messages on 'retain'). 

When I request a device list, it shows up. So the coordinator and the network (link quality 50-70) just works. 

## What did you expect to happen

## How to reproduce it (minimal and precise)

I need help to debug this. I have no idea where to start looking.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: Conbee2
Adapter firmware version {"maintrel":0,"majorrel":38,"minorrel":73,"product":0,"revision":"0x26490700","transportrev":0

Koenkk, 2020-09-06:
Please provide the herdsman debug logging so @ChrisHae can look further into this.

VincentSC, 2020-09-06:
Thanks.

Interactions with short pauses inbetween:
- pressed some button on Sunricher 4-buttons
- sent some value to Sunricher dimmer via nodered->mosquito
- sent some value to Hornbach Led via nodered->mosquito
- pressed some button on Sunricher 4-buttons again

It looks perfectly normal, but really nothing happens. Also the logs for the third interaction show up last.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5180040/log.txt)

VincentSC, 2020-09-06:
Update: put stick in laptop, installed DeCONZ and upgraded firmware to [latest](http://deconz.dresden-elektronik.de/deconz-firmware/). Now getting errors like
>	Sep 06 19:27:07 raspberrypi npm[3033]: Zigbee2MQTT:error 2020-09-06 19:27:07: Failed to execute LQI for '0x00158d0003910cf5'

Think I'm further from home now....

VincentSC, 2020-09-06:
Maybe a hint?

```
Sep 06 19:42:43 raspberrypi npm[664]: Zigbee2MQTT:debug 2020-09-06 19:42:43: TypeError: Read 0x00158d000390a7d7/1 genLevelCtrl(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
Sep 06 19:42:43 raspberrypi npm[664]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:221:45)
Sep 06 19:42:43 raspberrypi npm[664]:     at Generator.next (<anonymous>)
Sep 06 19:42:43 raspberrypi npm[664]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
Sep 06 19:42:43 raspberrypi npm[664]:     at runNextTicks (internal/process/task_queues.js:62:5)
Sep 06 19:42:43 raspberrypi npm[664]:     at processTimers (internal/timers.js:489:9)
Sep 06 19:42:44 raspberrypi npm[664]: Zigbee2MQTT:info  2020-09-06 19:42:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x00158d000390a7d7' failed: 'TypeError: Read 0x00158d000390a7d7/1 genLevelCtrl([\"currentLevel\"], {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Cannot read property 'frame' of null)'","meta":{"friendly_name":"0x00158d000390a7d7"},"type":"zigbee_publish_error"}'
Sep 06 19:47:31 raspberrypi npm[664]: Zigbee2MQTT:debug 2020-09-06 19:47:31: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 06 19:52:31 raspberrypi npm[664]: Zigbee2MQTT:debug 2020-09-06 19:52:31: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 06 19:52:32 raspberrypi npm[664]: Zigbee2MQTT:debug 2020-09-06 19:52:32: Received MQTT message on 'zigbee2mqtt/0xccccccfffee2dab1/set' with data '{"state":"ON"}'
Sep 06 19:52:32 raspberrypi npm[664]: Zigbee2MQTT:debug 2020-09-06 19:52:32: Publishing 'set' 'state' to '0xccccccfffee2dab1'
Sep 06 19:54:33 raspberrypi npm[664]: Zigbee2MQTT:error 2020-09-06 19:54:33: Publish 'set' 'state' to '0xccccccfffee2dab1' failed: 'TypeError: Read 0xccccccfffee2dab1/1 genLevelCtrl(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)'
Sep 06 19:54:33 raspberrypi npm[664]: Zigbee2MQTT:debug 2020-09-06 19:54:33: TypeError: Read 0xccccccfffee2dab1/1 genLevelCtrl(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
Sep 06 19:54:33 raspberrypi npm[664]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:221:45)
Sep 06 19:54:33 raspberrypi npm[664]:     at Generator.next (<anonymous>)
Sep 06 19:54:33 raspberrypi npm[664]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
Sep 06 19:54:33 raspberrypi npm[664]: Zigbee2MQTT:info  2020-09-06 19:54:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xccccccfffee2dab1' failed: 'TypeError: Read 0xccccccfffee2dab1/1 genLevelCtrl([\"currentLevel\"], {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Cannot read property 'frame' of null)'","meta":{"friendly_name":"0xccccccfffee2dab1"},"type":"zigbee_publish_error"}'
```

VincentSC, 2020-09-06:
I'm not managing to generate a network-map anymore (using Z2M Admin). Now getting `Failed to execute LQI...` all the time - even for devices very near. So I think there are 2 problems now. Z2M not working well with the latest firmware, and the communication being stalled.

ChrisHae, 2020-09-08:
I will have a look into this problem. Conbee2 FW should not be a problem - running fine here. 

Can you please enable full debug output with deconz adapter output:
Add -e DEBUG=zigbee-herdsman:* to your docker run command,
or add this to your docker-compose file:

```
    environment:
      - DEBUG=zigbee-herdsman:*
```

VincentSC, 2020-09-08:
The above log-file is generated using `DEBUG=zigbee-herdsman`. So I understand I should not provide the log-file, but the output to the screen? And if yes, how much time should I leave it running?

I'm doubting if I removed the data-dir when I changed the hardware. If I indeed forgot, could that be a problem? How to fix this the best way?

Chances are that the CB2-stick just broke. Could verify that by testing the system using its own software on my Ubuntu laptop.  Should I?

Yesterday evening the lights dimmed up and down for a while. Could it be that the Zigbee instructions finally arrived a day later? If so, how to see this?

ChrisHae, 2020-09-08:
When I use DEBUG=zigbee-herdsman:* (with: and *) then I see debug output from zigbee-hersdman e.g.: zigbee-herdsman:controller:log and also from deconz adapter e.g.: zigbee-herdsman:deconz:frameParser.

You can view logs by using following command: docker logs -f zigbee2mqtt or save to file by docker logs zigbee2mqtt> zigbee2mqtt.log 2>&1

I would like to see the logs from starting and pressing some buttons and perhaps pairing a device.

Perhaps you can also try to pair a device with the deCONZ Software. The easiest way is with the help of the phoscon app https://phoscon.de/de/app/doc (deCONZ must be running)

VincentSC, 2020-09-08:
Not sure if you follow me and I follow you. All answers are above. Made a typo, so I did add ":*". Are you saying the log-file is not right? I did see there was quite a lot of stuff not written to the log-file, but shown on the screen - possibly a bug that the log-file is incomplete? So my question again: should I provide the log-file or screen-output?

I'm *not* using docker. The info you provided is all in the documentation for how to create a herdsman-log.

The provided logs are exactly that. See the description. I did saw errors and weird stuff after a few hours, hence my question.

The question was *if* I should test it on my Linux PC, not "how". I cannot deduct from your sentence what you'd like me to do.

ChrisHae, 2020-09-08:
Yes test it on your Linux PC using deCONZ to check if the ConBee is working.
And provide the zigbee2mqtt Debug log with the output of the deconz adapter.

VincentSC, 2020-09-16:
Got tired of fiddling with the foundations - got a new zig-ah-zag-ah stick and that's just working (and fast!). Will later on test the stick, as it's quite new and is still under warranty. Will put my weekend-time into getting support for new devices instead, or to help out with documentation.

Will close for now. Thanks for the help!

closed 2020-09-16 by VincentSC