Koenkk/zigbee2mqtt #4283: devide sensors reporting status when they are offline

kasido, 2020-09-06:
Hi,
After set up Z2M with a cc2531 coordinator into home assistant and add 13 devices (xiaomi sensors all of them) I moved the raspeberry to its final location on the dinning room. I kept all the devices in the studio (out of range from the raspberry on the dinning room.
After home assistant restarted, all the sensors show the previous state, they don't show offline or unavailable,

It's seem to be a publish topic with all these data. I don't understand where this publish came from?

Is this normal, the devices will not show unvailable or offline?

[image]

sjorge, 2020-09-06:
Probably a combination of the `retain` device [property]( https://www.zigbee2mqtt.io/information/configuration.html#all-devices) or the `cache_state`, `cache_state_persistent`, and `cache_state_send_on_startup` [properties](https://www.zigbee2mqtt.io/information/configuration.html#configuration) under advanced.

kasido, 2020-09-06:
thanks, I forced cache_state, cache_state_persistent and cache_state_send_on_startup to true on the configuration panel por Z2M, but I can't find where the device properties are.
On the zigbee2mqtt.io website say that they should be on the configuration.yaml. Is this configuration.yaml the same for home assistant or is it a specific for Z2M? In that case, how can I find it??

sjorge, 2020-09-06:
I don't use z2m with home assistant myself, but AFAIK it should be a different configuration file.

domoticafacilconjota, 2020-09-08:
Hello there! It is under /share/zigbee2mqtt/devices.yaml at Home Assistant (false is default). You should remove the topic using, for example, MQTT Explorer too.

closed 2020-10-17 by github-actions[bot]