Koenkk/zigbee2mqtt #4285: Honeywell smoke detector (Xiaomi JTYJ-GD-01LM/BW) selftest fails with "No network route"

guardiande, 2020-09-06:
# Bug Report
## What happened

Redoing my whole setup I just recently paired the Honeywell smoke detector. I was paired directly to the CC2531 coordinator. To make sure the device works I pressed the button on the detector and Z2M received a normal payload message.

Then I decided it would be good to do a self test. That failed with the following error:

```
error 2020-09-06 12:41:00: Publish 'set' 'selftest' to 'SmokeDetector1' failed: 'Error: Write 0x00158d000215e015/1 ssIasZone({"65521":{"value":50397184,"type":35}}, {"timeout":35000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
info  2020-09-06 12:41:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'selftest' to 'SmokeDetector1' failed: 'Error: Write 0x00158d000215e015/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"SmokeDetector1"},"type":"zigbee_publish_error"}'
```

I read the other issues on the similar problems but they all refer to the device being connected to a router.

I once again had problems with pairing the device (see #4271). I had to resort to the secondary device and resetting the adapter configuration. This time I was not able to finish the interview sucessfully. I fixed the database to say interview successful true and the device seems to work with the exception of the selftest.

## What did you expect to happen

To hear a beep of the device after performing the selftest.

## How to reproduce it (minimal and precise)

See above.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: 20190619

Sandriekus, 2020-09-07:
Have the exact same issue after upgrading to 1.14.4.
Restored a backup to 1.14.3 and everything is working correct!
I'm using the slaesh stick

Koenkk, 2020-09-07:
@guardiande would you be able to sniff the traffic when you execute the command? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

@Sandriekus are you able to post the herdsman debug logging of this error on 1.14.4?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Enzokot, 2020-09-14:
@Koenkk can I help? I'm using the cc2538 stick

Koenkk, 2020-09-14:
@Enzokot yes please sniff the traffic like suggested in: https://github.com/Koenkk/zigbee2mqtt/issues/4285#issuecomment-688409270

Enzokot, 2020-09-15:
@Koenkk herdsman debug log?

Koenkk, 2020-09-15:
see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Enzokot, 2020-09-15:
@Koenkk 
0x00158d00039d9398 - without router
0x00158d0003943639 - with router
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5226586/log.txt)

Koenkk, 2020-09-15:
@Enzokot are you also able to provide a sniff? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

Enzokot, 2020-09-15:
> @Enzokot are you also able to provide a sniff? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

Yep, but later. Need to find one more stick

Enzokot, 2020-09-16:
@Koenkk sniff synchro with herdsman? at the same time?

Koenkk, 2020-09-16:
Preferably both.

Enzokot, 2020-09-17:
> Preferably both.

Done
0x00158d00039d9398 - without router
0x00158d0003943639 - with router
[logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/5240953/logs.zip)

Sandriekus, 2020-09-19:
sorry for the late response.
i can test also if you want.

current setup:
Zigbee2MQTT version: 1.14.4.1
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: CC2652RB_20200805

when performing a self-test to the HoneyWell Smoke detector:
`Zigbee2MQTT:error 2020-09-19 09:04:35: Publish 'set' 'selftest' to '0x00158d00052b7f72' failed: 'Error: Write 0x00158d00052b7f72/1 ssIasZone({"65521":{"value":50397184,"type":35}}, {"timeout":35000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:info  2020-09-19 09:04:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'selftest' to '0x00158d00052b7f72' failed: 'Error: Write 0x00158d00052b7f72/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x00158d00052b7f72"},"type":"zigbee_publish_error"}'`

i can also upgrade the adapter firmware to the latest dev for testing if you want.
and i have installed the zigbee map addin, here i see my 5 honeywell smoke detectors showing up without routes.

Sandriekus, 2020-09-19:
@Koenkk 
i've enabled herdsman debug on my current config and performed a selftest to one of the Smoke Detectors.

Zigbee2MQTT version: 1.14.4.1
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: CC2652RB_20200805
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5249779/log.txt)

Enzokot, 2020-09-19:
@Koenkk 
Sensitivity doesn't work too when the device connected to a router

Sandriekus, 2020-09-19:
ok, so for testing i've updated the firmware on the slaesh's CC2652RB to the latest dev version:

current setup:
Zigbee2MQTT version: 1.14.4.1
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: CC2652RB_20200830

logging when performing a self-test to the HoneyWell Smoke detector:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5249916/log.txt)

Sandriekus, 2020-09-19:
> @Koenkk
> Sensitivity doesn't work too when the device connected to a router

i can confirm that:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5249992/log.txt)

Koenkk, 2020-09-19:
I've did a lot of investigation in the smoke sensor lately and found out that many routers cannot handle this device (including our zstack coordinator). This is because the poll rate is very slow, causing commands to fail very often.

I expect pairing this device to a Xiaomi router (e.g. xiaomi plug) should solve this. The smoke sensor does not switch parents once it's paired so it should also keep working. You can check the parent via the networkmap: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgenetworkmap

Sandriekus, 2020-09-19:
@Koenkk Was there a major change in versions between 1.14.3 and 1.14.4?
Because with version 1.14.3 everything was working perfectly

Enzokot, 2020-09-19:
@Koenkk 

0x00158d00039a8358 - works great
0x00158d0003943639 - doesn't work selt-test and sensitivity (please, look at connections arrow. coordinator + router)

[image]

Sandriekus, 2020-09-19:
On my map i see some smoke detectors with routes, and some without routes:
[image]

the 0x00158d000528c65b (with routes) self test is working
the 0x00158d00054263de (without routes) self test is trowing the error

Koenkk, 2020-09-19:
@Sandriekus yes there was some change. Can you try this fw: 
[znp_CC2652RB_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/5250490/znp_CC2652RB_LAUNCHXL_tirtos_ccs.hex.zip)
 and repair the device (confirm that the device has a route now directly to the coordinator) and selftest again?

Sandriekus, 2020-09-19:
@Koenkk 

i've installed the firmware you've mentioned.
re-paired the 0x00158d00054263de honeywell smoke detector
performed a self-test.. and this is succesfull!

i've checked the routes, the device is not directly connected to the coordinator, but it is working!
see log:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5250546/log.txt)

Koenkk, 2020-09-19:
can you also check if it works when connected directly to the coordinator? Because when connected through the router the fw update shoulnt be necessary.

Try repairing it close to the coordinator.

Sandriekus, 2020-09-19:
> can you also check if it works when connected directly to the coordinator? Because when connected through the router the fw update shoulnt be necessary.
> 
> Try repairing it close to the coordinator.

Sure! Going to try tomorrow morning and let you know

Sandriekus, 2020-09-20:
@Koenkk 
re-paired the smoke sensor directly to the coordinator. performed a self-test and this is working.
i've checked the zigbee map addon and the smoke sensor is directly connected.

herdsman debug log:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5251230/log.txt)

So it is preferred to connect the HoneyWell Smoke Sensors directly to the coordinator? or was this just for testing?

Koenkk, 2020-09-20:
@Sandriekus have you tried to force remove it first before pairing? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

Sandriekus, 2020-09-20:
@Koenkk i have force removed the device, removed the battery for couple of minutes and then re-paired:

```
2020-09-20T16:43:26.985Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x00158d00052b7f72:48962,205,5)
2020-09-20T16:43:26.987Z zigbee-herdsman:controller:endpoint Read 0x00158d00052b7f72/1 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))
2020-09-20T16:43:26.990Z zigbee-herdsman:controller:device:log Interview procedure failed but got modelID matching 'lumi..*', assuming interview succeeded
2020-09-20T16:43:26.998Z zigbee-herdsman:controller:device:log Interview - completed for device '0x00158d00052b7f72' because of quirks ('Error: Read 0x00158d00052b7f72/1 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))')
2020-09-20T16:43:27.004Z zigbee-herdsman:controller:log Succesfully interviewed '0x00158d00052b7f72'
Zigbee2MQTT:info  2020-09-20 18:43:27: Successfully interviewed '0x00158d00052b7f72', device has successfully been paired
Zigbee2MQTT:info  2020-09-20 18:43:27: Device '0x00158d00052b7f72' is supported, identified as: Xiaomi MiJia Honeywell smoke detector (JTYJ-GD-01LM/BW)
Zigbee2MQTT:info  2020-09-20 18:43:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"MiJia Honeywell smoke detector","friendly_name":"0x00158d00052b7f72","model":"JTYJ-GD-01LM/BW","supported":true,"vendor":"Xiaomi"},"type":"pairing"}
```'

see full log here:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5252102/log.txt)

Koenkk, 2020-09-20:
Hmm, seems it still doesn't work indeed. 

I would recommend pairing the smoke sensor to a router instead.

Sandriekus, 2020-09-20:
> Hmm, seems it still doesn't work indeed.
> 
> I would recommend pairing the smoke sensor to a router instead.

an IKEA LED1842G3 bulb is oke for router?

Koenkk, 2020-09-20:
not sure, preferably a xiaomi device.

Sandriekus, 2020-09-20:
ok, so i don't have other xiaomi devices (routers) yet.
i have removed the batteries of the 5 smoke detectors, pushed the button on the smoke detectors a couple of times so all the power is gone and re-paired them with the closed ikea bulb.

lqi is way better now then before (could be the https://github.com/Koenkk/zigbee2mqtt/files/5250490/znp_CC2652RB_LAUNCHXL_tirtos_ccs.hex.zip firmware you told me to flash)
setting sensitivity is working instant
tomorrow i'm going to perform the self-tests

let you know the result

Sandriekus, 2020-09-21:
@Koenkk 
i've checked the routes this morning in the zigbee map addon - everything was ok
restarted hassio - everything ok
restarted host (raspberry Pi 4b) - everything ok
performed self-tests on all smoke detectors - all ok!

so it looks like it's working now with setup:
Zigbee2MQTT version: 1.14.4.1
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: https://github.com/Koenkk/zigbee2mqtt/files/5250490/znp_CC2652RB_LAUNCHXL_tirtos_ccs.hex.zip
performed steps:
- force remove the smoke detectors
- remove batteries from smoke detectors and push the test-button on the devices a couple of times (so all the power is released from the mainboard)
- put in back the batteries in the smoke detectors offcourse
- pair the smoke detectors nearby the closest router (ikea led bulbs in my case)

Koenkk, 2020-09-21:
@Sandriekus good, in that case the special firmware I provided should not be needed. If smoke sensor are connected through  a router it should work with the current fw: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

Sandriekus, 2020-09-21:
@Koenkk 
Oke, going to flash the CC2652RB_20200830 firmware in a couple of hours and let you know the result

Enzokot, 2020-09-21:
@Koenkk but if the device connected to the coordinator and router simultaneously like in my [example](https://github.com/Koenkk/zigbee2mqtt/issues/4285#issuecomment-695341619), self-test and sensitivity  doesn't work.

Koenkk, 2020-09-21:
@Enzokot a device cannot be connected to both, the networkmap shows this but one of the parent is lying. there is an updated cc2531 router fw available which should fix this: https://github.com/Koenkk/zigbee2mqtt/issues/4038

In you case I recommend removing the device from the network and repairing it.

Enzokot, 2020-09-21:
@Koenkk fw from [this post](https://github.com/Koenkk/zigbee2mqtt/issues/4038#issuecomment-686250233)?

Koenkk, 2020-09-22:
yes

Sandriekus, 2020-09-22:
@Koenkk 
i've flashed the slaesh’s CC2652RB stick with your latest dev firmware, but i get errors instantly
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5260718/log.txt)

all of the end devices are showing no routes in de zigbee map addon

Koenkk, 2020-09-22:
@Sandriekus can you now repair the smoke sensor to a router? Might be that this occured due to the reflashing itself.

Sandriekus, 2020-09-22:
@Koenkk 
force removed one smoke detector (0x00158d00054263de)
removed battery and pushed the button on the device a couple of times for releasing power of the mainbord
put back in battery
paired with closest ikea bulb (router)

is not working
when i check the zigbee map, every device (also 220v devices) give errors about network routes

logging:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5262102/log.txt)

Sandriekus, 2020-09-22:
as test i have flashed the znp_CC2652RB_LAUNCHXL_tirtos_ccs.hex firmware again.
restored a backup from last night

the smoke sensors are connected again, but in logging i see errors like:
```
Zigbee2MQTT:error 2020-09-22 18:56:07: Failed to execute LQI for '0xccccccfffe5de05e'
Zigbee2MQTT:debug 2020-09-22 18:56:07: Routing table succeeded for '0xccccccfffe5de05e'
Zigbee2MQTT:debug 2020-09-22 18:56:08: LQI succeeded for '0xccccccfffe5cc1ec'
Zigbee2MQTT:debug 2020-09-22 18:56:08: Routing table succeeded for '0xccccccfffe5cc1ec'
Zigbee2MQTT:error 2020-09-22 18:56:18: Failed to execute LQI for '0xccccccfffe9108bf'
Zigbee2MQTT:debug 2020-09-22 18:56:19: Routing table succeeded for '0xccccccfffe9108bf'
Zigbee2MQTT:error 2020-09-22 18:56:29: Failed to execute LQI for '0xccccccfffe6e7961'
Zigbee2MQTT:error 2020-09-22 18:56:39: Failed to execute routing table for '0xccccccfffe6e7961'
Zigbee2MQTT:debug 2020-09-22 18:56:39: LQI succeeded for '0xccccccfffe92e44e'
Zigbee2MQTT:debug 2020-09-22 18:56:39: Routing table succeeded for '0xccccccfffe92e44e'
Zigbee2MQTT:error 2020-09-22 18:56:49: Failed to execute LQI for '0x680ae2fffea9a1f6'
Zigbee2MQTT:debug 2020-09-22 18:56:50: Routing table succeeded for '0x680ae2fffea9a1f6'
Zigbee2MQTT:info  2020-09-22 18:56:50: Network scan finished
```

Koenkk, 2020-09-24:
@Sandriekus so this fw works fine: https://github.com/Koenkk/zigbee2mqtt/issues/4285#issuecomment-695346275?

Sandriekus, 2020-09-24:
@Koenkk that is correct.
with that specific firmware on my slaesh’s CC2652RB stick and zigbee2mqtt version 1.14.4.1 i can confirm the following is working:

- pairing the smoke sensors
- set sensitivity
- perform self-tests
- zigbee map addon shows routes

it is running stable now for 2 days (also after restarts of HA or complete host)

i can upgrade Zigbee2mqtt to 1.14.4.2 if you want me testing?

Koenkk, 2020-09-25:
@Sandriekus thanks, published the firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

pedroke, 2020-10-01:
Hi guys, I already created the duplicate of this issue which has been closed #4502 
After finding this discussion I applied the propossed solution:

1. I updated the firmware of my ZZH to CC26X2R1_20200925.zip
2. I force removed the xiaomi honeywell smoke detector from zigbee2mqtt
3. Removed batteries and pressed the button to discharge
4. Disconnected my router (ImmaxNeo socket) so that it doesn't mess things.
5. Smoke detector paired again with the network.

But the issue is still present even after those steps. Firmware is new for sure and it is still not possible to send messages to smoke detector and it still appears detached in the network map. Although reports correctly.

One thing I noticed after re-flashing of the firmware of ZZH. All devices seemed to be working but all were displayed as detached in the network map. I needed to press the join button on each of them to be displayed correctly again. Even though no routers were involved.  After re-joining they are displayed correctly in the map except of Sonoff humidity temperature and Xiaomi Smoke..
Strange..
I have to find how exactly is this "ping" working in zigbee networks, since almost all of my devices are baterry powered, only turning on when they report so theoretically they should also NOT respond to any kind of ping..

pedroke, 2020-10-01:
One additional remark from my side. Do I understand correctly that "ping" is implemented internally as "node descriptor request".. If so, then it is really strange how it should work for sleeping devices. If I understand the specification correctly it advises to cache node descriptors on coordinators and routers but it consumes memory.. so probably there is some limitation. It would mean, that may be in practice only a limited number of descriptors is cached and it may result in unsuccessful "pings" for sleeping devices and it cannot be solved only by changing the timeouts.. Does anybody know how many node descriptors are cached by particular firmwares? If we cannot change this, then the whole network map and pinging do not make much sense for larger networks. But the impossibility to send the message e.g. to smoke sensor is a different question and might be related to timeouts if I understand correctly.

Koenkk, 2020-10-02:
@pedroke to check whether the issue is present, use 20200925 in combination with Zigbee2MQTT 1.15.0.

What do you exactly mean with ping? If you mean availability you can read here how it works for EndDevices: https://www.zigbee2mqtt.io/information/availability.html#non-pingable-devices

pedroke, 2020-10-02:
@Koenkk Thanks for the hint, I will try with the newer version.

By "ping" I mean the mechanism of how a network map is built. What I can't understand is that when I trigger the network map build, all my battery-powered devices, which are sleeping for sure are displayed as connected. So it is probably only decided based on the received messages in the last 24 hours - as you mentioned. But e.g. smoke detector which is also SED is displayed as detached, although messages come regularly.. so I suppose that for that specific device different approach is taken. And what is even more interesting, that even Sonoff humidity and temperature which is almost the same as Aqara temperature sensor is also sometimes displayed as detached, even though it reports periodically...
So.. that was my confusion. whether ping is executed by "node descriptor request" or simply by the last 24 hours message or by any other approach and what are the rules and exceptions there.. And when I think about the "No route" error, it is even more interesting, why is it so. When the sensor joins the network directly to the coordinator, how is it possible that there is no route defined to it? I can understand that the message timeout can be a problem for a partially sleeping device, but a missing route? Hmm?
All in all, it is not that important, but one has to think about it when it's happening... :)

Koenkk, 2020-10-02:
@pedroke the way this works is that not the end device itself, but the parent of the end device (a router) report what childs it has, this is what you see on the network map.

pedroke, 2020-10-02:
Let's take an example. I reflashed firmware on ZZH..All devices were reporting correctly and had strong signal. But on the map all devices which are reporting to coordinator were detached. So coordinator was accepting messages from them, it was aware of them and still was not able to name them for networm map... strange. As soon as any of the device forcibly re-joined, it was again showing attachment in the map. Even stranger is that devices behind router were displayed correctly.
So.. something that was held in the memory of coordinator was deleted during flashing and resulted in detached nodes in the map. Since router was not flashed it preserved the list of its children.
So the question is, based on what parent returns the list of children?

pedroke, 2020-10-02:
Ok..now I see.I checked the code of network map sn the connections are based on routing table of the device. So in case there are detached devices there is no route defined. That also explains why smoke detector returns 'no route' error ehile it is detached in the map. And it also explains that flashing clears the map from RAM. But why are som routes cleared during runtime remains mistery for me. ED like xiaomi battery sensors keep the connection on map for ages even though they sleep and not listen at all..but smoke detector keep detaching and also e.g Sonoff temperature which is basically very similar to aqara...I need to find out what are the rules for routing table. What kicks some records out of it and why some of the sleepy devices remain untouched without any problem.

Anjerlaan, 2020-11-01:
Hi @koenkk,

You mentioned also a new firmware along with the Z2M update
https://github.com/Koenkk/zigbee2mqtt/issues/4832#issuecomment-720060815
I just saw you have created a new firmware for the ZZH adapter

https://electrolama.com/projects/zig-a-zig-ah/#flash-firmware
I am doing this for the first time, so I have some additonal questions about new firmware flashing:

- Do I first have to flash the new ZZH Firmware and then update Z2M, or the other way around?
- Do I need to to be in bootloader mode to flash the new firmware
- Do I need to erase the old flash first
- Do I need to repair all devices after the new flash
- Do I have o change the Pan-ID again

Greetzzz,

Gerben

Koenkk, 2020-11-01:
- Doesn't matter
- yes
- Just run the command here: https://electrolama.com/projects/zig-a-zig-ah/#flash-firmware (this will also erase)
- No
- No

andrej-peterka, 2020-11-17:
Hey @Koenkk ,

don't want to steal the conversation, but I noticed I also have a "No network route" error with a SilverCrest Motion sensor when pairing to zzh with the latest firmware.

I created a forum post here:
https://zigbee2mqtt.discourse.group/t/problem-pairing-to-zzh-working-with-cc2531/1773

And here's the relevant log file when pairing that device fails:
https://hastebin.com/ojimimozek.apache

Search for "No network route". 

Not sure if related, just though worth pointing it out.

pedroke, 2020-11-17:
@andrej-peterka For me it seems more like a failed interview - many 240 errors.. and subsequent removal from network and thus no route is found anymore. Have you tried to pair closer to the coordinator? Btw didn't you press pairing button while the interview was still in progress aprox after 17 seconds? Maybe Koen have better understanding, from the logs it seems to me like unfinished interview and device leaving the network.

andrej-peterka, 2020-11-17:
@pedroke I paired it right next to the coordinator and did not touch anything. Maybe some motion triggered something, not sure.
But yeah, I saw the "left the network" messages also, no idea what caused them, I assumed a failed interview.

I'm a newbie when it comes to the zigbee network, so my knowledge is lacking in this regard, sorry. Seems I jumped to conclusion too fast.

pedroke, 2020-11-17:
@andrej-peterka As it turned out with Honeywell smoke detector, the problem was with enrolling it to IAS zone. It is a special feature of zigbee specification for some security sensors.. in the end we found that coordinator was not triggering the enroll procedure in a way that sensor expected.. and it was fixed. But there were logs about enrolling.. In your case I see some data confirmation errors code 240 which is "transaction expired" that's why I thought that interview failed because of weak signal..but now..I don't know.. Btw..good to know that Lidl is also in Zigbee business ;)

Anjerlaan, 2020-11-17:
I had the same difficulty with paring by just holding it close to my ZZH adapter.
But with the domoticz-zigbee2mqtt-plugin the smoke detector was paired quite fast actually.

[3 more replies]

closed 2020-12-30 by github-actions[bot]