Koenkk/zigbee2mqtt #4290: zigbee2mqtt should remove 'retained' value from broker on device rename

sjorge, 2020-09-06:
# Bug Report
## What happened
I was testing an new temp/humidity sensor (sonoff) as I was not happy with my current TuYa one.

So I had 2:
- bedroom/sensor (TuYa)
- bedroom_aux/sensor (SONOFF)

I was happy with the SONOFF so I decided to replace the TuYa one.

1. renamed bedroom/sensor to bedroom_old/sensor
2. renamed bedroom_aux/sensor to bedroom/sensor

I however noticed node-red and some other mqtt clients still saw retained values for bedroom_aux/sensor being published on connect.

```
2020/09/06 16:51:31 INFO     mqtt2rrd: received 'zigbee2mqtt/bedroom_aux/sensor' with '{'temperature': 20.59, 'humidity': 55.82, 'linkquality': 60, 'battery': 100}'
```

I did some digging to verify it is indeed only the retained value still present on the broker.

## What did you expect to happen
I'd expect zigbee2mqtt to tell the broker to forgot the old retained value on rename (or on device removal too actually)

## How to reproduce it (minimal and precise)
- mosquitto broker with retain configured to be perstistantly stored on disk
- z2m connect to mosquitto
- other client connected to mosquitto subscribed to the z2m topics.

## Debug Info
```
[root@amethyst /opt/zigbee2mqtt]# mosquitto_sub -t zigbee2mqtt/bridge/info | jq .coordinator
{
  "meta": {
    "maintrel": 3,
    "majorrel": 2,
    "minorrel": 6,
    "product": 0,
    "revision": 20200728,
    "transportrev": 2
  },
  "type": "zStack12"
}
```

sjorge, 2020-09-06:
I got an easier test to show the retained value! You can just use mosquitto_sub to get it.

```
[root@amethyst /opt/zigbee2mqtt]# mosquitto_sub -t zigbee2mqtt/bedroom_aux/sensor
{"battery":100,"humidity":55.82,"linkquality":60,"temperature":20.59,"voltage":3200}
```

Koenkk, 2020-09-07:
Implemented!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-07 by sjorge