Koenkk/zigbee2mqtt #4291: Issues since a few days 'Failed to setup reporting for x'

svh1985, 2020-09-06:
Hi,

Since a few day's I'm having issues with my Z2M network.
I did not add or remove anything, the only thing changed is the upgrade to the latest docker image. Downgrading to the previous version gave the same error's so I don't think that's the issue. I also tried the dev version.

Used adapter: Zigbee CC2531, https://www.aliexpress.com/item/32869263224.html
I know the used adapter has it's limitations and an upgrade might be neccacary, so I would like to ask some help to verify if that's the issue or if something else is causing my problems.

Running on latest version of DietPi HW: RPi2

Docker-Compose
```
  zigbee2mqtt:
    container_name: zigbee2mqtt
#    image: koenkk/zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    volumes:
      - ${CONFIGDIR}/zigbee2mqtt:/app/data:rw
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    restart: always
    labels:
      com.centurylinklabs.watchtower.enable: "true"
    network_mode: host
    privileged: true
    environment:
      TZ: ${TIMEZONE}
```

Config:
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: '!secret server'
  user: '!secret user'
  password: '!secret password'
  client_id: Zigbee2MqttBridge
  include_device_information: true
  keepalive: 60
serial:
  port: /dev/ttyACM0
  disable_led: false
advanced:
  pan_id: '!secret pan_id'
  ext_pan_id: '!secret ext_pan_id'
  channel: 11
  cache_state: true
  log_level: debug
  log_directory: /var/log/
  log_output:
    - console
    - file
  baudrate: 115200
  rtscts: true
  soft_reset_timeout: 0
  network_key: '!secret network_key'
  last_seen: disable
  elapsed: false
  availability_timeout: 60
  report: true
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: hass/status
  timestamp_format: 'YYYY-MM-DD HH:mm:ss'
map_options:
  graphviz:
    colors:
      fill:
        enddevice: '#fff8ce'
        coordinator: '#e04e5d'
        router: '#4ea3e0'
      font:
        coordinator: '#ffffff'
        router: '#ffffff'
        enddevice: '#000000'
      line:
        active: '#009900'
        inactive: '#994444'
queue:
  delay: 250
  simultaneously: 5
devices:
  '0x0017880104b4fc86':
    friendly_name: Trap bewegingssensor
    qos: 2
  '0x00178801105faa57':
    friendly_name: Hue Dimmer switch
    qos: 2
  '0x00158d00042c04c9':
    friendly_name: Vaatwasser watersensor
    qos: 2
  '0x00158d0004016b96':
    friendly_name: Voordeur sensor
    qos: 2
  '0x00158d00034d1a4e':
    friendly_name: Buitenkast temperatuur sensor
    qos: 2
  '0x00158d000322b8f2':
    friendly_name: Badkamer temperatuur sensor
    qos: 2
  '0x0017880100f559ca':
    friendly_name: Zolder panty lamp
    qos: 2
  '0x0017880102108484':
    friendly_name: Zolder bewegingssensor
    qos: 2
  '0x0017880100df10b4':
    friendly_name: Gang Lamp
    qos: 2
  '0x001788010200589e':
    friendly_name: Gang bewegingssensor
    qos: 2
  '0x680ae2fffe7714ac':
    friendly_name: Badkamer signaal versterker
    qos: 2
  '0xec1bbdfffea3bcb2':
    friendly_name: Zolder Spot 1
    qos: 2
  '0xec1bbdfffe41c692':
    friendly_name: Zolder Spot 2
    qos: 2
  '0xec1bbdfffe910f36':
    friendly_name: Zolder Spot 3
    qos: 2
  '0x0000000000412487':
    friendly_name: Hue Tap Eetkamer
    qos: 2
  '0x001788011041f6b2':
    friendly_name: Eettafel Lamp
    qos: 2
  '0x0017880106f8a01a':
    friendly_name: Plant Lamp
    qos: 2
  '0x0017880100b369b4':
    friendly_name: Bank lamp 1
    qos: 2
  '0x0017880103da2dd6':
    friendly_name: Bank lamp 2
    qos: 2
  '0x0017880100cfe0eb':
    friendly_name: Bar Lamp
    qos: 2
  '0x0017880100e00589':
    friendly_name: Keuken lamp 1
    qos: 2
  '0x0017880100d90218':
    friendly_name: Keuken lamp 2
    qos: 2
  '0x0000000000424d78':
    friendly_name: Hue Tap Keuken
    qos: 2
  '0x000000000040085b':
    friendly_name: Hue Tap Slaapkamer
    qos: 2
  '0x0017880100e6e01e':
    friendly_name: Slaapkamer Plafond Lamp
    qos: 2
  '0x00178801026f2d2f':
    friendly_name: Slaapkamer Kast Lamp
    qos: 2
  '0x00158d0004852c89':
    friendly_name: Wasmachine watersensor
  '0x00158d0004a06c30':
    friendly_name: Zolderdeur sensor
  '0x00158d00042d864a':
    friendly_name: Dakterras deur sensor
  '0x00124b001fb984a9':
    friendly_name: ptvo
```

Log [debug.log](https://github.com/Koenkk/zigbee2mqtt/files/5180136/debug.log)

Koenkk, 2020-09-07:
@sjorge is it expected behaviour that setting up reporting for genOnOff cluster always fails for Hue bulbs?

sjorge, 2020-09-07:
I only had 2 examples, but sort of.

Out of the box only genOnOff reporting works. Once you OTA to the newest firmware this fails. Might be worth having that configureReporting in a try block.

closed 2020-10-15 by github-actions[bot]