Koenkk/zigbee2mqtt #4294: Publish 'set' 'state' to 'zb_switch_lvr' failed: 'Error: Command 0xec1bbdfffe66cb66/3 genOnOff.on({}

vuminh90, 2020-09-06:
## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2530
I have a problem with switch zigbee 3 gang - Model: ZM-L03E-Z
I click button on hass but state switch on hassio don't change (relay still do it). After 6s, state on hassio updated lastest status of switch
I checked log as below:
Sep 06 20:52:55 raspberrypi npm[5153]: Zigbee2MQTT:error 2020-09-06 20:52:55: Publish 'set' 'state' to 'zb_switch_lvr' failed: 'Error: Command 0xec1bbdfffe66cb66/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 20901 - 3 - 54 - 6 - 11 after 10000ms)'
Sep 06 20:52:55 raspberrypi npm[5153]: Zigbee2MQTT:info  2020-09-06 20:52:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'zb_switch_lvr' failed: 'Error: Command 0xec1bbdfffe66cb66/3 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 20901 - 3 - 54 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"zb_switch_lvr"},"type":"zigbee_publish_error"}'

My config in configuration.yaml:
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: ********
  password: ********
serial:
  port: /dev/ttyUSB0
advanced:
  rtscts: false
  channel: 15
  baudrate: 115200
  availability_timeout: 0
  
experimental:
  new_api: true
  
devices: devices.yaml
groups: groups.yaml

[image]
[image]

Koenkk, 2020-09-07:
If I understand correctly, command gives an error but relay status changes?

If so try changing the entry in `devices.js` to

```js
{
    zigbeeModel: ['TS0003'],
    model: 'ZM-L03E-Z',
    vendor: 'Zemismart',
    description: 'Smart light switch - 3 gang with neutral wire',
    supports: 'on/off',
    fromZigbee: [fz.ignore_basic_report, fz.on_off],
    toZigbee: [tz.on_off],
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    meta: {configureKey: 1, multiEndpoint: true, disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint) => {
        await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to edit `devices.js`

vuminh90, 2020-09-09:
ok i changed device.js
And my switch already update status
Thanks bro!

Koenkk, 2020-09-09:
So with this it all works?

vuminh90, 2020-09-23:
Yes, after edit all switch working perfect
Thanks!

Koenkk, 2020-09-25:
Added, will be in next release.

closed 2020-09-25 by Koenkk