Koenkk/zigbee2mqtt #4295: Ikea Fyrtur blinds battery state

amaia81, 2020-09-06:
Hi,

After updating via OTA my Fyrtur blind it doesnt report the battery state anymore. I tried to repair it and no luck so I removed it, restarted zigbee2mqtt and paired it again but the issue remained. Any suggestion?

Thanks!

closed 2020-10-14 by github-actions[bot]

EfficientSetting, 2021-02-04:
I have the same issue, battery is always reported as 0

ThomasSig, 2021-02-12:
I also have the same issue. Two identical devices, identical configuration and firmware (2.2.009). One with battery state and the other shows N/A. Any suggestions?

EfficientSetting, 2021-02-12:
I needed to click "configure" on the Z2M interface for battery to be reported. Worked afterwards. 

I believe this has to do with the recent per device reporting that was implemented.

ThomasSig, 2021-02-16:
> I needed to click "configure" on the Z2M interface for battery to be reported. Worked afterwards.
> 
> I believe this has to do with the recent per device reporting that was implemented.

This also fixed it for me. "Configure" and some patience :-)
Thanks,

aiolive, 2021-05-11:
@ThomasSig: i have the same issue, but i can't find the "configure" setting on the Z2M interface.... can you help me plese :)

filikun, 2022-02-10:
And where did you find that "configure" setting?

ThomasSig, 2022-02-10:
> > @ThomasSig: i have the same issue, but i can't find the "configure" setting on the Z2M interface.... can you help me plese :)
> > edit: it's fixed after a long time :)
> 
> And where did you find that "configure" setting?

@filikun : Sorry, I moved to the integrated zigbee feature in Home Assistant, because off some issues that eat-up the batteries  in just 1-2 weeks. 
As I remember out from some screenshots... 
Go to "Device" view in the web interface and in the right side of the Ikea Fyrtur device, click on the yellow button to re-configure. Be patience because it will take some time to update...

filikun, 2022-02-11:
> > And where did you find that "configure" setting?
> Go to "Device" view in the web interface and in the right side of the Ikea Fyrtur device, click on the yellow button to re-configure. Be patience because it will take some time to update...

Thanks m8, I did a reboot of Z2M and then re-configure. After some time it all worked again 👍

mgrcic, 2022-07-05:
hello, 
I have the similar issue. I am integrating this device into the 3rd party hub and the battery measurement reports are always 0.
This is confirmed by the ZigBee sniffer logs.
The configure report command was successfully processed by the roller blind, but it always reports 0 for battery voltage and battery percentage remaining attributes.
The device firmware version is 1.99.34-2.2.009
Do you have any suggestions?