Koenkk/zigbee2mqtt #4296: Is my zigbee network too big for CC2531?

quarcko, 2020-09-07:
After some runtime, my network fails, usually it starts with LQI scan failure for "Coordinator",
Then my networkmap becomes with no links at all, and devices slowly stop comunication with coordinator.
What helps is to shutdown the addon. Physically remove USB Stick to power it down, insert back
and restart the addon. Then it works for another few days.

## Debug Info
Zigbee2MQTT version: Latest on HASSIO, allways autoupdates to the latests
Adapter hardware: CC2531.
Adapter firmware version: Latest source routing

Logs when trying to turn on the switch, when in this failed state:
https://pastebin.com/ZTFHcyrV

The most interesting message from this error is:
Data request failed with error: 'MAC no resources' (26)

So just please confirm, is this the end for me with CC2531 and i should opt for a better stick?

way2many, 2020-09-07:
You don't mention your network size but this is a possibility or WiFi interference. I had on CC2531 a network of 43 devices with source firmware. It was functionning very well until I added 4 more devices and I have hit the physical threshold. I'm able to have all devices connected by resetting the CC2531 but then after a few hours 1 to 2 devices are ejected of the network systematically. I ordered a zig-a-zig-ah! last week so hope to migrate this week.

Wifi interference was a big issue for me months ago. I switched to channel 25 and it has been rock solid until the device threshold.

ALaDoffe, 2020-09-07:
Hi all,

Exactly same for me since 2 weeks.

Conf : 
CC2531 adaptor with 20190619 source-routing firmware.
Zigbee2MQTT version 1.14.4-dev (commit #b428070)
Currently 20 devices are joined

Everything works well for 3 or 4 days and then  : 
Failed to execute LQI for 'Coordinator'
failed (Data request failed with error: 'MAC no resources' (26))....

Most of all devices became unvailable.

I don't know if it is related to Z2M version but first error occured for me with : 
Zigbee2MQTT version 1.14.3-dev (commit #9b4ef70)

A Z2M service restart resolves issue for 3 or 4 days more...

My two cents.

quarcko, 2020-09-07:
Yeah sorry not mentioning my network size:
Currently i have 51 device, and this problem appeared only recently, like started a week ago.
But my network was already this big for around ~1 month.
And it was running stable with 45 devices for over a year.

And my wifi and bluetooth devices did not change during this period.
Also my channel separation is good between wifi and zigbee from start.

quarcko, 2020-09-08:
For me it does not help to just restart Z2M addon (service), i need to physically power down the adapter.
And it seems that problem gets even worse, yesterday after restart it worked for about 1 hour and failed.
Today in the morning i restarted it again, lets see....
Also there are now frequent LQI failures, map is incomplete, etc...

@Koenkk  help! Should i buy a more powerful adapter? Or is it some bug with Z2M?

Koenkk, 2020-09-08:
`MAC no resources' (26)` error means that the coordinator is out of memory. I'm surprised to read that your network still worked well with 45 devices. I think it depends on the type in it (e.g. having sensors only sending data isn't as stressful as controlling switches). 

It looks like you ran into the limits of the CC2531 and I recommend upgrading to a more powerful one.

quarcko, 2020-09-09:
Yeah, it was working so long okay with 45 devices, because all the time i was on old firmware 2018XXXX something.
this new source routing firmware is not so stable... but it has groups and supports ikea button... so that's why im here.
Anyway - just tried today "default" firmware. It performs even worse, it does not last longer than 5 minutes before 
whole stick just crashes and does not accept any commands until replugged.

And in the process of testing i lost my 2018XXXX firmware.. damn i still have file, but reflashing will not save me,
this firmware does not support setting PAN_ID forcefully...

@Koenkk do you have some internal log of your modifications that you did to firmware configurations?
Really i would like to build myself a firmware with all the perks of latest firmwares
but keep the settings of (max devices, heap, ram and what not) from 2018XXXX firmware?
If you could find some your notes or some changelog along the way...

quarcko, 2020-09-09:
Nah, maybe i just overglorify this old firmware, nevermind, sorry.
Just will get a better stick...

jhron, 2020-10-06:
I have similar problem. But in my case it was working ok when all devices are powered on. I have at the moment 14 IKEA  spot lights installed. Turned on by 2 relays. 4 one group and 10 second group. The relays are normally all the time turned on. But when they are off and coordinator is running already and I turn e.g. all 4 lights on using relay, whole network stops working. Only way how to fix this is to power off dongle ( reboot doesn't help). 

So in case of some power failure I must be sure that all lights are already powered on before I turn ZigBee 2Mqtt on.

closed 2020-11-13 by github-actions[bot]

pvillanyi, 2022-04-27:
I have experienced it out of sudden yesteray. Ma z2m network has worked stable for months... and now the automations don't work. I get the Data request failed with error: 'MAC no resources' (26)) error message... 
I have only 8 devices in my network and no changes since more then a month. What can cause this sudden problem?