Koenkk/zigbee2mqtt #4297: Ikea outlet E1603 letting current through when OFF

way2many, 2020-09-07:
Hi all,
Not really an issue with Z2M but this weekend I installed some 12V garden lights behind an Ikea outlet. Nothing fancy, some cheap LED with a 220V to 12V transformer. Every few seconds they flash, as if a condensator is discharging. This reminds me of the issues with a batch of Ikea GU10 which were also flashing (there are some discussion here about this).

Anybody experienced the same with an Ikea outlet? Should I be worried that it may be a faulty unit?

Thanks for help

bartekluk, 2020-09-08:
It probably is, I had a similar problem with mine. Relay would switch the power was still goging through despite the status led chaning state. Never got to return it since it was in the middle of covid.

way2many, 2020-09-08:
Ok great thanks. I will try to exchange it.

closed 2020-09-08 by way2many