Koenkk/zigbee2mqtt #4298: Device specific settings?

sirs2k, 2020-09-07:
Hi,

I'm a little confused, where do I put the config for hue_power_on_behavior so it recovers the hue settings before power loss?

I'm using hass.io plugin and I've tried putting the settings in devices.yaml like so
```
'0x0017880102d9e6ef':
  friendly_name: office_hue_3
  retain: true
  retrieve_state: true
  hue_power_on_behavior: recover
  hue_power_on_brightness: 125
```
But no luck, apologies if I've completely missed any instructions
Thank you

sjorge, 2020-09-07:
It's been ages but I think you to publish those to `zigbee2mqtt/office_hue_3/set/hue_power_on_behavior` with payload of `recover` and so forth.

sirs2k, 2020-09-08:
> It's been ages but I think you to publish those to `zigbee2mqtt/office_hue_3/set/hue_power_on_behavior` with payload of `recover` and so forth.

Thanks man, it's working but it's a hit and miss.
I turn off, turn on, it recovers
do it again seconds later, it's back to default
?

dozminic, 2020-09-15:
I'm having issues with this feature as well. According to the documentation, it should work by setting the device-specific option. However, I cannot get it to work no matter where I place it in the configuration. I can get it to work semi-reliably if I publish the MQTT message, but it does not persist past a stop/start of Z2M.

closed 2020-10-23 by github-actions[bot]

lolorc, 2020-10-25:
a bit sad it's being closed, it lacks some more detailed documentation, also it would be nice to be able to retrieve the current configuration

u17194907425, 2020-11-08:
@sirs2k 

> do it again seconds later, it's back to default

If I understand the manual right, then  _seconds later_ might be leading to a different behavior. In the Hue Bluetooth app it says that in order to force light on you need to quickly turn on-off-on. Then it will ignore the configs and turn on anyway,