Koenkk/zigbee2mqtt #4316: Tradfri set color_temp transition freezes bulb

teemu-kemppainen, 2020-09-09:
# Bug Report
Ikea Tradfri bulb freezes if color_temp is sent with transition. The freeze lasts for the transition time. 

## What happened
When changing color_temp with transition (alone or with brightness), the bulb does not respond until the transition is finished.
When changing brightness only with transition, the bulb stays responsive.

## What did you expect to happen
The bulb stays responsive during the transition.

## How to reproduce it (minimal and precise)
Send these within 20 seconds:
mosquitto_pub -t zigbee2mqtt/<friendly_name>/set -m '{"state": "ON", "color_temp": 500, "transition": 20}'
mosquitto_pub -t zigbee2mqtt/<friendly_name>/set -m '{"state": "OFF"}'

The bulb does not turn off. 

Send this after 20 seconds:
mosquitto_pub -t zigbee2mqtt/<friendly_name>/set -m '{"state": "OFF"}'

The bulb turns off properly.

Brightness only works:
mosquitto_pub -t zigbee2mqtt/<friendly_name>/set -m '{"state": "ON", "brightness": 254, "transition": 20}'
mosquitto_pub -t zigbee2mqtt/<friendly_name>/set -m '{"state": "OFF"}'

The bulb turns off immediately as expected.

## Debug Info
Zigbee2MQTT version: Docker koenkk/zigbee2mqtt:latest
Adapter hardware: CC2531
Adapter firmware version: 20190608
Bulb: TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (LED1537R6/LED1739R5)
Bulb firmware version: 2.0.029

sjorge, 2020-09-09:
I can reproduce this, looks like the bulb also stopped accepting color_temp when the state is off. Used to work on the previous firmware... I wonder if it's a firmware bug.

Flurkmark, 2020-09-14:
Afaik this is a known issue for some IKEA bulbs and is indeed firmware related.

closed 2020-10-22 by github-actions[bot]