Koenkk/zigbee2mqtt #4317: Updated to 1.14.4 (Hassio Addon), now PAN ID is already in in use ?

TheStigh, 2020-09-09:
# Bug Report
I don't know if this is a bug or not, seen some messages regarding PAN ID suddenly already in use. But I'm stuck.

## What happened
After updating from 1.14.3 to 1.14.4, my 51 zigbee devices became unavailable. Did a HA Reload and went back to Z2M logs and discovering message "pan id already in use...". Tried to change the default PAN ID to 6755, and then Z2M starts up again, of course with a clean database.  I did not edit or change anything in the configuration.

A major problem to me is that this is a remote location taking hours driving each way, and would love to resolve the issue remotely. Also, re-pairing 51 devices is not something I would look forward to... Can that be done?

## How to reproduce it (minimal and precise)
I don't know if it can be reproduced, updated to 1.14.4 just fine on my local HA instance without any issues.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC1352P-2
Adapter firmware version: 20200805

Koenkk, 2020-09-09:
The `pan id already in use..` error should only appear when the adapter is initialised for the first time, I'm not sure why it happend in your case. To investigate it we need debug logging (which I assume you don't have).  

When changing the pan ID, Zigbee2MQTT will reset itself (emptying the database.db) and start a new network. Do you have a backup of the `database.db` and `coordinator_backup.json`? In that case you should be able to recover your network by reverting back to the default pan ID, reflash the adapter and start Zigbee2MQTT. If not you unfortunately have to repair.

TheStigh, 2020-09-09:
The whole upgrade has given a weird experience. When trying to change the config (yes, I tried changing the pan_id and did see it started fresh with clean db), I got the **'error in config [options]...'** whatever I tried to add. I ended up with restoring the **share** and **z2m** back to 1.14.3 but still the error was persistent. Re-updated to 1.14.4 with same result.

I do have backups so that is no problem. Have decided to take the hours in the car tomorrow and will try to restore the backup with the 1352 disconnected, reflash it, re-connect and start up z2m with my fingers crossed. If not there will be some hours doing the re-pairing...

TheStigh, 2020-09-13:
> The `pan id already in use..` error should only appear when the adapter is initialised for the first time, I'm not sure why it happend in your case. To investigate it we need debug logging (which I [...]

That did it for me :) Thx @Koenkk 
CLosing the issue.

closed 2020-09-13 by TheStigh