Koenkk/zigbee2mqtt #4320: Zen Zen-01-W lacks cool / cooling

JasonSwindle, 2020-09-09:
# Bug Report
## What happened
Device can support Heat / Cool, but mqtt (and HASS) can only support heat

## What did you expect to happen
The device to be able to do more than heat.

## How to reproduce it (minimal and precise)
Docs and HASS MQTT discover both only have HEAT mode.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: LAUNCHXL CC1352P-2
Adapter firmware version: Z-Stack_3.x.0 (commit f4a0d30302b4757771f01420272b6ced89001dbc)

JasonSwindle, 2020-09-12:
Is there something more I can provide to help move this forward? :)

Koenkk, 2020-09-14:
@brubaked do you still have this device? Could you help with this issue?

brubaked, 2020-09-14:
Alright, my setup since I switched to back from smartthings to Zigbee2MQTT only has heating, so I hadn't noticed this before. I can confirm that when the thermostat is configured to support cooling ([page 19 of this document](https://www.homecontrols.com/homecontrols/products/pdfs/ZE-Zen/Zen-Thermostat-Manual.pdf)) the auto discovery only exposes the heating mode. If I manually configure the thermostat in the configuration.yaml file, the heating and cooling functions work as expected.

```yaml
climate:
  - platform: "mqtt"
    name: "Zen Thermostat"
    availability_topic: "zigbee2mqtt/bridge/state"
    temperature_unit: "C"
    min_temp: "10"
    max_temp: "30"
    mode_state_topic: "zigbee2mqtt/<FriendlyName>"
    mode_state_template: "{{ value_json.system_mode }}"
    mode_command_topic: "zigbee2mqtt/<FriendlyName>"
    current_temperature_topic: "zigbee2mqtt/<FriendlyName>"
    current_temperature_template: "{{ value_json.local_temperature }}"
    temp_step: 0.5
    action_topic: "zigbee2mqtt/<FriendlyName>"
    action_template: "{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}"
    modes: 
      - "off"
      - "heat"
      - "cool"
      - "fan only"
    temperature_state_topic: "zigbee2mqtt/<FriendlyName>"
    temperature_state_template: "{{ value_json.occupied_heating_setpoint }}"
    temperature_command_topic: "zigbee2mqtt/<FriendlyName>/set/occupied_heating_setpoint"
```

The debug log during pairing when the thermostat is configured for heating and cooling is below:

```
Zigbee2MQTT:info  2020-09-14 08:03:41: Device '<FriendlyName>' joined
Zigbee2MQTT:info  2020-09-14 08:03:41: Starting interview of '<FriendlyName>'
Zigbee2MQTT:info  2020-09-14 08:03:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"<FriendlyName>"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-09-14 08:03:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"<FriendlyName>"},"type":"pairing"}'
Zigbee2MQTT:debug 2020-09-14 08:03:41: Device '<FriendlyName>' announced itself
Zigbee2MQTT:info  2020-09-14 08:03:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"<FriendlyName>"},"type":"device_announced"}'
Zigbee2MQTT:debug 2020-09-14 08:03:55: Received Zigbee message from '<FriendlyName>', type 'readResponse', cluster 'genBasic', data '{"modelId":"Zen-01"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-14 08:03:55: No converter available for 'Zen-01-W' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"Zen-01"}'
Zigbee2MQTT:info  2020-09-14 08:03:55: MQTT publish: topic 'homeassistant/climate/<FriendlyName>/climate/config', payload '{"action_template":"{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}","action_topic":"zigbee2mqtt/<FriendlyName>","availability_topic":"zigbee2mqtt/bridge/state","current_temperature_template":"{{ value_json.local_temperature }}","current_temperature_topic":"zigbee2mqtt/<FriendlyName>","device":{"identifiers":["zigbee2mqtt_<FriendlyName>"],"manufacturer":"Zen","model":"Thermostat (Zen-01-W)","name":"<FriendlyName>","sw_version":"Zigbee2MQTT 1.14.4"},"json_attributes_topic":"zigbee2mqtt/<FriendlyName>","max_temp":"30","min_temp":"10","mode_command_topic":"zigbee2mqtt/<FriendlyName>/set/system_mode","mode_state_template":"{{ value_json.system_mode }}","mode_state_topic":"zigbee2mqtt/<FriendlyName>","modes":["off","auto","heat"],"name":"<FriendlyName>_climate","temp_step":0.5,"temperature_command_topic":"zigbee2mqtt/<FriendlyName>/set/occupied_heating_setpoint","temperature_state_template":"{{ value_json.occupied_heating_setpoint }}","temperature_state_topic":"zigbee2mqtt/<FriendlyName>","temperature_unit":"C","unique_id":"<FriendlyName>_climate_zigbee2mqtt"}'
Zigbee2MQTT:info  2020-09-14 08:03:55: MQTT publish: topic 'homeassistant/sensor/<FriendlyName>/linkquality/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_<FriendlyName>"],"manufacturer":"Zen","model":"Thermostat (Zen-01-W)","name":"<FriendlyName>","sw_version":"Zigbee2MQTT 1.14.4"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/<FriendlyName>","name":"<FriendlyName>_linkquality","state_topic":"zigbee2mqtt/<FriendlyName>","unique_id":"<FriendlyName>_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2020-09-14 08:03:55: Received MQTT message on 'homeassistant/climate/<FriendlyName>/climate/config' with data '{"action_template":"{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}","action_topic":"zigbee2mqtt/<FriendlyName>","availability_topic":"zigbee2mqtt/bridge/state","current_temperature_template":"{{ value_json.local_temperature }}","current_temperature_topic":"zigbee2mqtt/<FriendlyName>","device":{"identifiers":["zigbee2mqtt_<FriendlyName>"],"manufacturer":"Zen","model":"Thermostat (Zen-01-W)","name":"<FriendlyName>","sw_version":"Zigbee2MQTT 1.14.4"},"json_attributes_topic":"zigbee2mqtt/<FriendlyName>","max_temp":"30","min_temp":"10","mode_command_topic":"zigbee2mqtt/<FriendlyName>/set/system_mode","mode_state_template":"{{ value_json.system_mode }}","mode_state_topic":"zigbee2mqtt/<FriendlyName>","mo [... cut ...] pe":"pairing"}'
Zigbee2MQTT:info  2020-09-14 08:04:08: Successfully configured '<FriendlyName>'
Zigbee2MQTT:debug 2020-09-14 08:04:58: Received Zigbee message from '<FriendlyName>', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":544,"imageType":2,"manufacturerCode":4440}' from endpoint 1 with groupID 0
```

It seems like the thermostat is indeed only reporting that it has `off`, `auto` and `heat` modes available:
```
"modes":["off","auto","heat"]
```

JasonSwindle, 2020-09-18:
So does this mean it needs to be fixed at zigbee-herdsman level? Or a flaw with the device?

JasonSwindle, 2020-09-30:
@Koenkk  @brubaked Following up on this. I hate having to poke people. Thank you for the help. :)

brubaked, 2020-10-01:
@JasonSwindle Sorry! I haven't had time to dive into this deeper and figure out where the lack of feature autodetection lies. In the meantime, using the YAML code I posted above should give you access to cooling control in Home Assistant.

JasonSwindle, 2020-10-03:
Howdy,

I will give this a test. By `configuration.yaml` do you mean Zigbee2MQTT or HASS? They both have a configuration.yaml file, but I figure it goes into HASS, as it looks like a config for it.

JasonSwindle, 2020-10-07:
I'm not sure if this is the dev version, or something else, but I can update MQTT but the device never gets the update. I'm seeing this in the log:

```
ERROR: Publish 'set' 'occupied_heating_setpoint' to 'zen-thermostat' failed: 'Error: Write 0x00244600000eaea7/1 hvacThermostat({"occupiedHeatingSetpoint":1600}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 56085 - 1 - 10 - 513 - 4 after 10000ms)'
```

JasonSwindle, 2020-10-21:
@brubaked Using your setup, I can see MQTT is updated, but the unit never updates to the correct temp. Any advice?

JasonSwindle, 2020-11-21:
Not stale, but cannot get movement. Bump.

closed 2020-11-30 by github-actions[bot]

stegno, 2020-12-30:
I.have problem with integration also. Thermostat reporting temp and heating/idle status correctly, but set_temp works only 1 of 5 times and very laggy...I used deconz soft before and it works better with setting occupied temp.

JasonSwindle, 2021-04-17:
Bring this back, as this is still not working correctly.

gunnarsundberg, 2021-05-28:
Having this same issue as well

JasonSwindle, 2021-06-16:
@Koenkk Can we get this re-opened? The device still not work correctly.

gunnarsundberg, 2021-11-11:
Hey all! I submitted a fix for this to `zigbee-herdsman-converters` and it was merged. I'm also working on getting a cooling setpoint added.

JasonSwindle, 2021-11-12:
Thank you so much @gunnarsundberg !

ctjameson, 2022-06-04:
I'm unfortunately having this issue with a Zen stat. Current version of Zigbee2MQTT and experiencing the exact same issues as OP.

gunnarsundberg, 2022-06-05:
@ctjameson the problem is in `zigbee-herdsman-converters`. There's no cooling setpoint for the Zen thermostat. I am running a custom version that adds a cooling setpoint. I'll submit a PR when I get the chance

ctjameson, 2022-06-05:
Thanks @gunnarsundberg. How complicated is it for me to do the same until it's been integrated? Would you be willing to assist me with that? If not, that's fine as well. I can just wait.

gunnarsundberg, 2022-06-05:
I just submitted #4325

ctjameson, 2022-06-05:
Is it also possible to add a "fan only" setpoint as well? I leverage the fan in my HVAC very heavily.

gunnarsundberg, 2022-06-06:
@ctjameson it was merged so you should be able to use the cooling setpoint now! I'll look into the fan only mode

ctjameson, 2022-06-06:
@gunnarsundberg Apologies for my ignorance regarding how PRs work. I'm running A2M add-in on Home Assistant with a TI coordinator stick. I don't see an update for Z2M currently in the add-on store so I assume I'll need to wait for the next update for your PR to be available to me as a layman? Or is there some type of force update I can run to just pull the zegbee-herdsman-convertors component?

andynbaker, 2022-06-22:
Hey Z2M gurus! I finally made the jump from Deconz and have repaired all my devices to my conbeeii that is now attached to zigbee2mqtt running as a home-assistant add-on. Everything is working great, but like others here, I can't change the cooling setpoints of my three zen-01-w thermostats. I had this same problem with Deconz and worked with their devs to get that functionality added, so I know the hardware is capable of this function.

It looks like this functionality was merged with the edge branch, so I installed the edge add-on, but I'm still unable to change my cooling setpoints. A couple things I see: 1) there's now this extra temp input in the climate entity, and 2) I now get an error as shown in this screen shot:

[image]

Any thoughts?

ctjameson, 2022-06-22:
@andynbaker I'm glad someone smarter than I when it comes to these is also interested in getting this working. I can't wait to see what happens.

ctjameson, 2022-06-29:
So I'm on Zen thermostat #2 now. This one is brand new, never used. I can join it to my mesh and connect to the closest coordinator to it (about 6 feet away) but as soon as I put it back on the wall, it's like it loses ability to refresh values at all and just errors out. This is all while the screen is awake and the signal bars show as good. 

I'm still using main line Z2MQTT and not edge, so that may be why but I'm stumped on how to proceed as someone that doesn't know anything about the underlying code.

andynbaker, 2022-06-29:
I have to walk back my enthusiasm too. I’m seeing now that within a few minutes after pairing it and verifying that it works, they stop responding to the published commands. I too don’t know how to troubleshoot this, so I may sit tight until these changes are published and then create a new issue here.

shjryan, 2022-09-25:
I am experiencing the same issue as @andynbaker.

andynbaker, 2022-09-25:
I’ve been using zha since my last post and haven’t looked back. These thermostats work great in that environment.

shjryan, 2022-10-01:
@andynbaker I've used ZHA to start with, but the thermostat wasn't controllable with ZHA either... That's why I looked into using zigbee2mqtt.

A funny behaviour is that when I make a few changes using HomeKit, I can see them all in the Logbook. However, I don't see the logs in zigbee2mqtt. When I take all the batteries out of the thermostat and reinsert them, and as soon as it connects, all the temp changes get replayed on the thermostat. I really don't know what's happening, whether this is a zigbee2mqtt bug, or the thermostat issue.

andynbaker, 2022-10-01:
I have three of these and have no problems controlling them with HomeKit or Deconz.

shjryan, 2022-10-05:
@andynbaker I see. I will give ZHA another try... I've been hesitant, because it seems like I can't use both at the same time using a single Sonoff USB dongle. Thanks!