Koenkk/zigbee2mqtt #4321: Allow to tweak configureReporting parameters per 'device'

sjorge, 2020-09-09:
## Is your feature request related to a problem? Please describe
Sometimes the 'default' parameters for configureReporting are not exactly what you want in every circumstance.

For example a bunch of SNZB-02 sensors, they overall work great!
By default they are configured like this:
```
            await configureReporting.temperature(endpoint, {min: 5, change: 100});
            await configureReporting.humidity(endpoint);
```

But you might not always want them to update  at a 0.1C change or maybe you only want them to update 4x hour.

Currently I have one of these outside and it updates a lot more than the ones I have inside, because the temperature fluctuates more often. I currently manually forced it to

```
min = 300
max = 900
change = 500
```

It now updates at maximum once every 5 minutes, but will update 4x per hour if the temperature is constant, or if the temperature is changed by more than 0.5C, but I do not want this to be the default for all sensors, just the one outside on the balcony.

## Describe the solution you'd like
Also see https://github.com/Koenkk/zigbee2mqtt/issues/3812#issuecomment-688436105

I think something like this could work:

- On joining do the normal configuration flow as currently implemented
- On restart/deviceAnnounce check if the stored 'configureKey' is a number
  - yes && configureKey == store configureKey => do nothing
  - yes && configureKey > store configureKey => reconfigure
  - no && do nothing
- On `zigbee2mqtt/bridge/request/device/configure`
  - always call configure

You can then set custom values by calling a new endpoint, where you them specify what you want to configure.
e.g.

zigbe2mqtt/bridge/request/device/configure_reporting with payload `{"friendly_name": {"temperature": {"min": 300, "max": 900}}}`
zigbee2mqtt/bridge/request/device/configure with payload friendly_name

Would then call the configure flow but inside the  configureReporting.temperature() and friends, we take the values in this order:
- stored configureKey
- parameters
- default

In the case for the SNZB-02 mentioned above it will end up doing the following
humidy -> use whatever the default is
temperature -> use 'change' of 100 (as this is provided in the as a parameter for the device)
                     -> use 'min' of 300 (as overwritten in the stored configureKey for this device instance)
                     -> use 'max' of 900 (as overwritten in the stored configureKey for this device instance)

Now in the future if devices.js gets update with new values, for the device instance with different paremters nothing will happen. Unless the user explicitly calls zigbee2mqtt/bridge/request/device/configure, for instances that have not deviated from the default, the normal flow applies where we reconfigure the device because the configureKey we had for the device was lower than the one present in devices.js.

## Describe alternatives you've considered
Current work around:
- ensure all devices of the type are offline
- modify devices.js and enter prefered values
- update configureKey in devices.js to 999
- bring device that needs new config online
- verify that it got configure by checking database.db's meta.configured for the device
- restore devices.js
- bring other devices online.

## Additional context
n/a

sjorge, 2020-09-15:
Another benefit of this would be for people with lots of bulbs. Were you might one some to only report on/off and maybe other also report brightness/color.

.e.g. group of 4 bulbs,
bulb1: report as default
bulb2-4: on/off, others say 1x hour

You cut back a lot on network traffic and if opportunist group updates are on, we should still more or less have the group in sync.

That might also solve the mulitple group updates, that are the same... although fixing debounce for groups might be a better solution for the later.

maumi, 2020-10-06:
I have some sensors which report sometimes not so often (sometimes after 5 hours). 
They are WSDCGQ01LM
What I try to understand is if and how your request could help me. 
My understanding is that battery powered zb-devices sends requests when they think they need to and you cannot pull values from them.

sjorge, 2020-10-06:
@maumi if the device is using reporting, this will allow you to configure them to your liking if required.
reporting usually have the following knobs to tweak: minimum interval, maximum interval and change

minimum interval -> wait atleast X before reporting
e.g. a sensor is currently reporting between 1 and 5 minutes, say because it is in direct sunlight (blah), if you set minimum to 15 minutes... it won't report more than that.
maximum interval -> maximum time between updates. e.g. only refreshing once every 4 hours and you want it sooner? you can set this.
change => maybe you don't care about 0.1C increments for the outdoor sensor, you can change that to 0.5 or maybe even 1.0C. Or the reverse, maybe you have one near your servers/switches and do care about tiny increments.

No if the device is not using reporting, e.g. some TuYa stuff, this won't help at all.

But like Koen set, if might also be that the defaults just need some tweaking.

sjorge, 2020-10-06:
I just looked up the WSDCGQ01LM, sadly it's a Xiaomi/Aqara/Maiji device, they don't do reporting either. So this won't help.

maumi, 2020-10-06:
Okay thanks. Good to know.

maumi, 2020-10-06:
I've ordered one SNZB-02 to play with. 
If it's okay for me to set them all the same I could easily change parameters in devices.js and all my new devices (SNZB-02) will use these parameters, right?

sjorge, 2020-10-06:
Correct, if you want them all the same, you could just changes devices.js...

This feature request is for if you want some of them to have different settings...
or if you're needs are other than the default and want all of them to be different, you could do that too without editing devices.js... if this gets implemented.

sjorge, 2020-11-06:
Still think this would be a useful feature

sjorge, 2020-11-18:
@Koenkk any better idea to on how to implement something like this than what I proposed?
I have a free day this weekend and could work on it some if there is a clear direction we should be going about this.

Koenkk, 2020-11-19:
I think there are two possible solutions here:
- Via api: conceptually the same as [`zigbee2mqtt/bridge/request/device/bind`](https://github.com/Koenkk/zigbee2mqtt.io/blob/new_api/docs/information/mqtt_topics_and_message_structure.md#device)
- Via configuration.yaml: per device you specify the min/max/change overrides per cluster/attributes

Braindump:
- via api will be MUCH easier to implement
- via api will allow for easy configuration via the frontend (similar to how bind currently works, with dropdowns to select clusters/attributes)
- via configuration.yaml, figuring out the clusters and attributes can be hard and it is easy to make typos. Changing the clusters names and attributes would also be a breaking change if we do it like this
- via api will allow to configureReporting for attributes not in the `configure()`
- changes done via api will be lost when the device is reconfigured or the configureKey is increased (maybe not a problem, see *)
- via configuration.yaml: how to deal with the whole configureKey thing? (maybe not a problem, see *)
- via configuration.yaml: will require restarting zigbee2mqtt 

(*) This is another discussion, but I'm wondering why we need the configureKey after all.
- Bumping the configureKey and thus triggering a reconfigure on zigbee2mqtt doesn't work well with battery powered devices (because they are sleeping), this leads to errors in the startup log
-  With the frontend, people can very easily trigger a reconfigure of the device in case they have problems with it 
- `configure()` for a supported device is rarely changed 
- If the device works well for a user, why should the `configure()` be executed again if the configureKey is increased? In case the user has problems he will probably try to repair (leading to a leave and configure())

sjorge, 2020-11-19:
I'm in favor of getting rid of configureKey, like you said if a user is having issues we can tell him to repair or better yet, hit configure via the frontend.

Exposing it via the API is probably the way to go, that would potentially also allow someone working on a new device to get started more quickly by trying to configure reporting on different attributes.

How would we cleanly expose such a thing via API though... we can probably get the list of attribute from the cluster info, but can we read the current config somehow?

How or would it just be a 'set' only type of deal? How would way, say handle Trådfri bulbs, they require reconfiguration every deviceAnnounce if we don't have a way to 'get' the current reporting config?

Pigotka, 2020-11-19:
Oh I had no idea there is such cool discussion already. 
Why snzb-02 does not work the smart way as aquara sensors? They don't need to be configured and they report temperature when they feel it is important or at least I have the feeling from looking at the value history.

Second if we really need to configure sonoff device I would allow per device configuration. My suggestion in this case would be to use devices.yaml as device specific configuration is already located there.

Pigotka, 2020-11-19:
Also to note `configure()` does not work for SNZB-02 at all. Device need to be for removed and repaired every time.

sjorge, 2020-11-19:
> Also to note `configure()` does not work for SNZB-02 at all. Device need to be for removed and repaired every time.

Good point, would but it would be hard to specify these in device.yaml too... I wonder if they do reconfigure if we don’t remove them and just reset them, and maybe for battery device hold the new commands until we see a deviceAnnohncement or something

Koenkk, 2020-11-19:
- Reading the current config is something that can be added later, this would involve storing it zigbee-herdsman similar like how it is done for bind, lets first focus on the MVP
- This won't work for TRADFRI and SNZB-02 indeed. For the TRADFRI a good alternative is the report feature, SNZB-02 should be documented properly. Also I think the main use case for this feature is changing reportings of temperature and power related things, I don't see a good use case for TRADFRI bulbs yet.

Also involving @nurikk (frontend) and @kirovilya (for dropping `configureKey`) here

nurikk, 2020-11-21:
Eventually this can brings us to metadata driven converter, for zcl compatible devices this would be just a configuration what to bind and reporting config

sjorge, 2020-11-21:
Been playing some more with one of my SNZB-02 that don't allow configuration after the initial one, for these it looks like it would indeed somehow be useful:

specify a default for the device in devices.js (or somewhere)
allow specifying a per-device default in configuration.yaml/devices.yaml based on the IEEE address
specifying them via de API (and when doing so store them in configuration.yaml)

That would allow for the SNZB-02 to be forced removed, have them added to configuration.yaml before repairing... 

Keeping the info in database.db would probably be cleaner, but since that gets removed on a force remove of a device, that won't work for a SNZB-02... and maybe other... still not sure why I can't just reset the device and have it join again though, the device_announce happens but the device never succesfully rejoins of I just reset it with joining enabled before removing it first.

I wonder if some steps we do during initial interviewing is what keeps the device wake that won't happen when z2m doesn't kick of the interviewing step again.

sjorge, 2020-12-22:
Unstale, koenkk can the stale bot ignore feature requests?

Koenkk, 2020-12-22:
@sjorge no, it's a good reminder and moment to check if feature requests are still relevant.

sjorge, 2020-12-30:
Looks like this now sort of works, with the recent change I can change the config, hit apply and wake up the device.
Or in the case of the SNSB-02 'reset' the device and it gets applied when it joins and have about ~30 seconds window to push the config after.

closed 2020-12-30 by sjorge

cadavre, 2020-12-30:
What is the "recent change"? Could you contribute on how to re-configure i.e. SNSB-02? There will be many ppl looking for this, getting into this – without actual answer to the question on closed issue.

sjorge, 2020-12-30:
Very recent, I just queue op a change via mqtt or api, then insert the battery again, seems to take the change.

~ sjorge

Pigotka, 2021-01-09:
@sjorge I'm sorry but I don't see any option to tweak configureReporting per device as this feature requested. Could you link PR or documentation how I could set configureReporting for one of my SNZB-02 and keep other untouched?

sjorge, 2021-01-09:
[image]

You can change a value, hit apply and then hold the button on the sensor to reset it. Takes a bit of trial and error to get the timing right. But it seems to take the new settings.

maumi, 2021-01-09:
> @sjorge I'm sorry but I don't see any option to tweak configureReporting per device as this feature requested. Could you link PR or documentation how I could set configureReporting for one of my SNZB-02 and keep other untouched?

You can do it in the http webserver for example.

maumi, 2021-01-09:
> [image]
> 
> You can change a value, hit apply and then hold the button on the sensor to reset it. Takes a bit of trial and error to get the timing right. But it seems to take the new settings.

Are you sure you need to reset it?
I think it is enough to wake the device up with a single press. That's what I did and it seems to work.

sjorge, 2021-01-09:
Hmm single press did not work for me, I just tried. Looks like long press doesn't work either.

I had to force remove the device, long press to have it join, I had about 10-15 seconds after the 'Successfully configured device XXX' to make changes via the web ui and those got applied to, after that I got timeouts.

maumi, 2021-01-09:
> Hmm single press did not work for me, I just tried. Looks like long press doesn't work either.
>
> I had to force remove the device, long press to have it join, I had about 10-15 seconds after the [...]

One device only needed one single press. Another needed multiple single presses. I think it is some timing problem. 
so in short. Hit apply, do multiple single button press until the green message with success pop up.

sjorge, 2021-01-09:
I grabbed all 5 of mine...

Indeed seems to be a timing issue. You don't need to long press.
Works: 1. hit apply, 2. short press
Worked better: 1. remove battery, 2. insert battery, 3. hit apply after seeing 'deviceAnnouncement' in the log

Removing the battery and hitting apply after the deviceAnnouncement gave me the best success rate.

Pigotka, 2021-01-09:
That is awesome guys. Thanks a lot.

cadavre, 2021-01-11:
I cannot see any endpoint to modify on "Reporting" tab:

[image]

Any idea why?

maumi, 2021-01-11:
> I cannot see any endpoint to modify on "Reporting" tab:
> 
> [image]
> 
> Any idea why?

What kind of device? SNZB-O2?
Or a device without reporting?

cadavre, 2021-01-11:
> What kind of device? SNZB-O2?
> Or a device without reporting?

This particular SNZB-02. But I cannot see anything on this tab for any other sensor (I assume Xiaomi sensors doesn't support this).

sjorge, 2021-01-11:
I think of the device was paired before the feature got added to z2m, the table might be empty.

~ sjorge

Pigotka, 2021-01-11:
Yes. Seeing the same. All sensors added before have no configureReport entries. Try repairing it.