Koenkk/zigbee2mqtt #4322: Unable to join an Osram Smart+ motion sensor using Conbee II stick.

jonathantullett, 2020-09-09:
## What happened

Using a Conbee II adapter with Home Assistant on a Raspberry Pi4 and trying to join an Osram Smart+ motion sensor.

Device doesn't join, keeps failing the interview.  I have tried several of the devices.

I do see a device created within HA, but there's no information published.

## What did you expect to happen

I expect the device to join the network and start sending data from the sensors.

## How to reproduce it (minimal and precise)

Reset the device by removing the battery for 10 seconds and holding the button on the side while inserting the battery. Various lengths of time tested from 3 to 10+ seconds.

## Debug info
Zigbee2MQTT version:  Zigbee2MQTT 1.14.4-dev
Adapter hardware: Conbee II
Adapter firmware version: 0x26580700

The logs from the joining attempts can be seen at https://pastebin.com/VmmR1vrY

I have tried another sensor - the Smart Things one, and that was successful.

Koenkk, 2020-09-09:
Please provide the herdsman debug logging when joining the device.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

jonathantullett, 2020-09-09:
I am having no luck enabling the herdsman debug logging.

If I include `zigbee_herdsman_debug: true` in the root of configuration.yaml, it's removed on restart, same if it's included in the `advanced` section.

I am running as a Hass.io add-on so trying to do through the yaml (both direct on the file and through the UI).

Are you able to advise?  Thank you.

jonathantullett, 2020-09-09:
I have captured the pairing debug into the following and it's available at https://pastebin.com/7gW1NGsN

Please let me know if there's anything else I can provide to assist.

Koenkk, 2020-09-10:
For some reason the device keeps rejoining again and again (in the log: `Device '0x000d6f000e63a27f' joined`). Do you perhaps have a CC2531 to sniff traffic with? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

jonathantullett, 2020-09-10:
Unfortunately not.  I have ordered a Electrolama zig-a-zig-ah! stick if that would help?

I note the sensors work perfectly with SmartThings, so they're functioning.

Koenkk, 2020-09-10:
@jonathantullett once you receive your ZZH, could you check if the OSRAM sensor can join when running z2m with a zzh?

jonathantullett, 2020-09-10:
Certainly.  I'll let you know how that goes.  Fingers crossed it'll be here tomorrow so not long to wait.

Thanks for your help so far.

jonathantullett, 2020-09-11:
I have just configured the ZZH and the Osram has paired immediately and is streaming information through mqtt:
```
{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x000d6f000e63a27f"],"manufacturer":"OSRAM","model":"SMART+ Motion Sensor (AC01353010G)","name":"0x000d6f000e63a27f","sw_version":"Zigbee2MQTT 1.14.4-dev"},"device_class":"motion","json_attributes_topic":"zigbee2mqtt/0x000d6f000e63a27f","name":"0x000d6f000e63a27f_occupancy","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x000d6f000e63a27f","unique_id":"0x000d6f000e63a27f_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}
{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x000d6f000e63a27f"],"manufacturer":"OSRAM","model":"SMART+ Motion Sensor (AC01353010G)","name":"0x000d6f000e63a27f","sw_version":"Zigbee2MQTT 1.14.4-dev"},"json_attributes_topic":"zigbee2mqtt/0x000d6f000e63a27f","name":"0x000d6f000e63a27f_tamper","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0x000d6f000e63a27f","unique_id":"0x000d6f000e63a27f_tamper_zigbee2mqtt","value_template":"{{ value_json.tamper }}"}
{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x000d6f000e63a27f"],"manufacturer":"OSRAM","model":"SMART+ Motion Sensor (AC01353010G)","name":"0x000d6f000e63a27f","sw_version":"Zigbee2MQTT 1.14.4-dev"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x000d6f000e63a27f","name":"0x000d6f000e63a27f_battery_low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x000d6f000e63a27f","unique_id":"0x000d6f000e63a27f_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low}}"}
{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x286d9700010916dd"],"manufacturer":"SmartThings","model":"Motion sensor (2018 model) (IM6001-MTP01)","name":"0x286d9700010916dd","sw_version":"Zigbee2MQTT 1.14.4-dev"},"device_class":"motion","json_attributes_topic":"zigbee2mqtt/0x286d9700010916dd","name":"0x286d9700010916dd_occupancy","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x286d9700010916dd","unique_id":"0x286d9700010916dd_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}
{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x000d6f000e63a27f"],"manufacturer":"OSRAM","model":"SMART+ Motion Sensor (AC01353010G)","name":"0x000d6f000e63a27f","sw_version":"Zigbee2MQTT 1.14.4-dev"},"device_class":"temperature","json_attributes_topic":"zigbee2mqtt/0x000d6f000e63a27f","name":"0x000d6f000e63a27f_temperature","state_topic":"zigbee2mqtt/0x000d6f000e63a27f","unique_id":"0x000d6f000e63a27f_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}
{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x000d6f000e63a27f"],"manufacturer":"OSRAM","model":"SMART+ Motion Sensor (AC01353010G)","name":"0x000d6f000e63a27f","sw_version":"Zigbee2MQTT 1.14.4-dev"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x000d6f000e63a27f","name":"0x000d6f000e63a27f_linkquality","state_topic":"zigbee2mqtt/0x000d6f000e63a27f","unique_id":"0x000d6f000e63a27f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}
{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x286d9700010916dd"],"manufacturer":"SmartThings","model":"Motion sensor (2018 model) (IM6001-MTP01)","name":"0x286d9700010916dd","sw_version":"Zigbee2MQTT 1.14.4-dev"},"device_class":"temperature","json_attributes_topic":"zigbee2mqtt/0x286d9700010916dd","name":"0x286d9700010916dd_temperature","state_topic":"zigbee2mqtt/0x286d9700010916dd","unique_id":"0x286d9700010916dd_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}

{"battery_low":false,"linkquality":78,"occupancy":true,"tamper":true}
{"battery_low":false,"linkquality":60,"occupancy":true,"tamper":true}
{"message":"interview_successful","meta":{"description":"SMART+ Motion Sensor","friendly_name":"0x000d6f000e63a27f","model":"AC01353010G","supported":true,"vendor":"OSRAM"},"type":"pairing"}
{"battery_low":false,"linkquality":75,"occupancy":false,"tamper":true}
{"battery_low":false,"linkquality":78,"occupancy":false,"tamper":true,"temperature":20.36}
{"battery_low":false,"linkquality":84,"occupancy":true,"tamper":true,"temperature":20.36}
```

Koenkk, 2020-09-11:
@ChrisHae looks like this is Conbee specific, do you know of any known issues with the osram sensor?

elgeniskogen, 2020-09-11:
Is it so that the supported devices for Conbee II on Zigbee2MQTT is limited to the supported device list from Conbee? I have notised that Zigbee2MQTT havs a much longer list of supported devices than what Conbee is claiming on their site. Is the Zigbee2MQTT devicde support independent of the HW?

Koenkk, 2020-09-11:
@elgeniskogen device support is independent of the hardware, but as been seen in this issue there are some very rare exceptions to this

elgeniskogen, 2020-09-12:
Thanks. So I can buy Conbee II (which is easiest available for me) and be future proof with Zigbee2MQTT? I would like to have support for the SmartThings stuff that DeConz doesn't.

Koenkk, 2020-09-12:
Yes, no problem

ChrisHae, 2020-09-15:
I have also a osram smart+ motion sensor and tested it. It interviewed and configured without problems. I also sniffed the traffic with wireshark and didn't see why it would joind repeatedly.
But I used a brand new FW for conbee2. Perhaps you want to try it for yourself: http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26650700.bin.GCF

I think the error could also be if the device is to far away from the coordinator or if you have to less routers in your network.
Could you tell me please how many devices (enddevices and routers) do you have?

ChrisHae, 2020-09-15:
P.S. reading corrdinator version is bugged with this new version but I'm already on it.

jonathantullett, 2020-09-15:
I was nearly the end of my return period for the Conbee so it's been returned to the vendor.  The device was between 50cm and 5cm (I tried difference placement) from the controller, so I don't believe it was a range issue.

i was running the latest firmware at the time (0x26580700), and it was the only router running for part of the testing as I powered off the SmartThings hub.

elgeniskogen, 2020-09-15:
I was planning to buy Conbee II (which is easily available here in Norway), but after reading this I became a bit sceptical after all. Is the device support HW dependent in Z2M so I should buy a Electrolama zig-a-zig-ah!  after all for best possible compatibility?!

jonathantullett, 2020-09-16:
I can only speak from my very limited experience. I have had zero problems with the zig-a-zig-ah, but I lost an entire day to this weirdness on a Conbee II.

ChrisHae, 2020-09-16:
Device support is not HW depended. It is implemented in software. The Osram Smart+ Sensor runs fine in my setup with ConBee2. Not trying to defend the ConBee but: In a zigbee network there can be many error sources depending on the number and types of devices, their distances, firmware, the firmware of the coordinator or the software. You can see those issues also for all other zigbee HW in the corresponding forum threads.

closed 2020-10-24 by github-actions[bot]

Pockrandt, 2020-12-21:
can confirm this likely is a bug in the conbee implementation.
just switched from CC2531 to conbee II (current fw)
(sadly no access to the debugger to flash sniffer fw)

- keep config, no re-pair => Osram Motion not sending data (only once  temperature when powering up)
    - all other devices work (Hue lights, Xiaomi motion, Xiaomi door contact, Osram Plugs, Osram lights)
- remove all devices, new pan id, new network key with conbee II => osram motion fails to join 
    - icon in webguide is that generic red one
    - Device itself is blinking blue, blue, green  (as in joining/pairing) 
    - direct in front of the stick with no other devices joined/powered on
- replace conbee with CC2531 > motion sensor can join and work as expected
- stop z2m and setup home assistants own zha, join all devices => sensors joins & works even when joined through 2 lights