Koenkk/zigbee2mqtt #4323: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

NE8XT, 2020-09-09:
Zigbee2MQTT version:1.14.4
Adapter hardware: CC2531
**After updating:**
pi@ZigBee:~$ cd /opt/zigbee2mqtt && DEBUG=* npm start

> zigbee2mqtt@1.14.4 start /opt/zigbee2mqtt
> node index.js

winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warn"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "http"
winston:create-logger: Define prototype method for "verbose"
winston:create-logger: Define prototype method for "debug"
winston:create-logger: Define prototype method for "silly"
winston:create-logger: Define prototype method for "emerg"
winston:create-logger: Define prototype method for "alert"
winston:create-logger: Define prototype method for "crit"
winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warning"
winston:create-logger: Define prototype method for "notice"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "debug"
Zigbee2MQTT:info  2020-09-09 08:32:23: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-09-09.08-32-23' filename: log.txt
winston:file: written true false
Zigbee2MQTT:debug 2020-09-09 08:32:23: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-09-09.07-30-59'
winston:file: written true false
Zigbee2MQTT:debug 2020-09-09 08:32:25: Loaded state from file /opt/zigbee2mqtt/data/state.json
winston:file: written true false
winston:file: logged 127 info  2020-09-09 08:32:23: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-09-09.08-32-23' filename: log.txt

winston:file: logged 229 debug 2020-09-09 08:32:23: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-09-09.07-30-59'

winston:file: logged 312 debug 2020-09-09 08:32:25: Loaded state from file /opt/zigbee2mqtt/data/state.json

winston:file: ENOENT ok /opt/zigbee2mqtt/data/log/2020-09-09.08-32-23/log.txt
winston:file: stat done: log.txt { size: 0 }
winston:file: create stream start /opt/zigbee2mqtt/data/log/2020-09-09.08-32-23/log.txt { flags: 'a' }
winston:file: create stream ok /opt/zigbee2mqtt/data/log/2020-09-09.08-32-23/log.txt
winston:file: file open ok /opt/zigbee2mqtt/data/log/2020-09-09.08-32-23/log.txt
Zigbee2MQTT:info  2020-09-09 08:32:25: Starting Zigbee2MQTT version 1.14.4 (commit #037d54a)
winston:file: written true false
Zigbee2MQTT:info  2020-09-09 08:32:25: Starting zigbee-herdsman...
winston:file: written true false
Zigbee2MQTT:debug 2020-09-09 08:32:25: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
winston:file: written true false
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
winston:file: logged 81 info  2020-09-09 08:32:25: Starting Zigbee2MQTT version 1.14.4 (commit #037d54a)

winston:file: logged 136 info  2020-09-09 08:32:25: Starting zigbee-herdsman...

winston:file: logged 563 debug 2020-09-09 08:32:25: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'

  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyACM0 +457ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +9ms
  serialport/stream Binding #open had an error [Error: Error Resource temporarily unavailable Cannot lock port] +7ms
Zigbee2MQTT:error 2020-09-09 08:32:26: Error while starting zigbee-herdsman
winston:file: written true false
Zigbee2MQTT:error 2020-09-09 08:32:26: Failed to start zigbee
winston:file: written true false
Zigbee2MQTT:error 2020-09-09 08:32:26: Exiting...
winston:file: written true false
Zigbee2MQTT:error 2020-09-09 08:32:26: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
winston:file: written true false
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.4 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.14.4 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-09T06_32_26_492Z-debug.log

Koenkk, 2020-09-09:
Check https://www.zigbee2mqtt.io/information/FAQ.html#error-resource-temporarily-unavailable-cannot-lock-port

NE8XT, 2020-09-09:
pi@ZigBee:~$ ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0
ls: cannot open directory '/proc/1/fd/': Permission denied
ls: cannot open directory '/proc/2/fd/': Permission denied
ls: cannot open directory '/proc/3/fd/': Permission denied
ls: cannot open directory '/proc/4/fd/': Permission denied
ls: cannot open directory '/proc/5/fd/': Permission denied
ls: cannot open directory '/proc/6/fd/': Permission denied
ls: cannot open directory '/proc/8/fd/': Permission denied
ls: cannot open directory '/proc/9/fd/': Permission denied

Pferdebockwurst, 2020-09-09:
Try with root rights, eg
`sudo ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0`

NE8XT, 2020-09-09:
pi@ZigBee:~$ ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0
ls: cannot open directory '/proc/1/fd/': Permission denied
ls: cannot open directory '/proc/2/fd/': Permission denied
ls: cannot open directory '/proc/3/fd/': Permission denied
ls: cannot open directory '/proc/4/fd/': Permission denied
ls: cannot open directory '/proc/8/fd/': Permission denied
ls: cannot open directory '/proc/9/fd/': Permission denied
pi@ZigBee:~$ sudo ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0
pi@ZigBee:~$

NE8XT, 2020-09-09:
pi@ZigBee:~$ sudo journalctl -u zigbee2mqtt.service -o cat -f
Zigbee2MQTT:info  2020-09-09 21:06:57: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2020-09-09 21:06:57: Currently 0 devices are joined:
Zigbee2MQTT:warn  2020-09-09 21:06:57: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-09-09 21:06:57: Allowing new devices to join.
Zigbee2MQTT:warn  2020-09-09 21:06:57: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2020-09-09 21:06:57: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2020-09-09 21:06:57: Connecting to MQTT server at mqtt://11.11.11.168:1883
Zigbee2MQTT:info  2020-09-09 21:06:58: Connected to MQTT server
Zigbee2MQTT:info  2020-09-09 21:06:58: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2020-09-09 21:06:58: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"037d54a","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.14.4"}'

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/5197667/debug.txt)

NE8XT, 2020-09-24:
moved to Openwrt and Zigate!

closed 2020-09-24 by NE8XT