Koenkk/zigbee2mqtt #4324: Miele DGC 6805 XL

Pferdebockwurst, 2020-09-09:
## Information about the device + link
Miele DGC 6805 combined steamer and oven
https://www.miele.lu/de/c/combi-dampfgarer-1473.htm?mat=09505230&name=DGC_6805
https://www.miele.de/pmedia/ZGA/TX0021/9848220-000-01_09848220_001_0214.pdf
https://www.miele.de/pmedia/ZGA/TX2070/9895770-000-00_09895770_000.pdf

This is one of the earlier smart home capable devices from Miele. Later they switched to Wifi+Cloud, but this device (and several others) are using Zigbee. It is intended to be used with a gateway from Miele (XGW 3000) or Qivicon (Home Base).

The device was able to pair my zigbee network:

`info  2020-09-09 20:20:42: Device '0x001d63fffe0206fc' joined
info  2020-09-09 20:20:42: Starting interview of '0x001d63fffe0206fc'
info  2020-09-09 20:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001d63fffe0206fc"},"type":"device_connected"}'
info  2020-09-09 20:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001d63fffe0206fc"},"type":"pairing"}'
info  2020-09-09 20:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x001d63fffe0206fc"},"type":"device_announced"}'
info  2020-09-09 20:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x001d63fffe0206fc"},"type":"device_announced"}'
info  2020-09-09 20:20:42: Successfully interviewed '0x001d63fffe0206fc', device has successfully been paired
warn  2020-09-09 20:20:42: Device '0x001d63fffe0206fc' with Zigbee model 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2020-09-09 20:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x001d63fffe0206fc","supported":false},"type":"pairing"}'
info  2020-09-09 20:34:25: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'`

Unfortunately i wasn't able to get a report about unknown actions, even after triggering them by using the combi steamer. Maybe there is some special configuration needed, like the documentation suggests, but I don't know what and where to start. I'm really lost now, so please tell me how to go on. I could also buy the gateway, they only cost about 10 Euros, and try to sniff some traffic...

Pferdebockwurst, 2020-09-12:
@Koenkk Sorry for the reminder - what do you think about this one?

Koenkk, 2020-09-13:
@Pferdebockwurst sniffing the traffic with the original gateway should allow you to figure out what commands can be send (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

Pferdebockwurst, 2020-09-13:
OK, then I will buy the gateway and try to sniff the traffic. But I hope
you will help to analyze the traffic...?

----------

(https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

https://github.com/Koenkk/zigbee2mqtt/issues/4324#issuecomment-691669854

Koenkk, 2020-09-13:
I can only provide limited support

Pferdebockwurst, 2020-09-13:
That should be enough. Thanks, that keeps me motivated.

----------

https://github.com/Koenkk/zigbee2mqtt/issues/4324#issuecomment-691672230

closed 2020-10-28 by github-actions[bot]