Koenkk/zigbee2mqtt #4325: report color

zhapwey, 2020-09-10:
Is there a way to control how a light bulb returns the color to you, because I see that some use x, and others hue, x ... so that they all return their state the same?.

sjorge, 2020-09-10:
It depends on the bulb... and some other factors

Via reporting bulbs can report:
- currentX and currentY, most RGB bulbs like trådfri and hue do this.
- currentHue and currentSaturation, innr and hue do this for example (some like hue use enhancedHue vs normal hue)
- colorTemperature, most white spectrum bulbs do this... and some rgb bulbs...

There is also a 'colorMode' which sometimes gets returned to indicate which one of the above is currently in use. But I've never seen it been update for my bulbs even when i switch setting the color via temp and xy.

For example my innr bulbs return both a xy combo and a hue/sat combo at the same time.

Now when you set the color via zigbee2mqtt there is a 3rd option where it will also return what you set.
So if you set it using `{"color": {"r": 255, "g": 0, "b": 0}}` it will return that too.

For sure this is something that can be improved, but I am not sure what the right thing to do is. As different people want different things.

e.g. personally I never want it to return what i've set but prefer it to return either xy or hue/sat. But other people want it to return exactly what has been set.

A sort of 'prefered' color payload as you mentioned could work I guess. But that means a whole lot more converting. As of now xy, temp and hue/sat are mostly passed to the bulb as-is, while others get converted to one of those (all xy I think).

Not all conversion are 'lossless' for example rgb has a brightness element too it too, that gets lost in xy, although one could send it out as the actual level for the bulb I guess, but 'pure red' for example looks very much the same on my innr bulbs at 50 or 100% brightness, while on my osram bulbs at 100% brightness it looks pinkish.

zhapwey, 2020-09-10:
Thanks for your answer.
I would see interesting that the report of the bulbs could be defined in the configuration but it is clear that it is a personal opinion. The issue I see in that if I write the color it returns xy and if it reports the bulb it returns hue x.
One more question, is there a conversion between "xy" and "hue y x", how do you go from one to the other? The rest of the conversions are clear to me, but the issue of the lost hole?

sjorge, 2020-09-10:
AFAIK we're not doing any conversions between xy and hue/sat, if both are present, the bulb probably reports both.

We only do conversion from rgb/hsv/hsb/hex -> xy (unless I missed another one somewhere), the only way to get these in the mqtt message is also by first setting it, you'll never get those from the bulb reporting it state.

zhapwey, 2020-09-10:
What happens to me is that when I change the color of a bulb (philips) it is that the bulb reports the color in xy
zigbee2mqtt/0x00178801064c84cb/set {"color":"#11ff00"}
zigbee2mqtt/0x00178801064c84cb {"color":{"x":0.1746,"y":0.7449},"color_temp":128}
zigbee2mqtt/0x00178801064c84cb {"color":{"x":0.17,"y":0.7},"color_temp":153,"linkquality":69}

When I receive a report from the automatic bulb.
zigbee2mqtt/0x00178801064c84cb {"color":{"hue":142,"x":0.17},"color_temp":153,"linkquality":72}

Is the light bulb the one that reports in the types of color?

sjorge, 2020-09-10:
If you enable debug logging, it will show the reported values (if any)

~ sjorge

zhapwey, 2020-09-10:
I will review, I had not thought, thank you very much

zhapwey, 2020-09-10:
You are absolutely right, when I change the light I report xy and then in the automatic reports it reports as hue, it was strange to me that it worked like this, very grateful for your time.

sjorge, 2020-09-10:
Is it a modern Hue bulb? I only have an older (but up to date) E14 RGB candle, and it doesn't report anything (which is kind of disappointing). If so, it might be worth trying to configure reporting for the device to see if we can get it to report at least currentSaturation in addition to the hue you are now getting. And maybe even currentX and currentY... just with currentHue, we only have half the picture :(

zhapwey, 2020-09-10:
It is one of the new ones, your message gave me the hint to update the bulb that had an update available, and now it reports everything, xy and hue.
With what I think it was my mistake for having the bulb without updating.

Thank you very much once and I apologize.

closed 2020-10-18 by github-actions[bot]