Koenkk/zigbee2mqtt #4328: Support for Lonsonho 4 Button Wireless Wall Switch

milobt, 2020-09-10:
## Information about the device + link
Lonsonho 4 button wireless wall switch
Tuya compatible device
Each button supports click, double click and lone press
https://www.aliexpress.com/item/1005001287093949.html?spm=a2g0s.9042311.0.0.13744c4d8FaS7m

## data/database.db entry of the device

`{"id":51,"type":"EndDevice","ieeeAddr":"0xbc33acfffe5e6f93","nwkAddr":21923,"manufId":4098,"epList":[1,4,3,2],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0012","65506":28,"appVersion":66}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"epId":4,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":255,"interviewCompleted":false,"meta":{},"lastSeen":1599766546290} `

Koenkk, 2020-09-11:
Please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

milobt, 2020-10-13:
Closing, as support for this device has since been added.

closed 2020-10-13 by milobt