Koenkk/zigbee2mqtt #4329: Support for Tuya USB Zigbee Repeater

milobt, 2020-09-10:
## Information about the device + link

Pairs as Water leak detector (TS0207)
https://www.aliexpress.com/item/4001230151655.html

## data/database.db entry of the device
`{"id":62,"type":"Router","ieeeAddr":"0x14b457fffe364323","nwkAddr":29180,"manufId":4098,"manufName":"_TZ3000_m0vaazab","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":22,"modelId":"TS0207","manufacturerName":"_TZ3000_m0vaazab","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1599393638227} `

Koenkk, 2020-09-11:
Please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Diddlik, 2020-09-26:
Hi, I tried to add it as a new device. The problem is that a device with ID: TS0207 (https://www.zigbee2mqtt.io/devices/TS0207.html) already exists. And if you edit devices.js according to the instructions, this repeater will still be detected as a water leak sensor.

Koenkk, 2020-09-26:
@arteck added support for this device (https://github.com/Koenkk/zigbee-herdsman-converters/pull/1581). Supported in the dev branch now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-26 by Koenkk

ahmaddxb, 2021-06-23:
I have this issue and im on v1.19.1